MichaIng/DietPi #4559: Dietpi upgrade issue - apt-utils is not installed

tfretze, 2021-07-15:
#### Details:                                                                                                                                                                                                                    
- Date           | Thu Jul 15 07:26:53 BST 2021                                                                                                                                                                                  
- DietPi version | v7.1.2 (MichaIng/master)                                                                                                                                                                                      
- Image creator  | DietPi Core Team                                                                                                                                                                                              
- Pre-image      | Raspbian Lite                                                                                                                                                                                                 
- Hardware       | RPi 3 Model B (armv7l) (ID=3)                                                                                                                                                                                 
- Kernel version | Linux DietPi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux                                                                                                                              
- Distro         | buster (ID=5,RASPBIAN=1)                                                                                                                                                                                      
- Command        | apt-get -qq upgrade                                                                                                                                                                                           
- Exit code      | 100                                                                                                                                                                                                           
- Software title | DietPi-Update                                                                                                                                                                                                 
#### Steps to reproduce:                                                                                                                                                                                                         
                                                                                                                                                                                      
1. entering the command dietpi-update                                                                                                                                                                                                                           
2. ...                                                                                                                                                                                                                           
#### Expected behaviour:                                                                                                                                                                                                         
update the version of dietpi                                                                                                                                                                                                    
- ...                                                                                                                                                                                                                            
#### Actual behaviour:                                                                                                                                                                                                           
upgrade is failing                                                                                                                                                                                                    
- ...                                                                                                                                                                                                                            
#### Extra details:                                                                                                                                                                                                              
                                                                                                                                                           
- ...                                                                                                                                                                                                                            
#### Additional logs:                                                                                                                                                                                                            
```                                                                                                                                                                                                                              
debconf: delaying package configuration, since apt-utils is not installed                                                                                                                                                        
dpkg: warning: files list file for package 'php7.3-cgi' missing; assuming package has no files currently installed                                                                                                               
dpkg: warning: files list file for package 'lighttpd' missing; assuming package has no files currently installed                                                                                                                 
dpkg: warning: files list file for package 'pinentry-curses' missing; assuming package has no files currently installed                                                                                                          
dpkg: warning: files list file for package 'libdns1104:armhf' missing; assuming package has no files currently installed                                                                                                         
dpkg: warning: files list file for package 'libzstd1:armhf' missing; assuming package has no files currently installed                                                                                                           
dpkg: warning: files list file for package 'libjson-c3:armhf' missing; assuming package has no files currently installed                                                                                                         
dpkg: warning: files list file for package 'systemd-sysv' missing; assuming package has no files currently installed                                                                                                             
dpkg: warning: files list file for package 'adduser' missing; assuming package has no files currently installed                                                                                                                  
dpkg: warning: files list file for package 'libc-l10n' missing; assuming package has no files currently installed                                                                                                                
dpkg: warning: files list file for package 'php7.3-sqlite3' missing; assuming package has no files currently installed                                                                                                           
dpkg: warning: files list file for package 'apt' missing; assuming package has no files currently installed                                                                                                                      
[... cut ...]
Errors were encountered while processing:                                                                                                                                                                                        
 /var/cache/apt/archives/base-files_10.3+rpi1+deb10u10_armhf.deb                                                                                                                                                                 
E: Sub-process /usr/bin/dpkg returned an error code (1)                                                                                                                                                                          
```

Joulinar, 2021-07-15:
Many thanks for your report.
```
Structure needs cleaning
```
This indicates a filesystem corruption.

Please try to have it repaired on reboot and check back fsck log:
```
> /forcefsck
reboot
# after reboot
journalctl -t systemd-fsck
```

tfretze, 2021-07-15:
this is the output from the commands you asked me to enter

[image]

should i attempt update again?

tfretze, 2021-07-15:
#### Details:                                                                                                                                                                     
- Date           | Thu Jul 15 11:04:59 BST 2021                                                                                                                                   
- DietPi version | v7.1.2 (MichaIng/master)                                                                                                                                       
- Image creator  | DietPi Core Team                                                                                                                                               
- Pre-image      | Raspbian Lite                                                                                                                                                  
- Hardware       | RPi 3 Model B (armv7l) (ID=3)                                                                                                                                  
- Kernel version | Linux DietPi 5.10.17-v7+ #1421 SMP Thu May 27 13:59:01 BST 2021 armv7l GNU/Linux                                                                               
- Distro         | buster (ID=5,RASPBIAN=1)                                                                                                                                       
- Command        | apt-get -qq upgrade                                                                                                                                            
- Exit code      | 100                                                                                                                                                            
- Software title | DietPi-Update                                                                                                                                                  
#### Steps to reproduce:                                                                                                                                                          
                                                                                                                                       
1. ...                                                                                                                                                                            
2. ...                                                                                                                                                                            
#### Expected behaviour:                                                                                                                                                          
                                                                                                                                                      
- ...                                                                                                                                                                             
#### Actual behaviour:                                                                                                                                                            
                                                                                                                                                       
- ...                                                                                                                                                                             
#### Extra details:                                                                                                                                                               
                                                                                                            
- ...                                                                                                                                                                             
#### Additional logs:                                                                                                                                                             
```                                                                                                                                                                               
debconf: Perl may be unconfigured ("use" not allowed in expression at /usr/share/perl/5.28/Carp.pm line 3, at end of line                                                         
syntax error at /usr/share/perl/5.28/Carp.pm line 3, near "use strict"                                                                                                            
BEGIN not safe after errors--compilation aborted at /usr/share/perl/5.28/Carp.pm line 4.                                                                                          
Compilation failed in require at /usr/lib/arm-linux-gnueabihf/perl/5.28/IO/File.pm line 7.                                                                                        
BEGIN failed--compilation aborted at /usr/lib/arm-linux-gnueabihf/perl/5.28/IO/File.pm line 7.                                                                                    
[... cut ...]
 php7.3-json                                                                                                                                                                      
 php7.3-readline                                                                                                                                                                  
E: Sub-process /usr/bin/dpkg returned an error code (1)                                                                                                                           
```

MichaIng, 2021-07-15:
No errors found, that is strange 🤔. Can you check the following:
```sh
dmesg -l emerg,alert,crit,err
ls -l /lost+found
```

Also, do you have a backup where you can restore those missing list files from? Even when those are older, better to have an outdated file list than none. E.g. this one was expected but not found by DPkg:
```
/var/lib/dpkg/info/php7.3-cgi.list
```
So if you have a backup, you could run this loop to restore those lists:
```sh
#!/bin/bash
for i in /path/to/backup/var/lib/dpkg/info/*.list
do
package=${i##*/}
package=${package%.list}
[[ -f $i && ! -f /var/lib/dpkg/info/${package}.list ]] || continue
echo "$package package list found in backup but not on system"
dpkg-query -s "$package" &> /dev/null || continue
echo "$package also currently installed on system"
mv -nv "$i" "/var/lib/dpkg/info/${package].list"
done
```
Copy paste into console or a file to execute.

tfretze, 2021-07-15:
i don't have a backup of my dietpi to work off unfortunately.

[image]

looks like there are many files in this directory
[image]

Joulinar, 2021-07-15:
in general there is no need to create screen prints. You can simply copy the output from SSH terminal and post it into GitHub.

In general you have an issue with `under voltage`. Means something wrong with power supply or cabling. Pls use original power supply if possible. This perhaps have lead to file system corruptions.

@tfretze 
do you have another Linux box you could use to perform some file system checks on your SD card?

MichaIng, 2021-07-15:
Actually, looking at the stage of the system, it is much work to get it back working, so probably it's better to save data and configs and flash a fresh image, **after voltage issues have been solved, of course**, which are nearly guaranteed to be the reason for the lost files.

As there is no real file system corruption, it seems, the alternative would be open each and every file in `/lost+found`, and if it can be identified by it's content (for the package `.list` files this will be easy, for the missing perl `.pm` files very difficult), move it back to where it belongs to. Some are also whole directories, so there it might be easier to identify.

tfretze, 2021-07-15:
ok i will use the do that from now on, 

I had my raspberry pi plugged directly into the router via usb, with my router powering it so that could be the culprit. 

It was working fine for months up until i attempted this update though which is strange.

i will look at flashing a fresh image. 

thankyou for your time

MichaIng, 2021-07-15:
Always good to do backups on a regular basis, e.g. using `dietpi-backup`. SD cards do also die relatively fast, so similar issues can also happen without voltage issues, and then it's good to have critical files doubled. The Raspberry Pi tries to mitigate voltage issues by itself by throttling down the CPU automatically and restoring frequency when voltage normalises, so often one won't see effects of this for a long time, especially when no external drive is involved, which then may suffer from random disconnects and such. But when doing larger file writes and mixing with CPU and RAM usage, like during larger APT package installs/upgrades, when the voltage drops in an unlucky moment, e.g. the file's metadata or filesystem link can be corrupted/lost.

You can also try to reduce power consumption of the RPi in other ways:
- If you don't use WiFi and Bluetooth, assure that those are disabled firmware-wise: `grep 'dtoverlay=disable-' /boot/config.txt`
- If you don't use an attached screen, enable headless mode: `dietpi-config` > `Display Options` > `Display Resolution` > `Headless`
- And of course also the max CPU and GPU frequencies can be reduced: `arm_freq` and `core_freq` settings in `/boot/config.txt`.

Btw, as you mention `apt-utils` in the heading and see the warning on APT installs/upgrades: This is not an issue but an expected warning on DietPi, unless this package is installed manually. `apt-utils` ships a command line tool `apt-extracttemplates`, which can be used to extract `debconf` templates (definitions for interactive package configuration dialogues, used by some packages) from a package archive. `debconf` has an APT config hook `/etc/apt/apt.conf.d/70debconf`, which tries to extract those templates of all downloaded packages, before those are completely extracted and installed, so that all interactive configuration dialogues can be shown and answered before the actual package install starts. But when `apt-utils` is not installed, hence `apt-extracttemplates` not available, `debconf` cannot do that and hence prints the warning. The interactive dialogues are then instead shown and can be answered when the respective package is extracted and installed.

tfretze, 2021-07-15:
good info on the power reduction steps, i will take that on board on my next install. Thankyou!

MichaIng, 2021-07-28:
I'll mark this issue as closed. Feel free to reopen if required.

closed 2021-07-28 by MichaIng