MichaIng/DietPi #4560: Using BBR anyhow on ComputeModule 1?

danergo, 2021-07-15:
Hi,

I found a very promising article on massively increasing network performance by applying "BBR" as TCP congestion control:

https://djangocas.dev/blog/huge-improve-network-performance-by-change-tcp-congestion-control-to-bbr/

This involves adding two lines to `/etc/sysctl.conf`, one of which is `net.core.default_qdisc = fq`.

However if I do this, and try applying it, I got an error:

`sysctl: setting key "net.core.default_qdisc": No such file or directory`.

Is it anyhow possible to use BBR on DietPi by any chance?

Here is another article:
https://www.cyberciti.biz/cloud-computing/increase-your-linux-server-internet-speed-with-tcp-bbr-congestion-control/

Check the stats at the end ;)

closed 2021-07-15 by danergo

MichaIng, 2021-07-15:
Could you answer it yourself? It depends on the system hardware and kernel, whether this is available or not, so we'd need to check on the specific device, whether it's possible, what the default is, and then run benchmarks.

I do not believe that all kernels are compiled with with a default that is slower and has only downsides compared to BBR, so we'd need to check this carefully 😉.

danergo, 2021-07-16:
I just had an extra characters at the end of "fq", like "fq    # this is explanation". After removing those, problem was solved and everything works as intended.

MichaIng, 2021-07-17:
Okay, great 🙂.