MichaIng/DietPi #4562: myMPD v8.0.0 has no mympd.conf

jcorporation, 2021-07-16:
Today myMPD v8.0.0 was released and there are major changes for configuration. The mympd.conf file is droped in favor of a configuration directory.

https://jcorporation.github.io/myMPD/configuration/

Proposed configuration script:

```
# Adjust config file on fresh install
# - On reinstall /var/lib/mympd exists, so use its existence as reinstall flag
if [[ ! -d '/var/lib/mympd' ]]; then
  # create default myMPD configuration
  export MYMPD_HTTP_PORT=1333
  export MYMPD_SSL=false
  export MYMPD_MPD_HOST=/run/mpd/socket
  mympd -c
fi
```
playlist_directory is configurable in the web gui. If you want to configure it at install time:
- create /var/lib/mympd/state
- write the directory in the file /var/lib/mympd/state/playlist_directory
- directory and file must be writeable by myMPD.

MichaIng, 2021-07-17:
Many thanks for the info. I'll update it for the upcoming release. For migrations of existing configurations: Would it work to move the old config file into `/var/lib/mympd/config/` (respectively `<workdir>/config/`)?

jcorporation, 2021-07-17:
myMPD should migrate valid configuration options at first startup.

MichaIng, 2021-07-17:
Works great 👍. Just to be sure, `/var/lib/mympd/config/` is intended to not be owned/writable by myMPD, while instead `/var/lib/mympd/state/` is used by myMPD to override configs done via web UI, right?

jcorporation, 2021-07-17:
This is right.

Thanks for your fast response and the pull requests!

closed 2021-07-24 by MichaIng