MichaIng/DietPi #4563: LAN9512: TCP/UDP checksum offload support

danergo, 2021-07-16:
Hi!

I'm using an old RPi architecture along with DietPi.

It utilizes a LAN9512 USB2ETH NIC, which offers "TCP/UDP checksum offload support".

I'm wondering if this is enabled by default and I don't really found any information regarding this.

Guys, you might have a better view on this topic.

What I found, is that the kernel sources are (somewhere) here: https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/net/usb/smsc95xx.c

And indeed, in ln#685 (https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/net/usb/smsc95xx.c#L685) they defined a function for enabling/disabling these features:

```
/* Enable or disable Tx & Rx checksum offload engines */
static int smsc95xx_set_features(struct net_device *netdev,
	netdev_features_t features)
```

I'd like to make sure these engines are enabled, and would love to see how can I check it.

MichaIng, 2021-07-17:
Check out the following:
```sh
ethtool -k eth0
```
Or if you have an onboard Ethernet but use an external Ethernet adapter, it's most likely `eth1`:
```sh
ethtool -k eth0
```

E.g. on RPi 2 the first output lines show:
```
rx-checksumming: on
tx-checksumming: on
        tx-checksum-ipv4: on
        tx-checksum-ip-generic: off [fixed]
        tx-checksum-ipv6: off [fixed]
        tx-checksum-fcoe-crc: off [fixed]
        tx-checksum-sctp: off [fixed]
```
Which means that checksum offloading is enabled, where supported, obviously for outgoing transfer and IPv6 it is not supported (`off` and `[fixed]`).

danergo, 2021-07-17:
Thx, I've checked these, and all values are the same as yours. :)

closed 2021-07-17 by danergo