MichaIng/DietPi #4564: PINE A64 - A dietpi-update throws missing stretch release file error

twikedk, 2021-07-17:
#### Details:
- Date           | Sat Jul 17 10:26:00 CEST 2021
- Bug report     | 7c2dbcc2-716f-4323-aa34-fde2ac06226f
- DietPi version | v7.3.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | ARMbian
- Hardware       | PINE A64 (aarch64) (ID=40)
- Kernel version | `Linux development 4.19.63-sunxi64 #5.92 SMP Fri Aug 2 00:18:27 CEST 2019 aarch64 GNU/Linux`
- Distro         | stretch (ID=4)
- Command        | `apt-get -q update`
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. run: dietpi-update

#### Expected behaviour:

- system should read package lists to see if there are any updates

#### Actual behaviour:

- following error get thrown

#### Extra details:

- if I check http://apt.armbian.com is is true there are no stretch Release :-)

#### Additional logs:
```

Hit:2 https://packages.sury.org/php stretch InRelease
Ign:3 https://deb.debian.org/debian stretch InRelease
Get:4 https://deb.debian.org/debian stretch-updates InRelease [93.6 kB]
Get:5 https://deb.debian.org/debian-security stretch/updates InRelease [53.0 kB]
Get:6 https://deb.debian.org/debian stretch-backports InRelease [91.8 kB]
Hit:7 https://deb.debian.org/debian stretch Release
Ign:1 https://armbian.systemonachip.net/apt stretch InRelease
Get:9 https://deb.debian.org/debian-security stretch/updates/main arm64 Packages [678 kB]
Err:8 https://mirrors.dotsrc.org/armbian-apt stretch Release
  404  Not Found
Get:10 https://deb.debian.org/debian-security stretch/updates/main armhf Packages [678 kB]
Reading package lists...
E: The repository 'http://apt.armbian.com stretch Release' does no longer have a Release file.
```

MichaIng, 2021-07-17:
Many thanks for your report.

Armbian dropped support for Debian Stretch already. I'm not sure how to handle best. The kernel, dtb and bootloader packages from the Buster repository should work fine on Stretch as well, but everything else may cause conflicts with system libraries.

Quick workaround to allow the update going through:
```sh
mv -v /etc/apt/sources.list.d/armbian.list{,_bak}
```

I'll come back to you in a few hours with a configuration to allow pulling kernel/dtb/bootloader upgrades from the Buster repository and as alternative Buster upgrade instructions.

MichaIng, 2021-07-17:
Okay, here is a better solution:
```sh
[[ -f '/etc/apt/sources.list.d/armbian.list_bak' ]] && mv -v /etc/apt/sources.list.d/armbian.list{_bak,}
echo 'deb https://apt.armbian.com/ buster main' > /etc/apt/sources.list.d/armbian.list
echo -e '# Only allow install of kernel, device tree, bootloader and firmware packages from Armbian Buster suite on Debian Stretch systems:
Package: linux-image-* linux-headers-* linux-dtb-* linux-u-boot-* armbian-firmware\nPin: origin apt.armbian.com\nPin-Priority: 500
\nPackage: *\nPin: origin apt.armbian.com\nPin-Priority: -1' > /etc/apt/preferences.d/dietpi-armbian
```
This allows only a defined set of compatible and essential packages to be installed from the Buster suite of Armbian's APT repository, while everything else will be blocked to avoid conflicts and broken dependencies.

WR1171, 2021-07-20:
Thanks @MichaIng Almost created another thread until I searched 404 and found this.  Thanks!

closed 2021-07-24 by MichaIng