MichaIng/DietPi #4568: Kernel panic - not syncing: VFS: Unable to mount fs on unknown-block(8,1)

carrotice, 2021-07-18:
Often after a reboot DietPi (with the command `reboot`) or after a proper shtudown of the DietPi system (with the command `shutdown -h now`) an error occurs when the system starts:
`---[ end Kernel panic - not syncing: VFS: Unable to mount fs on unknown-block(8,1) ]---`
see picture:
[image]

At first thought that this causes execution of upgrade by command `apt update` or `apt upgrade`. However this is happened also at time, when upgrades of system wasn't executing. Just a normal system reboot or normal shutdown and its startup. Like in my case.

For booting DietPi I have a SanDisk Ultra 32 GB Micro SD memory card. I have RootFS on an external USB hard drive Verbatim 2 TB.

To repair a "damaged" file system, I need to run the Linux operating, such as Ubuntu 20.04.1 LTS, on another computer and connect a "damaged" external USB hard drive to it.

Then I need to run command `sudo fsck.ext4 -fty -C 0 /dev/sdb1` in the terminal window to repair the file system.

It is very annoying and incorrect that almost every reboot of a DietPi system requires a "repair" of the file system.

### Required Information
- DietPi version
	G_DIETPI_VERSION_CORE=7
	G_DIETPI_VERSION_SUB=1
	G_DIETPI_VERSION_RC=2
	G_GITBRANCH='master'
	G_GITOWNER='MichaIng'
- Distro version
	buster (10.9)
- Kernel version
	Linux DietPi 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux
- SBC model
	RPi 4 Model B (armv7l)
- Memory
	4 GB
- Power supply used
	Official Raspberry Pi 4 USB-C 15.3W power supply, 5.1V, 3.0A (KSA-15E-051300-HX, SC0213)
- SDcard used
	SanDisk Ultra 32 GB Micro SD
- External HDD
	External USB HDD Verbatim 2 TB
- Case
	Argon One for Raspberry Pi 4

### Additional Information (if applicable)
- Was the software installed freshly or updated/migrated?
Previously the DietPi system was partially upgraded several times using the `apt update` and `apt upgrade` command from version 6.27 to the current version 7.1.2.
I will not currently upgrade to the latest possible version (currently 7.3.2.) until DietPi starts normally.

- Bug report ID
	78598894-8cf3-453f-8faf-35cfdcb6633f
- DietPi version
	[image]
- DietPi CPU info
	[image]
- DietPi Services
	[image]
- DietPi Drive Manager
	[image]
- DietPi Drive Manager - External USB HDD
	[image]

### Steps to reproduce
Reboot DietPi system with command `sudo reboot` or proper shutdown DietPi system with command `shutdown -h now` and then power on Raspberry Pi with power button.
I have Argon One case for Raspberry Pi 4.

### Expected behaviour
Normal boot from Micro SD card, mount and run file system from external USB HDD.

### Actual behaviour
Not always but often after I (re)boot DietPi system I get an error
`---[ end Kernel panic - not syncing: VFS: Unable to mount fs on unknown-block(8,1) ]---`

### Extra details
In this case I need to remove the external hard drive and connect it to the computer running Ubuntu linux operating system and run the following commands `sudo lsblk` to check name for external HDD and `sudo fsck.ext4 -fty -C 0 /dev/sdb1` to repair external USB HDD.

Joulinar, 2021-07-18:
Hi,

This looks more like a dieing SD card and has no relation to DietPi. But I guess @MichaIng will explain it more in detail.

MichaIng, 2021-07-18:
@Joulinar 
It's not the SD card that gets corrupted, but the external HDD which holds the root filesystem.

@carrotice 
Many thanks for your report.

I had a look into your bug report:
- One thing I recognise (also in your screenshot above) is that your root filesystem is nearly full. Okay relative, as 44 GiB are still free, but depending on how write-intense your services are, 98% disk usage is probably not a stable long-term state.
- A minor enhancement you may want to apply is to switch to the modern entropy daemon: `apt install rng-tools5`
- Aria2 fails to start, as the log file is missing:
    ```sh
    > /var/log/aria2.log
    chown aria2: /var/log/aria2.log
    systemctl restart aria2
    ```
- Unbound is failing as well. Can you check: `journalctl -u unbound`
- And the Argon fan service fails. Check via: `journalctl -u argononed`
- You have DietPi-RAMlog enable but rsyslog installed and running as well. That does not make much sense. If you require persistent system logs, you need to disable/uninstall DietPi-RAMlog (`dietpi-software uninstall 103; reboot`), if logs for the current boot session are sufficient, uninstall rsyslog (`dietpi-software uninstall 102`) and use `journalctl` to view system logs, hold in memory.
- You may also remove the obsolete elevator entry from your cmdline.txt: `sed -i 's/elevator=deadline //' /boot/cmdline.txt`
- And there is still an obsolete CloudPrint service file installed: `apt purge cloudprint-service cloudprint`
- Ah and the install state of the DHCP client (you don't use it, I know) is not as it should be: `apt install isc-dhcp-client`

But I couldn't see an issue/error that would explain kernel panics at boot. How is the drive powered? Via dedicated PSU or via USB only? Could you also run and paste the output of the following command:
```sh
vcgencmd get_throttled
```
Also checking the above listed issues/steps might give a hint, when the issue and the boot problems had the same root cause.

Joulinar, 2021-07-18:
@MichaIng damn it. Mixed the devices

carrotice, 2021-07-27:
I stopped DietPi with `shutdown` command and disconnected the power supply.
A few days later, I startet DietPi and I had a Kernel panic error again. See picture:

[image]

I removed the external hard drive and connect it to the computer running Ubuntu linux operating system and run the
following commands `sudo lsblk` to check name for external HDD and `sudo fsck.ext4 -fty -C 0 /dev/sdb1` to repair external USB HDD.

I connected the external hard drive to the DietPi and booted the system.
A new error occurred on startup. `You are in emergency mode. After logging in, type "journalctl -xb" to view system logs ...`. See picture:

[image]

After entering the root password, I got the following error:
`[FAILED] DietPi-Login | Failed to load DietPi-Globals. Skiping DietPi login scripts...`

[image]

After entering the `systemctl reboot`, the DietPi system could normaly start. See picture:

[image]

[image]

If you needed I can forward the screenshots of journalctl -xb log.
I can also recreate the bug report ID.

Joulinar, 2021-07-27:
Looks like the disk was not mounted correctly. Can you check kernel log?
`dmesg -l err,crit,alert,emerg`

If possible try to connect via SSH and copy the output from SSH terminal instead of creating screen prints.

Did a reboot is working without issues now?

MichaIng, 2021-09-01:
I'll mark this as closed. Feel free to reopen if the issue persists.

closed 2021-09-01 by MichaIng

carrotice, 2021-09-26:
Some detail for post on 27.7.2021:
Connection via SSH is possible only from last two picture, when DietPi is started correctly and SSH service is running.

Yesterday I noticed again that DietPi stopped working, SSH access was also not possible. When I connected the monitor to the HDMI connector I noticed that something was wrong with the file system. More details in the picture below:
[image]
After turning off DietPi (with the button on the ArgonOne case) and turning it on again, I got a file system error message again, as shown in the image below:
[image]
I had to fix the file system again on another computer with Ubuntu operating system with the command `sudo fsck.ext4 -fty -C 0 / dev / sdb1` After turning on DietPi, I received the following message:
[image]
When I interrupted the "emergency mode" by pressing `Control-D`, the DietPi started and I was also able to access it from the SSH console.
Kernel log after running command `dmesg -l err, crit, alert, emerg` shows nothing. However, if I run the `dmesg -l warn` command, it shows me the following:
```
root@DietPi:~# dmesg -l warn
[    0.086682] usb_phy_generic phy: supply vcc not found, using dummy regulator
[  116.283285] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[  116.605753] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
root@DietPi:~#
```
I currently still have DietPi version 7.1.2 and have not yet upgraded to the latest version 7.6.2.

MichaIng, 2021-09-26:
Please check for all kernel errors, not just warnings:
```sh
dmesg -l emerg,alert,crit,err
```
Sadly the `-l` option is not "show all logs with severity above" but "show all logs with severity equal" :wink:.

carrotice, 2021-09-27:
I ran the commands again and the results showed nothing wrong. See the results below:
```
root@DietPi:~# dmesg -l emerg,alert,crit,err
root@DietPi:~# dmesg -l emerg
root@DietPi:~# dmesg -l alert
root@DietPi:~# dmesg -l crit
root@DietPi:~# dmesg -l err
root@DietPi:~#
```

MichaIng, 2021-09-28:
Okay, if it still boots into emergency mode, then obviously something on the boot partition is still broken. You repaired only the ext4 partition (?) so you may try to do the same for the FAT partition.

Alternatively you can update to latest DietPi which includes APT package upgrades and hence a kernel/dtb/bootloader upgrade. The latter may include fixing whatever was broken on the FAT partition.

carrotice, 2021-12-08:
When I run above command I receive next errors:
```
root@DietPi:~# apt install raspberrypi-kernel raspberrypi-bootloader libraspberrypi0 libraspberrypi-bin
Branje seznama paketov ... Narejeno
Gradnja drevesa odvisnosti
Branje podatkov o stanju ... Narejeno
Naslednji paketi bodo nadgrajeni:
  libraspberrypi-bin libraspberrypi0 raspberrypi-bootloader raspberrypi-kernel
4 nadgrajenih, 0 na novo nameščenih, 0 bo odstranjenih in 91 ne nadgrajenih.
Potrebno je dobiti 0 B/85,5 MB arhivov.
Po tem opravilu bo porabljenega 3770 kB dodatnega prostora.
dpkg: warning: files list file for package 'fake-hwclock' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'mysql-common' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'fontconfig' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'deluge-console' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'fbset' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'deluge-common' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'mime-support' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'dosfstools' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'gettext-base' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'debhelper' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'diffutils' missing; assuming package has no files currently installed
dpkg: unrecoverable fatal error, aborting:
 loading files list file for package 'libcairo2:armhf': cannot open /var/lib/dpkg/info/libcairo2:armhf.list (Bad message)
E: Sub-process /usr/bin/dpkg returned an error code (2)
root@DietPi:~#
```

carrotice, 2021-12-08:
I tried to update DietPi from version v7.1.2 to v7.8.2 but the upgrade failed:
`root@DietPi:~# dietpi-update

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[ INFO ] DietPi-Update | Getting repository version: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid repository version
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v7.1.2
[ INFO ] DietPi-Update | Latest version  : v7.8.2

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for update pre-requirements

[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=17181 MiB | required=100 MiB
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : docker
[  OK  ] DietPi-Services | stop : cups
[  OK  ] DietPi-Services | stop : syncthing
[  OK  ] DietPi-Services | stop : deluge-web
[  OK  ] DietPi-Services | stop : plexmediaserver
[  OK  ] DietPi-Services | stop : deluged
[  OK  ] DietPi-Services | stop : lighttpd
[  OK  ] DietPi-Services | stop : php7.3-fpm
[  OK  ] DietPi-Services | stop : smbd
[  OK  ] DietPi-Services | stop : nmbd
[  OK  ] DietPi-Services | stop : avahi-daemon

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying pre-patches

[  OK  ] DietPi-Update | Downloading pre-patches
[  OK  ] DietPi-Update | Applying execute permission
[ INFO ] DietPi-Pre-patches | Connecting to Mosquitto and Webmin repositories via plain HTTP once until latest OpenSSL has been installed: https://github.com/MichaIng/DietPi/issues/4795
[  OK  ] DietPi-Pre-patches | sed -Ei s#https://(repo.mosquitto.org|download.webmin.com)#http://\1# /etc/apt/sources.list
[  OK  ] DietPi-Update | Successfully applied pre-patches

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 https://download.docker.com/linux/raspbian buster InRelease [33.6 kB]
Get:3 https://downloads.plex.tv/repo/deb public InRelease [6685 B]
Get:4 https://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:5 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:6 https://downloads.plex.tv/repo/deb public/main armhf Packages [428 B]
Get:7 https://download.docker.com/linux/raspbian buster/stable armhf Packages [15.8 kB]
Get:8 https://archive.raspberrypi.org/debian buster/main armhf Packages [393 kB]
Get:9 http://raspbian.raspberrypi.org/raspbian buster/non-free armhf Packages [104 kB]
Get:10 http://raspbian.raspberrypi.org/raspbian buster/contrib armhf Packages [58.8 kB]
Fetched 13.7 MB in 32s (423 kB/s)
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
Preconfiguring packages ...
dpkg: warning: files list file for package 'fake-hwclock' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'mysql-common' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'fontconfig' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'deluge-console' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'fbset' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'deluge-common' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'mime-support' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'dosfstools' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'gettext-base' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'debhelper' missing; assuming package has no files currently installed
dpkg: warning: files list file for package 'diffutils' missing; assuming package has no files currently installed
dpkg: unrecoverable fatal error, aborting:
 loading files list file for package 'libcairo2:armhf': cannot open /var/lib/dpkg/info/libcairo2:armhf.list (Bad message)
E: Sub-process /usr/bin/dpkg returned an error code (2)
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -qq upgrade
[ INFO ] DietPi-Update | APT upgrade, please wait...
root@DietPi:~#`

MichaIng, 2021-12-08:
It looks like several essential files are missing. Can you have a look into:
```sh
ls -l /lost+found
```
But this is extremely difficult/much work to recover and no guarantee that it succeeds at all. I suggest you recover a backup, at best offline/externally, or externally extract all important data and setup a new image. When using the same SD card, keep having an eye on `dmesg -l emerg,alert,crit,err` to detect I/O error early. As we are not sure about the origin of the data loss, it could be that the SD card simply has bad block (physically broken), also insufficient PSU or a power loss in the middle of vast file writes.