MichaIng/DietPi #4569: RasPi Zero DietPi Update from 7.3.2 to 7.4 not possible

omavoss, 2021-07-18:
Hi,

DietPi v7.3.2 : 22:50 - So 18.07.2021
 ─────────────────────────────────────────────────────
 - Device model : RPi Zero W (armv6l)
 - CPU temp : 42'C : 107'F (Optimal temperature)
 - LAN IP : 192.168.115.98 (wlan0)
 - MOTD : DietPi v7.4 beta is ready for testing:
          https://github.com/MichaIng/DietPi/issues/4567
 ─────────────────────────────────────────────────────

 ...
...
root@DietPi:~# dietpi-update

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[ INFO ] DietPi-Update | Getting repository version: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[  OK  ] DietPi-Update | Got valid repository version
[  OK  ] DietPi-Update | No update required, your DietPi installation is already up to date:
[ INFO ] DietPi-Update | Current version : v7.3.2
[ INFO ] DietPi-Update | Latest version  : v7.3.2
[ INFO ] DietPi-Update | APT update, please wait...
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | No APT upgrades were found, not creating file: /run/dietpi/.apt_updates
root@DietPi:~#

Please give me any hints for fixed this, thank you.

MichaIng, 2021-07-18:
Many thanks for your report.

Note that DietPi v7.4 is still a beta, as noted in the MOTD, so it won't be offered unless you explicitly apply to the beta branch. Check out the link the comes along with the beta info to find all info and how to apply to the beta:

omavoss, 2021-07-18:
DietPi Update on Raspi1, Raspi2, Raspi3 and Raspi4 (64bit) was succesful. Why not RaspiZero?

omavoss, 2021-07-18:
Uh, I find my error; I must changed into the beta-channel; please delete this thread, thank you.

MichaIng, 2021-07-18:
No problem.

To apply to the beta:
```sh
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=beta' /boot/dietpi.txt
```

To switch back to stable releases:
```sh
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /boot/dietpi.txt
```

closed 2021-07-18 by MichaIng