MichaIng/DietPi #4571: Secured dietpi installation on Raspbery Pi 4B

andreicaba, 2021-07-20:
Hello

This might look more like a feature request than an issue.
After carefully checking many raspberry pi distributions, your DietPi seems to have the best lightweight/reliable ratio.
We are working on a project that requires the raspberry pi device to be left unattended in public places, running our sensitive, proprietary software. The software is protected inside an encrypted .img file (using luks).
The .img file is decrypted only if the integrity of important OS files (/boot, /etc, /usr, /root) is intact.
As we found out that DietPi has a lot of "doors" into the OS [for example, someone could just plug the microsd card into the PC and he could just type in some commands in the /boot/dietpi/postboot file (or in other files that are executed inside the OS) and he can get even root access to the running OS through it].
We would like to know all the possible "backdoors" into the OS - to protect the raspberry pi properly.

best regards,
A. Caba

Joulinar, 2021-07-20:
Hi,

Just for clarification, DietPi is not an own OS. It is a set of script on top of a base image. For Raspberry Pi device we use Raspberry OS. Means it's basically a Raspberry OS with reduced amount of packages (diet) and some automation scripts on top. Whatever is possible on Raspberry OS will be possible on DietPi as well. This include the scenario you describe.

MichaIng, 2021-07-20:
We collected some general security recommendations here: https://github.com/MichaIng/DietPi/wiki/Security-recommendation
Those apply to any OS, so do your concerns.

If you want to protect your system against physical access, then indeed disk/image encryption is the only way. You should be able to set this up with DietPi just like with any other Debian- or Ubuntu-based distribution, just note that `dietpi-drive_manager` does currently not detect encrypted drives properly, so I'd simply avoid using it, but setting up `/etc/fstab` manually, as you'd do on any other Linux distro. There is a request open to add support for it, any contribution to start with this is highly appreciated: #3377

`/boot/dietpi/postboot` is invoked via systemd, like many other DietPi-independent system services and scripts, and like init scripts/services are called on any Linux distro. This file is only writable by the root user, and if one has root access to any Linux distro, one can install any init service/script, so DietPi does not add any "door" here. Actually, as DietPi comes with a lot less packages pre-installed and a lot services disabled by default, there are fewer attack vectors of this kind. With root access, it however doesn't matter at all, as stated, so this is what should be prevented, from physical disk access via disk encryption and in other ways by e.g. disabling the root account for direct login (`usermod -L root`), limiting `dietpi` users sudo permissions (`/etc/sudoers.d/dietpi`), or locking that one as well and creating own users with strict sudo permissions. And otherwise following the security recommendations form the wiki, e.g. to harden SSH via key authentication, if required at all.

andreicaba, 2021-07-21:
yes, thanks for the quick answer :) That was how I viewed DietPi so far - and, by the way, congrats for it: great work!... :)
It seems that the Raspbian OS was kinda covered (we're still researching/testing/etc this part) by fingerprinting everything we thought would be important. We were asking about the DietPi part.

MichaIng, 2021-11-07:
I'll mark this as closed. Feel free to reopen when required.

closed 2021-11-07 by MichaIng