MichaIng/DietPi #4573: OrangePi Zero not booting anymore after converting to DietPi

timocapa, 2021-07-22:
Hello, I've been trying to convert an Armbian (Debian) image to DietP on an OrangePi Zero, however post reboot the Pi doesn't come online anymore.

The PREP script runs without any errors and this occurs on every dp (stable / beta / dev) and Debian (buster / bullseye) version.

I've had DietPi running on it without any issues prior for a few years. After updating it it stopped booting a few days ago.

The Pi is connected via Ethernet.

I don't think it's a SD Card failure - Armbian flashes and runs without issues

Any ideas?

elchmartin, 2021-07-22:
I've had something like this before ... at least I think so.
my problem back then was the oPi not correctly setting up the IP adress after installing dietpi.
from what i remember, the oPi-zero doesnt have HDMI, so you cannot really check what it is doing during startup/boot...
did you use a fixed IP or a DHCP-assigned one? if you used a fixed IP, try with DHCP and the other way around.
a tool like fing will easily tell you if the oPi got an IP from your router...

another Idea: try to start with "real" [debian for the oPi-zero](http://www.orangepi.org/downloadresources/OrangePiZero2/2020-11-04/orangepi3_8849be56b419022d3a1780ac6d.html) instead of armbian. if you succeed, then armbian is the issue, whysoever.

timocapa, 2021-07-22:
> did you use a fixed IP or a DHCP-assigned one

DHCP but it gets a fixed IP from the router (so DHCP will always respond with 192.168.0.4, in my case)

> another Idea: try to start with "real" debian for the oPi-zero instead of armbian. 

I'll give that a try once I get up, although

> then armbian is the issue, whysoever.

I don't see how that could be a thing, yet

MichaIng, 2021-07-22:
Armbian should work just fine.

How did you check whether it's not coming online? Did you try to ping it or check whether the router/DHCP server sees it coming online and e.g. requesting it's IP? Do you have a UART cable to check early boot output via debug port from an external system?

Trying a static IP via `dietpi.txt` (right after running PREP) would be an idea indeed, just to rule out any issues related to DHCP, e.g. a changing MAC address which invalidates the reserved IP in the DHCP server. Many SBCs have no fixed MAC addresses but those rotating on each boot.

timocapa, 2021-07-22:
> How did you check whether it's not coming online? Did you try to ping it or check whether the router/DHCP server sees it coming online and e.g. requesting it's IP?

Both pinging and checking the OpenWrt log and lease, which doesn't renew.

> Many SBCs have no fixed MAC addresses but those rotating on each boot.

Even if that was the case I would see it reappearing under a new IP, which isn't the case. With that said it has never changed its MAC

> Do you have a UART cable to check early boot output via debug port from an external system?

Unfortunately not - I had ordered one from amazon a while ago but they sent me the wrong thing and I've not bothered since 

> Trying a static IP via dietpi.txt (right after running PREP) would be an idea indeed

Anything special I'd need to enter? Haven't edited it before as I've always managed IPs through the router

MichaIng, 2021-07-22:
E.g. this block:
```
# Enter your static network details below, if applicable.
AUTO_SETUP_NET_USESTATIC=1
AUTO_SETUP_NET_STATIC_IP=192.168.0.4
AUTO_SETUP_NET_STATIC_MASK=255.255.255.0
AUTO_SETUP_NET_STATIC_GATEWAY=192.168.0.1
AUTO_SETUP_NET_STATIC_DNS=9.9.9.9 149.112.112.112
```

You used the Armbian Buster image from Jun 21 2021 from [here](https://www.armbian.com/orange-pi-zero/), right?

timocapa, 2021-07-22:
> You used the Armbian Buster image from Jun 21 2021 from here , right?

Correct, I've also tried an older custom image I've made with F2FS as rootfs, which prior worked fine converting to DietPi. Also fails now.

timocapa, 2021-07-22:
Install log: https://hastebin.com/wazunitiwu.properties (this had a temporary pihole install running)

Set the block you gave me and double checked it's fine for my current network and it still didn't come up again

For the install log, I've had armbian's beta repo enabled for a newer kernel, however that doesn't make a difference, already tried

MichaIng, 2021-07-22:
There was a webserver and PHP installed as well, so that image was not a clean Armbian server image? However, that shouldn't be an issue and indeed the whole PREP run looks pretty fine.

How long did you wait for the system to come up? As it's an H2, it may take a while for e.g. [these](https://github.com/MichaIng/DietPi/blob/master/rootfs/var/lib/dietpi/services/dietpi-firstboot.bash) steps to finish. Also is there any LED indicating CPU or I/O activity?

Another idea: I know from my old notebook that some systems fail to produce enough entropy for systemd to start. There is haveged as entropy daemon installed, but my notebook doesn't even reach a state where this starts and hangs a long time, or until I hit a few keys on the keyboard (filling the entropy pool). So just to rule it out, when the system does not come up for a long time (give it 10 minutes or so) and does not show any activity either, probably having a keyboard via USB attached and hitting some random keys helps. As Armbian additional has `rng-tools` installed for using the hardware random generator, if the board + kernel supports this, and that daemon would start earlier than haveged, it could explain that Armbian boots. But this is highly speculative and so far my (>12 years old!) notebook has been the only case.

timocapa, 2021-07-22:
> There was a webserver and PHP installed as well, so that image was not a clean Armbian server image?

"this had a temporary pihole install running"

> How long did you wait for the system to come up?

40 minutes later and it's still dead

> Also is there any LED indicating CPU or I/O activity?

it has two leds that can be configured, but by default only one is enabled and just shows if it's on or off

> probably having a keyboard via USB attached and hitting some random keys helps

Plausible.

[image]

MichaIng, 2021-07-22:
Wow, at least we found a solution.

I'm not exactly sure when the need for an entropy daemon appeared, but I think it's since Debian Buster as first reports arrived in first half of 2015 on RPi, which switched to Buster earlier as of Raspberry Pi 4 release: https://github.com/MichaIng/DietPi/issues/2760
If I remember right involved are at least parts of systemd, but the kernel itself might be a reason as well. In the meantime, all distros come with entropy daemons.

So now an interesting question is whether there is a hardware random generator available or not:
```sh
ls -l /dev/hwrng
```

I remember now that Armbian by default configures `rng-tools` to feed `/dev/urandom` into `/dev/random`, which means that available hardware random generators are not used and the whole purpose of separate blocking and non-blocking random sources is broken, degrading the security of `/dev/random` when the pool is empty. So that's a bad default behaviour, but a functional workaround indeed. `haveged.service` however should work alone as well, being a software-based random generator, WHEN it starts early enough. So would be also interesting to see boot logs and when haveged was started:
```sh
journalctl
```

timocapa, 2021-07-22:
Currently on armbian again to convert it to DietPi again to verify things, so this comes from armbian. I'll make a new post once I'm on DietPi again.

> So now an interesting question is whether there is a hardware random generator available or not

```
root@orangepizero:~# ls -l /dev/hwrng 
crw------- 1 root root 10, 183 May 26 03:43 /dev/hwrng
```
> So would be also interesting to see boot logs and when haveged was started

Apparently armbian has haveged as well.

About 4 seconds into the (first)boot
[journal.txt](https://github.com/MichaIng/DietPi/files/6862764/journal.txt)

MichaIng, 2021-07-22:
Yes Armbian has both installed while `rng-tools` should be not required then and the way it is configured on Armbian implies a theoretical security issue and breaks the intention of `/dev/random`. But interesting is hat I cannot see it starting at all in your journal output. Probably it's not enabled? `systemctl status rng-tools`

But since a hardware random generator device is present, let's see if that is usable:
```sh
apt install rng-tools5
sleep 2
journalctl -u rngd
```

> e: ot: why does `mount -o remount /tmp` tend to fail so often in the script?

Where? In the hastebin you linked above I only see it succeeding once.

timocapa, 2021-07-22:
> systemctl status rng-tools

```
root@orangepizero:~# systemctl status rng-tools
Unit rng-tools.service could not be found.
```
I believe it's missing as I currently use a custom minimal armbian build, sorry. rng-tools seem present on a stock armbian build

> let's see if that is usable

Nope
```
root@orangepizero:~# journalctl -u rngd
-- Logs begin at Wed 2021-05-26 03:44:05 UTC, end at Thu 2021-07-22 15:03:51 UTC. --
Jul 22 15:03:51 orangepizero systemd[1]: Started Start entropy gathering daemon (rngd).
Jul 22 15:03:51 orangepizero rngd[2572]: read error
Jul 22 15:03:51 orangepizero rngd[2572]: read error
Jul 22 15:03:51 orangepizero rngd[2572]: No entropy sources working, exiting rngd
Jul 22 15:03:51 orangepizero systemd[1]: rngd.service: Succeeded.
```

> Where? In the hastebin you linked above I only see it succeeding once.

It fails every other DietPi install for me. The hastebin one was fine, but when I redid it just now the command failed and I replaced it with a dummy command to pass it. The same as others that you find if you look for the error in issues :)

MichaIng, 2021-07-22:
I think the error happens when Armbian's zram drive is still mounted at `/tmp`, while we already adjusted `/etc/fstab` to contain a `tmpfs` entry. The remount is done to adjust the `/tmp` tmpfs size after the swap file size has been adjusted, but it is only possible (it seems) to remount when the filesystem is the same, as content etc is kept. Currently the remount is done whenever /tmp is a mount point already (so it's skipped e.g. on default Debian, where it is not a mount point at all), but we need to also skip it when it's no tmpfs yet.

Strange is that when I try to remount a tmpfs with a completely changed fstab entry, it works pretty fine, the current options are simply kept unchanged.

timocapa, 2021-07-22:
Cool, glad to see it's (hopefully) fixed.

My Pi seems to manage to boot without a keyboard again as well, runs 5.13.4-sunxi and DietPi on Beta just fine. With PiHole running, entropy hovers around 2700-2900, so I'll close this unless it happens again. Still kinda confused about it, especially about why it's suddenly working fine again.

Schönen Tag noch :)

MichaIng, 2021-07-22:
I do not yet understand the doubled fstab entry. We reset it completely during the DietPi-PREP run and `dietpi-set_swapfile` (afterwards) only changes an existing entry but does not add one: https://github.com/MichaIng/DietPi/blob/bc0387f/dietpi/func/dietpi-set_swapfile#L71

This was after DietPi-PREP finished?

Strange with the entropy indeed, especially when that Armbian image came as well with haveged only. Please report back it happens again. It's quite a big problem when an image does not boot at all because of this, even with an entropy daemon installed exactly to address it. Probably we can tweak haveged to start at an earlier boot stage or so.

timocapa, 2021-07-22:
> This was after DietPi-PREP finished?

A bit before it prompted me to reboot, so I don't think PREP finished

> especially when that Armbian image came as well with haveged only.

I've mentioned that the lack of rng-tools seems to be from me running a more minimal image at that moment - the official Armbian image seems to have it

https://github.com/MichaIng/DietPi/issues/4573#issuecomment-884993352

But yeah if both systems just run haveged there's not much of a reason why DietPi wouldn't boot

timocapa, 2021-08-08:
We back :sunglasses: 

Wanted to move to Bullseye due to it being in a hard freeze now

DietPi Buster -> DietPi Bullseye and dead
Made a minimal armbian bullseye (w/ linux 5.13), booted - after running (and finishing) prep, dead

Joulinar, 2021-08-08:
How to upgrade Armbian first before installing DietPi on top?

timocapa, 2021-08-08:
not sure what you mean - the device was up to date in all cases

elchmartin, 2021-08-08:
not sure about the zero, mine is sleeping tightly.
but i remember it has sth like 256mb RAM ... could this be a problem? maybe check before/while updating ... (htop?)

timocapa, 2021-08-08:
It has 2 variants, one with 256 and 512 MB, and a running DietPi instance with PiHole sits at around 90 MB used. I've the 512 MB one.

elchmartin, 2021-08-08:
OK so that doesnt seem to be the issue ... if I find the time, amybe i can try to do the same as you, maybe yours is 'just broken' ;-)

elchmartin, 2021-08-09:
so I dug up my opi0 with 256RAM last night.
installing buster and updating everything went well. so far so good, but still on buster.

prep with dietpi ran supposedly fine until reboot. no IP, nothing afterwards.
i think dietpi removes the sunxi-kernel and the poor little h2+ needs that.
how to fix? add support for h2+ ... I did not try selecting h3 as target machine, but im not going tfor that now.
btw tried the experimental hirsuite image and it appears to run just fine ...

timocapa, 2021-08-09:
DietPi reinstalls the kernel however so it's present. If I chroot into the sd card (requires some setting up, see binfmt and qemu), i think everything is present

But good to know it's not just me

MichaIng, 2021-08-09:
The installed kernel package should be detected automatically and kept installed, yes. The Ubuntu image cannot be converted as part of the process is applying the Debian repository to `sources.list` and then you'll run into package upgrade failures due to version conflicts etc. Ah, actually it will fail right at the start when trying to detect the Debian version 😉.

Whether you choose Generic Allwinner H3 or Generic Device during PREP btw makes no difference at that stage. It is only during a few software installs and config options where this is taken into account. Generally every board is "supported" (means it should boot) as long as a functional kernel, bootloader and in case device tree is installed, for network additionally the firmware needs to be present (which usually is as we pre-install a bunch of common generic firmware packages for WiFi and Ethernet chips and dongles). Every Armbian image "should" work, as they have all those packages installed, which are detected and kept by PREP, including the `firmware-armbian` package which should be guaranteed to hold firmware for all onboard network adapters of boards they support.

Hard to say where it fails without any screen output, e.g. whether its at kernel level (shouldn't be, as the kernel package is only upgraded but stays the same from Armbian) or something during [dietpi-firstboot](https://github.com/MichaIng/DietPi/blob/dev/rootfs/var/lib/dietpi/services/dietpi-firstboot.bash), or bringing up the network only. E.g. I'm currently testing a Raspbian Bullseye build on Raspberry Pi Zero, which seems to hang at `systemd-machine-id-setup` during first boot: https://github.com/MichaIng/DietPi/blob/dev/rootfs/var/lib/dietpi/services/dietpi-firstboot.bash#L218
And I have no idea why, yet.

timocapa, 2021-08-09:
Should there be any logs if I was to chroot into the Zero? I can do that, at least.

MichaIng, 2021-08-09:
You can use the `dev` branch version of the PREP script where this typo is solved. But it does not break anything. When chrooting into the Zero, please have a look at logs in `/var/tmp/dietpi/logs/` which contains a log for initial partition + filesystem expansion, and the output of the above linked `dietpi-firstboot` script, if it was actually started. Good idea to check for those.

timocapa, 2021-08-09:
> /var/tmp/dietpi/logs/

Wouldn't tmp imply they're gone once shut off?

MichaIng, 2021-08-09:
Nope, while `/tmp` is a tmpfs, `/var/tmp` is boot-persistent.

timocapa, 2021-08-09:
Okay, reran the prep script from a running (and working) buster dietpi install for bullseye, we'll see how that goes

timocapa, 2021-08-09:
Assuming this is the last boot attempt, which seems about right time wise (for some reason set to EU/London instead of Berlin), it stalled after dropbear key generation

https://hastebin.com/efayegolac.csharp

I'll remove the file and let the Pi attempt to boot again to be sure I think?

MichaIng, 2021-08-09:
Found the issue. It's not the machine ID but the Dropbear configuration. Since some recent systemd update, I guess, `invoke-rc.d`, called during Dropbear reconfiguration, hangs forever when called to start/restart the Dropbear service. Not sure if there is some logic behind it, e.g. that it actually is usually configured to start only after `dietpi-boot.service` has finished, while `dietpi-firstboot.service` runs before. But actually direct service starts/restarts do not respect (intentionally/reasonably) do not bother with init orders.

This workaround fixes it: https://github.com/MichaIng/DietPi/commit/99292013a49aab1ed0eadc0e0dc4dc214332b4b4

timocapa, 2021-08-09:
So, try again with both dev prep and dev branch selected?

MichaIng, 2021-08-09:
Jep, that would do it.

timocapa, 2021-08-09:
It booted.

[image]

MichaIng, 2021-08-09:
Just verified, it is indeed the init order, where we define [here](https://github.com/MichaIng/DietPi/blob/dev/rootfs/etc/systemd/system/dietpi-boot.service#L6) that Dropbear shall not start before DietPi-Boot has finished, to assure network is up and time synced first. `invoke-rc.d dropbear start` now waits forever (without any timeout) until `dietpi-boot.service` finishes, which never happens as it is configured to run after `dietpi-firstboot.service` from where `invoke-rc.d` is waiting.

I think here is the related changelog entry:
```
init-system-helpers (1.59) unstable; urgency=medium

  * invoke-rc.d: Remove unnecessary systemctl daemon-reload.
    We have proper support in update-rc.d and debhelper nowadays, which
    makes this superfluous.
  * invoke-rc.d,service: Stop using ignore-dependencies.
    Ignoring dependencies during bootup and shutdown can have undesired
    side-effects, like changing the ordering of services and ultimately
    causing them to fail. So drop this hack, which was added to work around
    issues in hook scripts. We only keep a special case for reload requests,
    where we schedule those with "no-block" to avoid unnecessary dead-locks.
    (Closes: #960595)
```
Just verified with a very simple test service, which does nothing but orders itself behind dietpi-boot.service and added `invoke-rc.d test start` to /boot/dietpi/boot, and voila, it hangs there forever.

timocapa, 2021-08-09:
I'm curious as to why this has been a hit or miss on Buster. I could consequently reboot the system with no issues - then a kernel update landed, rebooted and it died. But then just today, armbian buster to dietpi buster worked again? Possible race condition?

MichaIng, 2021-08-09:
Armbian pushes kernel upgrades very regularly, which is nice to be up-to-date but here and there they come with issues. As DietPi uses the very same kernel packages, the issue would then appear on both, and whether it is an issue on first boot of course depends on the kernel version installed on the image.

Btw, I applied the workaround to the PREP script now, so it is not required to use DietPi `dev` (only PREP script from dev branch): https://github.com/MichaIng/DietPi/commit/fc704e3a3652567904668c61b701275d09125ef9

The issue is solved more elegantly and in a long-term fashion now: https://github.com/MichaIng/DietPi/commit/1ecf972b16bb486c026c3ce56af0507cd6bcaade
Each existing host key is regenerated individually, instead of running the whole package reconfiguration, which implies a lot of additional checks and steps we do not want here. Little bonus is that it respects manual host key removals, e.g. when all SSH clients support `ed25519` already, then `rsa` and `ecdsa` host keys can be removed from the image (e.g. right after running PREP before poweroff). Or on Buster images, the legacy `dss` host key, which is still auto-generated there on package install/configuration.

timocapa, 2021-08-09:
> here and there they come with issues

I don't think the kernel was ever an issue here though, as Armbian has always worked. It was mostly just an example.

Good to see it was fixed though. Now trying to figure out why the Query log of pihole doesn't load on Bullseye but that's a whole different issue :p

MichaIng, 2021-08-09:
> then a kernel update landed, rebooted and it died

That sounds like an issue with the kernel upgrade. But of course could be a side-effect of the disk writes or so.

Depending on how you installed Pi-hole, there is an issue with the default Lighttpd config shipped with the Pi-hole installer to load a deprecated module: https://github.com/pi-hole/pi-hole/pull/4221
The fix has not yet been released, but AFAIK they are only waiting for a dnsmasq release expected the next days.

timocapa, 2021-08-09:
> That sounds like an issue with the kernel upgrade. But of course could be a side-effect of the disk writes or so.

Maybe, maybe not, we'll never know. I just moved to `linux-dtb-edge-sunxi` / `linux-image-edge-sunxi` and it's still happy now, so that's good.

> Depending on how you installed Pi-hole

Official, so curling their install script. 

> pi-hole/pi-hole#4221

I checked out to dev, so I should have this. Query log gets stuck on `Processing...`, wiping browser data doesn't fix it, like Google suggests.

But I don't think this is on topic anymore, so heyo.

MichaIng, 2021-08-09:
Probably something for the Pi-hole forum then. I think real world testing on Debian Bullseye has not yet been done much. Check browser console, webserver logs in `/var/log/lighttpd/error.log` and FTL logs in `/var/log/pihole-FTL.log` first, probably they provide a hint already.

timocapa, 2021-08-09:
Both logs are fine, browser console brings this

[image]

I'll close this as a "the actual issue is fixed" in hopes it gets less confusing if we go OT :)

closed 2021-08-09 by timocapa

MichaIng, 2021-08-09:
That should help the Pi-hole devs to identify the issue in the `queries.js` where the `length` property is called a few times: https://github.com/pi-hole/AdminLTE/blob/master/scripts/pi-hole/js/queries.js
In your browser devtools you can check the exact line `104`, as in the source this does not match, when something on the way strips empty lines or such away. Probably a recent PHP v7.4 version became stricter, as typically such TypeErrors started to appear with PHP8.0.

timocapa, 2021-08-09:
This line: https://github.com/pi-hole/AdminLTE/blob/devel/scripts/pi-hole/js/queries.js#L104

[image]

Added here: https://github.com/pi-hole/AdminLTE/commit/a120e725243361b20024bfab51dca7db249dc12d#diff-b36e3890f389f9a9166cf8cbbdf1e54ec1778ac715e7548dc9f84d569dd4a768R101

MichaIng, 2021-08-09:
Indeed, there is a check missing whether the variable is set at all. Easiest is probably to assign an empty string to the variable in the first place, if the EDE is not given:
```js
var ede = data[11];
```
=>
```js
var ede = (data[11]) ? data[11] : '';
```

timocapa, 2021-08-09:
Worked. Query log loads.

MichaIng, 2021-08-09:
I just wonder why this is an issue on Debian Bullseye (only), since all of that is executed in the browser and not on the server 🤔.

timocapa, 2021-08-09:
Perhaps newer php / lighttpd version?

`PHP 7.4.21 (cli) (built: Jul  2 2021 03:59:48) ( NTS )`
`lighttpd/1.4.59 (ssl) - a light and fast webserver`

MichaIng, 2021-08-09:
But neither PHP not the webserver manipulate or execute JavaScript code, but your browser does. But probably either the PHP version or Lighttpd are the reason why EDE is not set in the first place 🤔. However, I'll open a PR to fix/workaround via assignment, then probably the devs can have a closer look why this applies in the first place.

timocapa, 2021-08-09:
i wasnt sure if i should have opened an issue on an unsupported debian version, but i suppose at some point they wouldve run into the same issue

MichaIng, 2021-08-09:
PR open: https://github.com/pi-hole/AdminLTE/pull/1862