MichaIng/DietPi #4574: Bazarr crash after 0.9.6 update

Dr0bac, 2021-07-22:
With 0.9.6 update i see in release info that something is added to requirements.txt or changed.
here is the systemctl log.

````
root@DietPi:~# systemctl -l status bazarr.service
● bazarr.service - Bazarr Daemon (DietPi)
   Loaded: loaded (/etc/systemd/system/bazarr.service; disabled; vendor preset: enabled)
   Active: inactive (dead)

Jul 21 22:49:28 DietPi Bazarr[1973]: 2021-07-21 22:49:28,472 - root                             (76fc1ad0) :  ERROR (init:65) - BAZARR requirements.txt installation result: b'  ERROR: Comma
nd errored out with exit status 1:\n   command: /usr/bin/python3 /usr/local/lib/python3.7/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-pacbj030/overlay
 --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- \'setuptools >= 40.8.0\' wheel \'Cython >= 3.0a6\' \'cffi >= 1.12.3 ; platform_python_implem
entation == \'"\'"\'CPython\'"\'"\'\' \'greenlet >= 0.4.17, < 2.0 ; platform_python_implementation == \'"\'"\'CPython\'"\'"\'\'\n       cwd: None\n  Complete output (146 lines):\n  Collecti
ng cffi>=1.12.3\n    Using cached cffi-1.14.6.tar.gz (475 kB)\n  Collecting Cython>=3.0a6\n    Using cached Cython-3.0.0a9-py2.py3-none-any.whl (1.1 MB)\n  Collecting greenlet<2.0,>=0.4.17\
n    Using cached greenlet-1.1.0-cp37-cp37m-linux_armv7l.whl\n  Collecting wheel\n    Using cached wheel-0.36.2-py2.py3-none-any.whl (35 kB)\n  Collecting pycparser\n    Using cached pycpar
ser-2.20-py2.py3-none-any.whl (112 kB)\n  Collecting setuptools>=40.8.0\n    Using cached setuptools-57.4.0-py3-none-any.whl (819 kB)\n  Building wheels for collected packages: cffi\n    Bu
ilding wheel for cffi (setup.py): started\n    Building wheel for cffi (setup.py): finished with status \'error\'\n    ERROR: Command errored out with exit status 1:\n     command: /usr/bin
/python3 -u -c \'import sys, setuptools, tokenize; sys.argv[0] = \'"\'"\'/tmp/pip-install-bedxyxx9/cffi_5da16ba02ebd48b59525968e0260414d/setup.py\'"\'"\'; __file__=\'"\'"\'/tmp/pip-install-
bedxyxx9/cffi_5da16ba02ebd48b59525968e0260414d/setup.py\'"\'"\';f=getattr(tokenize, \'"\'"\'open\'"\'"\', open)(__file__);code=f.read().replace(\'"\'"\'\\r\\n\'"\'"\', \'"\'"\'\\n\'"\'"\');
f.close();exec(compile(code, __file__, \'"\'"\'exec\'"\'"\'))\' bdist_wheel -d /tmp/pip-wheel-pzw5rpbe\n         cwd: /tmp/pip-install-bedxyxx9/cffi_5da16ba02ebd48b59525968e0260414d/\n    C
omplete output (56 lines):\n    Package libffi was not found in the pkg-config search path.\n    Perhaps you should add the directory containing `libffi.pc\'\n    to the PKG_CONFIG_PATH env
ironment variable\n    No package \'libffi\' found\n    Package libffi was not found in the pkg-config search path.\n    Perhaps you should add the directory containing `libffi.pc\'\n    to
 the PKG_CONFIG_PATH environment variable\n    No package \'libffi\' found\n    Package libffi was not found in the pkg-config search path.\n    Perhaps you should add the directory contain
ing `libffi.pc\'\n    to the PKG_CONFIG_PATH environment variable\n    No package \'libffi\' found\n    Package libffi was not found in the pkg-config search path.\n    Perhaps you should a
dd the directory containing `libffi.pc\'\n    to the PKG_CONFIG_PATH environment variable\n    No package \'libffi\' found\n    Package libffi was not found in the pkg-config search path.\n
    Perhaps you should add the directory containing `libffi.pc\'\n    to the PKG_CONFIG_PATH environment variable\n    No package \'libffi\' found\n    running bdist_wheel\n    running buil
d\n    running build_py\n    creating build\n    creating build/lib.linux-armv7l-3.7\n    creating build/lib.linux-armv7l-3.7/cffi\n    copying cffi/verifier.py -> build/lib.linux-armv7l-3.
7/cffi\n    copying cffi/vengine_gen.py -> build/lib.linux-armv7l-3.7/cffi\n    copying cffi/vengine_cpy.py -> build/lib.linux-armv7l-3.7/cffi\n    copying cffi/setuptools_ext.py -> build/l
ib.linux-armv7l-3.7/cffi\n    copying cffi/recompiler.py -> build/lib.linux-armv7l-3.7/cffi\n    copying cffi/pkgconfig.py -> build/lib.linux-armv7l-3.7/cffi\n    copying cffi/model.py -> b
uild/lib.linux-armv7l-3.7/cffi\n    copying cffi/lock.py -> build/lib.linux-armv7l-3.7/cffi\n    copying cffi/ffiplatform.py -> build/lib.linux-armv7l-3.7/cffi\n    copying cffi/error.py ->
 build/lib.linux-armv7l-3.7/cffi\n    copying cffi/cparser.py -> build/lib.linux-armv7l-3.7/cffi\n    copying cffi/commontypes.py -> build/lib.linux-armv7l-3.7/cffi\n    copying cffi/cffi_o
pcode.py -> build/lib.linux-armv7l-3.7/cffi\n    copying cffi/backend_ctypes.py -> build/lib.linux-armv7l-3.7/cffi\n    copying cffi/api.py -> build/lib.linux-armv7l-3.7/cffi\n    copying c
ffi/__init__.py -> build/lib.linux-armv7l-3.7/cffi\n    copying cffi/_cffi_include.h -> build/lib.linux-armv7l-3.7/cffi\n    copying cffi/parse_c_type.h -> build/lib.linux-armv7l-3.7/cffi\n
    copying cffi/_embedding.h -> build/lib.linux-armv7l-3.7/cffi\n    copying cffi/_cffi_errors.h -> build/lib.linux-armv7l-3.7/cffi\n    running build_ext\n    building \'_cffi_backend\' e
xtension\n    creating build/temp.linux-armv7l-3.7\n    creating build/temp.linux-armv7l-3.7/c\n    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-stron
g -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DUSE__THREAD -DHAVE_SYNC_SYNCHRONIZE -I/usr/include/ffi -I/usr/include/libffi -I/usr/include/python3.7m -c c/_cffi_
backend.c -o build/temp.linux-armv7l-3.7/c/_cffi_backend.o\n    c/_cffi_backend.c:15:10: fatal error: ffi.h: No such file or directory\n     #include <ffi.h>\n              ^~~~~~~\n    com
pilation terminated.\n    error: command \'arm-linux-gnueabihf-gcc\' failed with exit status 1\n    ----------------------------------------\n    ERROR: Failed building wheel for cffi\n
Running setup.py clean for cffi\n  Failed to build cffi\n  Installing collected packages: pycparser, wheel, setuptools, greenlet, Cython, cffi\n      Running setup.py install for cffi: star
ted\n      Running setup.py install for cffi: finished with status \'error\'\n      ERROR: Command errored out with exit status 1:\n       command: /usr/bin/python3 -u -c \'import sys, setu
ptools, tokenize; sys.argv[0] = \'"\'"\'/tmp/pip-install-bedxyxx9/cffi_5da16ba02ebd48b59525968e0260414d/setup.py\'"\'"\'; __file__=\'"\'"\'/tmp/pip-install-bedxyxx9/cffi_5da16ba02ebd48b5952
5968e0260414d/setup.py\'"\'"\';f=getattr(tokenize, \'"\'"\'open\'"\'"\', open)(__file__);code=f.read().replace(\'"\'"\'\\r\\n\'"\'"\', \'"\'"\'\\n\'"\'"\');f.close();exec(compile(code, __fi
le__, \'"\'"\'exec\'"\'"\'))\' install --record /tmp/pip-record-wcr2pil6/install-record.txt --single-version-externally-managed --prefix /tmp/pip-build-env-pacbj030/overlay --compile --inst
all-headers /tmp/pip-build-env-pacbj030/overlay/include/python3.7m/cffi\n           cwd: /tmp/pip-install-bedxyxx9/cffi_5da16ba02ebd48b59525968e0260414d/\n      Complete output (56 lines):\
n      Package libffi was not found in the pkg-config search path.\n      Perhaps you should add the directory containing `libffi.pc\'\n      to the PKG_CONFIG_PATH environment variable\n
    No package \'libffi\' found\n      Package libffi was not found in the pkg-config search path.\n      Perhaps you should add the directory containing `libffi.pc\'\n      to the PKG_CONF
IG_PATH environment variable\n      No package \'libffi\' found\n      Package libffi was not found in the pkg-config search path.\n      Perhaps you should add the directory containing `li
bffi.pc\'\n      to the PKG_CONFIG_PATH environment variable\n      No package \'libffi\' found\n      Package libffi was not found in the pkg-config search path.\n      Perhaps you should
add the directory containing `libffi.pc\'\n      to the PKG_CONFIG_PATH environment variable\n      No package \'libffi\' found\n      Package libffi was not found in the pkg-config search
path.\n      Perhaps you should add the directory containing `libffi.pc\'\n      to the PKG_CONFIG_PATH environment variable\n      No package \'libffi\' found\n      running install\n
 running build\n      running build_py\n      creating build\n      creating build/lib.linux-armv7l-3.7\n      creating build/lib.linux-armv7l-3.7/cffi\n      copying cffi/verifier.py -> bu
ild/lib.linux-armv7l-3.7/cffi\n      copying cffi/vengine_gen.py -> build/lib.linux-armv7l-3.7/cffi\n      copying cffi/vengine_cpy.py -> build/lib.linux-armv7l-3.7/cffi\n      copying cffi
[... cut ...]
ges/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-10r0yses/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- \'s
etuptools >= 40.8.0\' wheel \'Cython >= 3.0a6\' \'cffi >= 1.12.3 ; platform_python_implementation == \'"\'"\'CPython\'"\'"\'\' \'greenlet >= 0.4.17, < 2.0 ; platform_python_implementation =
= \'"\'"\'CPython\'"\'"\'\' Check the logs for full command output.\n'
Jul 22 09:59:48 DietPi Bazarr[28317]: Bazarr starting...
Jul 22 09:59:48 DietPi Bazarr[28317]: Bazarr exited.
Jul 22 09:59:48 DietPi systemd[1]: bazarr.service: Succeeded.
````

Joulinar, 2021-07-22:
Hi,

Looks like some package is missing.  `Package libffi was not found`

Can you try following and restart Bazarr afterwards
`apt install libffi-dev`

MichaIng, 2021-07-22:
`setuptools` has been added, but that is installed already along with Python 3 pip. 3 months earlier, this has been added:
```
gevent>=21
gevent-websocket>=0.10.1
```
and see related changelog entry. I guess if pulls `cffi` as dependency. The only strange thing is that there are pre-compiled wheels for all platform.

Can you check whether piwheels.org is already installed as source, and in case install it?
```sh
cat /etc/pip.conf | grep piwheels || echo -e '[global]\nextra-index-url=https://www.piwheels.org/simple/' > /etc/pip.conf
```

Joulinar, 2021-07-22:
> Can you check whether piwheels.org is already installed as source, and in case install it?

This is applicable for 32bit systems only as piwheels.org did not provide aarch64 packages.

MichaIng, 2021-07-22:
According to the error log, builds are done against armv7l and aarch64 wheels are available via regular pypi.org: https://pypi.org/project/cffi/#files

Dr0bac, 2021-07-23:
@MichaIng i did what you said:
``cat /etc/pip.conf | grep piwheels || echo -e '[global]\nextra-index-url=https://www.piwheels.org/simple/' > /etc/pip.conf``
After that i got message something like this: ``cat: no such directory``
But when i started bazarr after that with ``python3 bazarr.py`` it started successfully, and after rpi reboot i had no issues as well.

closed 2021-07-23 by Dr0bac

Joulinar, 2021-07-23:
> After that i got message something like this: cat: no such directory

That's fine. The availability of `/etc/pip.conf` was checked and as it was not available, config file was created. This enabled Bazarr to start successfully.

MichaIng, 2021-07-23:
Shall we check and in case create that config on next DietPi update, when Bazarr is installed on an ARMv6/7 system? 🤔

Joulinar, 2021-07-23:
Or for all Python installation in general? Not only Bazarr?

MichaIng, 2021-07-23:
I'm just a little concerned as how much trouble it brought to the Raspberry Pi foundation when a package update silently added the MS VS Code repository on their system. piwheels.org can be seen as a completely different case, but in principle we're doing something similar here when adding a 3rd party software source that is used automatically by a package manager. It's okay when doing this during a software install, where it's transparent and clear what it is for and in which context it was added, but during updates I'm a bit careful with such.

I think it's okay when combining it with a whiptail prompt, giving a choice. In theory someone could have also manually removed it and don't want it back.

Joulinar, 2021-07-23:
There you have a point. So going with a whiptail should be fine

MichaIng, 2021-07-23:
Done: https://github.com/MichaIng/DietPi/commit/f85149b894987d7653193154ffb8279eae9d4189