MichaIng/DietPi #4576: RoonBridge not working on

GrahamGIGA, 2021-07-22:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=3
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
10.10
- Kernel version | `uname -a`
Linux Roon-Demo 5.10.17-v8+ #1421 SMP PREEMPT Thu May 27 14:01:37 BST 2021 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 3 Model B (aarch64)
- Power supply used | (EG: 5V 1A RAVpower)
Standard wall wart 5V 1a
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
RoonBridge
- Was the software title installed freshly or updated/migrated?
Fresh install
- Can this issue be replicated on a fresh installation of DietPi?
It is!
 Reference code: f9d291cd-46a2-4d60-bdae-819725acbf1a
- Bug report ID | `echo $G_HW_UUID`

See also video of stop behavior at: https://www.dropbox.com/s/qze7tpxltygd5f7/Roon%20htop%20demo.mp4?dl=0

#### Steps to reproduce

1. When configured as a Roon endpoint using IQaudIO Dac+ card, the endpoint is not visible to Roon.  

I notice when using top that process reaper and RoonBridgeHelper appear, drop and the reappear in a regular pattern (see Dropbox video link above)
[image]

#### Expected behaviour

All my other Rpi endpoints (all Diepi / RoonBridge configured bu diet setup) are visible.

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2021-07-23:
Many thanks for your report.

Can you please check the output of:
```sh
journalctl -u roonbridge
```
If it's too much:
```sh
journalctl -u roonbridge | tail -50
```

MichaIng, 2021-07-23:
Ah, I think I found it already in your bug report:
```
could not open log (app may be open already):System.IO.IOException: The file '/mnt/dietpi_userdata/roon/RoonBridge/Logs' already exists.
```
That is strange since those are supposed to be directories. Can you try:
```sh
G_EXEC mkdir -p /var/log/roon /mnt/dietpi_userdata/roon/{RoonBridge,RAATServer}
G_EXEC rm -Rf /mnt/dietpi_userdata/roon/{RoonBridge,RAATServer}/Logs
G_EXEC ln -s /var/log/roon /mnt/dietpi_userdata/roon/RoonBridge/Logs
G_EXEC ln -s /var/log/roon /mnt/dietpi_userdata/roon/RAATServer/Logs
```
I'll try to replicate the issue.

GrahamGIGA, 2021-07-23:
> 
> 
> Ah, I think I found it already in your bug report:
> 
> could not open log (app may be open already):System.IO.IOException: The file '/mnt/dietpi_userdata/roon/RoonBridge/Logs' already exists.
> That is strange since those are supposed to be directories. Can you try:
> 
> G_EXEC mkdir -p /var/log/roon /mnt/dietpi_userdata/roon/{RoonBridge,RAATServer}
> G_EXEC rm -Rf /mnt/dietpi_userdata/roon/{RoonBridge,RAATServer}/Logs
> G_EXEC ln -s /var/log/roon /mnt/dietpi_userdata/roon/RoonBridge/Logs
> G_EXEC ln -s /var/log/roon /mnt/dietpi_userdata/roon/RAATServer/Logs
> I'll try to replicate the issue.

root@Roon-Demo:~# G_EXEC mkdir -p /var/log/roon /mnt/dietpi_userdata/roon/{RoonBridge,RAATServer}
[  OK  ] mkdir -p /var/log/roon /mnt/dietpi_userdata/roon/RoonBridge /mnt/dietpi_userdata/roon/RAATServer
root@Roon-Demo:~# G_EXEC rm -Rf /mnt/dietpi_userdata/roon/{RoonBridge,RAATServer}/Logs
[  OK  ] rm -Rf /mnt/dietpi_userdata/roon/RoonBridge/Logs /mnt/dietpi_userdata/roon/RAATServer/Logs
root@Roon-Demo:~# G_EXEC ln -s /var/log/roon /mnt/dietpi_userdata/roon/RoonBridge/Logs
[  OK  ] ln -s /var/log/roon /mnt/dietpi_userdata/roon/RoonBridge/Logs
root@Roon-Demo:~# G_EXEC ln -s /var/log/roon /mnt/dietpi_userdata/roon/RAATServer/Logs
[  OK  ] ln -s /var/log/roon /mnt/dietpi_userdata/roon/RAATServer/Logs
root@Roon-Demo:~# 

Cheers

GrahamGIGA, 2021-07-23:
Too fast, on the return key.

root@Roon-Demo:~# G_EXEC mkdir -p /var/log/roon /mnt/dietpi_userdata/roon/{RoonBridge,RAATServer}
[  OK  ] mkdir -p /var/log/roon /mnt/dietpi_userdata/roon/RoonBridge /mnt/dietpi_userdata/roon/RAATServer
root@Roon-Demo:~# G_EXEC rm -Rf /mnt/dietpi_userdata/roon/{RoonBridge,RAATServer}/Logs
[  OK  ] rm -Rf /mnt/dietpi_userdata/roon/RoonBridge/Logs /mnt/dietpi_userdata/roon/RAATServer/Logs
root@Roon-Demo:~# G_EXEC ln -s /var/log/roon /mnt/dietpi_userdata/roon/RoonBridge/Logs
[  OK  ] ln -s /var/log/roon /mnt/dietpi_userdata/roon/RoonBridge/Logs
root@Roon-Demo:~# G_EXEC ln -s /var/log/roon /mnt/dietpi_userdata/roon/RAATServer/Logs
[  OK  ] ln -s /var/log/roon /mnt/dietpi_userdata/roon/RAATServer/Logs
root@Roon-Demo:~# 

That appeared to do it, after a reboot!

Cheers

MichaIng, 2021-07-23:
Ah yes, reboot or
```sh
systemctl restart roonbridge
```
should have been enough.

Not sure how the log directory was gone, it looks like completely reset, so if you have issues with other services, the reason is likely the same (a missing log file or directory in `/var/log`).

closed 2021-07-24 by MichaIng