MichaIng/DietPi #4578: Fresh install fails on orangepi zero lts

pcfreaxx, 2021-07-23:
#### Details:
- Date           | Fri Jul 23 09:22:08 CEST 2021
- DietPi version | v7.3.2 (MichaIng/master)
- Image creator  |
- Pre-image      |
- Hardware       |  (ID=22)
- Kernel version | `Linux orangepizero 5.10.43-sunxi #21.05.6 SMP Mon Jun 21 15:07:45 UTC 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | `/boot/dietpi/func/dietpi-set_software useradd dietpi`
- Exit code      | 127
- Software title | DietPi-PREP
#### Steps to reproduce:

1. ... virgin armbian buster converting to dietpi
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
./dietpi-globals: line 846: /boot/dietpi/func/dietpi-set_software: No such file or directory
```

Joulinar, 2021-07-23:
Can you check `ls -la /boot/dietpi/func/`

pcfreaxx, 2021-07-23:
sry system won't boot again :-( install is completely broken now

Joulinar, 2021-07-23:
Not sure if it is related but there was a similar request to create an image for OrangePI. If I recall correctly, there was an issue with network connection #4309

MichaIng, 2021-07-23:
If that file is missing, that sounds more like a broken SD card. Do you have another Linux system to run an fsck over the ext4 partition? Also, in case you still have it, a complete output of the PREP script until that error could help as well, probably a hint or (unhandled) error can be found prior to this already.

pcfreaxx, 2021-07-23:
Uhhhhh, i found the Problem. I had moved rootfs from armbian to USB before i started to Install dietpi.
Now in the next try, i just used the sd card everything ist working perfectly. Sorry for wasting your time. 
It was my first time using dietpi.

Joulinar, 2021-07-23:
Ok thx for sharing. I will close the issue for now. Feel free to reopen if needed.

closed 2021-07-23 by Joulinar