MichaIng/DietPi #4581: Kodi fails to start; RPi4; Debian Buster 10.10; aarch64; beta 7.4.1

inspector71, 2021-07-24:
```
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=4
G_DIETPI_VERSION_RC=1
G_GITBRANCH='beta'
G_GITOWNER='MichaIng'

buster 10.10

Linux DietPi 5.10.17-v8+ #1421 SMP PREEMPT Thu May 27 14:01:37 BST 2021 aarch64 GNU/Linux

RPi 4 Model B (aarch64)
```

PSU: DeskPi Pro

SanDisk USB stick

- _Software title_ Kodi
- _Fresh / upgrade_ New but maybe not completely fresh
- _Can this issue be replicated on a fresh installation of DietPi?_ Untried
- _Bug report ID_ 674c1438-5300-4d46-92fa-adfd1a274071

1. Installed DietPi
2. Installed a bunch of apps after tweaking config
3. Tried ...

```
dietpi@DietPi:~ $ xinit kodi-standalone

X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 5.4.0-54-generic aarch64 Debian
Current Operating System: Linux DietPi 5.10.17-v8+ #1421 SMP PREEMPT Thu May 27 14:01:37 BST 2021 aarch64
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:9B:1A:BD vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  console=tty1 root=PARTUUID=6a73df9f-6140-41e8-840e-f9aaa4b6aa1e rootfstype=ext4 rootdelay=10 elevator=deadline fsck.repair=yes rootwait quiet net.ifnames=0
Build Date: 19 April 2021  03:23:36PM
xorg-server 2:1.20.4-1+rpt3+deb10u3 (https://www.debian.org/support)
Current version of pixman: 0.36.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/home/dietpi/.local/share/xorg/Xorg.0.log", Time: Sat Jul 24 19:18:19 2021
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE)
Fatal server error:
(EE) parse_vt_settings: Cannot open /dev/tty0 (Permission denied)
(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE) Please also check the log file at "/home/dietpi/.local/share/xorg/Xorg.0.log" for additional information.
(EE)
(EE) Server terminated with error (1). Closing log file.
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error
```

Joulinar, 2021-07-24:
Hi,

Pls try to start Kodi as user `root`

inspector71, 2021-07-24:
> 
> 
> Hi,
> 
> Pls try to start Kodi as user `root`

Sure :)

```
root@DietPi:/home/dietpi# xinit kodi-standalone

X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 5.4.0-54-generic aarch64 Debian
Current Operating System: Linux DietPi 5.10.17-v8+ #1421 SMP PREEMPT Thu May 27 14:01:37 BST 2021 aarch64
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 smsc95xx.macaddr=DC:A6:32:9B:1A:BD vc_mem.mem_base=0x3eb00000 vc_mem.mem_size=0x3ff00000  console=tty1 root=PARTUUID=6a73df9f-6140-41e8-840e-f9aaa4b6aa1e rootfstype=ext4 rootdelay=10 elevator=deadline fsck.repair=yes rootwait quiet net.ifnames=0
Build Date: 19 April 2021  03:23:36PM
xorg-server 2:1.20.4-1+rpt3+deb10u3 (https://www.debian.org/support)
Current version of pixman: 0.36.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Sat Jul 24 21:56:23 2021
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(II) modeset(0): Initializing kms color map for depth 24, 8 bpc.
xinit: Unable to run program "xterm": No such file or directory
Specify a program on the command line or make sure that /usr/bin
is in your path.

xinit: connection to X server lost

waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.
```

No much luck though:

_Bug report_ 674c1438-5300-4d46-92fa-adfd1a274071

Joulinar, 2021-07-24:
Pls try `kodi --standalone`

MichaIng, 2021-07-24:
On RPi, the Kodi build does not support and use the X server. RPi 4 is using the KMS/GBM instead, previous RPi models DispmanX, if I'm not mistaken.

As long as the user is in `video` and `render` groups, `kodi --standalone` from console should work:
```sh
usermod -aG video,render
kodi --standalone
```

inspector71, 2021-07-24:
Tried a fresh install starting from 7.0.2 with the intention to test these suggestions (thanks!).

Looks like it has become stuck after DNS resolution problems I think I saw somewhere else:

```
 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Upgrading APT packages

[ INFO ] DietPi-Update | APT update, please wait...
Get:1 https://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Get:2 https://archive.raspberrypi.org/debian buster/main armhf Packages [376 kB]
Err:3 https://deb.debian.org/debian buster InRelease
  Could not resolve 'deb.debian.org'
Err:4 https://deb.debian.org/debian buster-updates InRelease
  Could not resolve 'deb.debian.org'
Err:5 https://deb.debian.org/debian-security buster/updates InRelease
  Could not resolve 'deb.debian.org'
Err:6 https://deb.debian.org/debian buster-backports InRelease
  Could not resolve 'deb.debian.org'
Get:7 https://archive.raspberrypi.org/debian buster/main arm64 Packages [287 kB]
Fetched 696 kB in 1min 10s (9989 B/s)
Reading package lists...
W: Failed to fetch https://deb.debian.org/debian/dists/buster/InRelease  Could not resolve 'deb.debian.org'
W: Failed to fetch https://deb.debian.org/debian/dists/buster-updates/InRelease  Could not resolve 'deb.debian.org'
W: Failed to fetch https://deb.debian.org/debian-security/dists/buster/updates/InRelease  Could not resolve 'deb.debian.org'
W: Failed to fetch https://deb.debian.org/debian/dists/buster-backports/InRelease  Could not resolve 'deb.debian.org'
W: Some index files failed to download. They have been ignored, or old ones used instead.
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
```
Cannot test with it stuck at this point.

Joulinar, 2021-07-24:
Do you use PiHole or any other AdBlocker? You have issues to resolve global Debian apt repository `deb.debian.org`

inspector71, 2021-07-24:
> 
> 
> Do you use PiHole or any other AdBlocker? You have issues to resolve global Debian apt repository `deb.debian.org`

No, the device is directly connected via wired ethernet to the router. Nothing but a DHCP lease and the corresponding router-based DNS resolver. A separate device can load the deb.debian.org web page, albeit that device is using a VPN,hopefully without DNS leakage :)

*sigh*

Joulinar, 2021-07-24:
Well you would need to check why you are not able to resolve the global Debian apt repository. Maybe you can give a static IP to the RPi using a different DNS server

inspector71, 2021-07-24:
> 
> 
> Well you would need to check why you are not able to resolve the global Debian apt repository. Maybe you can give a static IP to the RPi using a different DNS server

Flipped router DNS to `9.9.9.9`. Did not seem to be a complete fix but something enough APT updating seems to have happened to let app installs progress.

inspector71, 2021-07-24:
Finally *appeared* to get through all the DNS failures and so on. Only for this to happen, unfortunately:

```
root@DietPi:~# kodi --standalone
ERROR: Unable to create GUI. Exiting
double free or corruption (out)
Aborted
Crash report available at /root/kodi_crashlog-20210725_004524.log
```
Bug report: _89b7423a-2c44-4621-b511-8e61f105d106_

Joulinar, 2021-07-24:
Can you share the crash log `/root/kodi_crashlog-20210725_004524.log`

inspector71, 2021-07-24:
OK, here's a strange one. I re-ran `apt update` three times without DNS errors.

I used `dietpi-config` to set Kodi to autostart with the 0 uid.

I rebooted. 

Kodi is running!

inspector71, 2021-07-24:
> 
> 
> Can you share the crash log `/root/kodi_crashlog-20210725_004524.log`

Sure. Now that I changed SSH servers from Dropbear to OpenSSH, I can SFTP it to this device and ...

[kodi_crashlog-20210725_004524.log](https://github.com/MichaIng/DietPi/files/6872658/kodi_crashlog-20210725_004524.log)

Joulinar, 2021-07-24:
Usually there is no need to switch SSH server. You could have installed OpenSSH client (ID 0) and use SCP 😉

inspector71, 2021-07-24:
prefer OpenSSH

inspector71, 2021-07-24:
unfortunately the Kodi version, `17.6 Krypton`, is quite old.

FYI DietPi self upgraded from `7.0.2` to `7.3.2`

Joulinar, 2021-07-24:
Kodi is installed from Debian repository as you use 64bit. On 32bit, the Raspbian repository is used and it should ship version 18 or something. Not really something related to DietPi. Option would be to switch to Bullseye, use Buster backports or compile it yourself.

inspector71, 2021-07-24:
> 
> 
> Kodi is installed from Debian repository as you use 64bit. On 32bit, the Raspbian repository is used and it should ship version 18 or something. Not really something related to DietPi. Option would be to switch to Bullseye, use Buster backports or compile it yourself.

Thanks for the feedback. 64bit is still so very messy in the SBC world. Oh well.

Let me know if I can help debug the original issue in this thread.

MichaIng, 2021-07-24:
Ah sorry, I also completely forgot about the aarch64 thing. Yes in that case, there is not yet a Raspberry Pi build but instead the Debian version (old and X11 based) is installed. Our Kodi start script, used by the autostart option as well, knows that and starts Kodi accordingly via xinit: https://github.com/MichaIng/DietPi/blob/master/dietpi/misc/start_kodi

And your first attempt `xinit kodi-standalone` failed because xinit requires a full path to executables. `xinit /usr/bin/kodi-standalone` should have worked. Now I also understand why it was looking for xterm as default X client 😄.

Btw, there is a much newer Kodi available from backports: https://packages.debian.org/buster-backports/kodi
Try this one:
```sh
apt install -t buster-backports kodi
```

inspector71, 2021-07-25:
@Michaelng: brilliant summary. Thanks for that detailed effort. I will try to get into the ins and outs of it soon. Cannot continue with the current Kodi `17.6` with no H/W acceleration. Completely under-utilising the capabilities of the RPi4 that I am quite impressed with having experienced them for about 6 months with LibreELEC. Hoping you suggestions will help because the *arr and Jackett installations went like a breeze. If they are that easy to update, incredible!

MichaIng, 2021-07-25:
Yes indeed via X11 by default it is not accelerated. But probably when using the KMS DDX it will be 🤔. Try the following:
```sh
/boot/dietpi/func/dietpi-set_hardware rpi-opengl vc4-fkms-v3d
```
or the full KMS driver:
```sh
/boot/dietpi/func/dietpi-set_hardware rpi-opengl vc4-kms-v3d-pi4
```

inspector71, 2021-07-25:
> Ah sorry, I also completely forgot about the aarch64 thing. 

Not at all. Honestly, happy to use 32 bit as I _only_ have a 4GB RPi4 ... it was just hard to opt for 32 instinctively as it, well, felt backwards ... even though I knew this to not be so much the case is the SBC world.

> Yes in that case, there is not yet a Raspberry Pi build but instead the Debian version (old and X11 based) is installed. Our Kodi start script, used by the autostart option as well, knows that and [...]

Ah, I see what you mean there.

> And your first attempt `xinit kodi-standalone` failed because xinit requires a full path to executables. `xinit /usr/bin/kodi-standalone` should have worked. Now I also understand why it was looking for xterm as default X client 😄.

Honestly, I just saw `xinit` on some page somewhere and thought it might have been the missing piece of the puzzle. Poking in the dark really, I'm afraid.

> Btw, there is a much newer Kodi available from backports: https://packages.debian.org/buster-backports/kodi
> Try this one:
> 
> ```shell
> apt install -t buster-backports kodi
> xinit /usr/bin/kodi --standalone
> ```

Thanks! Wow, I would like to do that but this DNS malarkey has got me miffed. Just cannot get reliable DNS! Even with my router pointing to `9.9.9.9` Sonarr.tv; *.debian.org; ... more or less every domain in `sources.list` has failed at one stage or another whilst doing various APT updating within the last two days. Mad. Never, ever, had this sort of trouble before.

> `kodi --standalone` vs `kodi-standalone` btw skips one unnecessary wrapper script which starts PulseAudio (not used by default on DietPi) and a failure loop. Aside of that, internally that script calls `kodi --standalone` as well.

Hmmm, there's a bit too much granularity here, I reckon. Maybe it's just me though. Best not to complain :)

> I was thinking already for a while whether we should install a config on DietPi which lifts the backports priority so that everything is pulled from backports (where available) in the first place. This also includes newer kernel versions on x86_64 systems etc. The only concern was that Raspbian has no backports, so we cannot reliably know which package/software versions are installed only based on the distro version, but need to differentiate between Raspbian and Debian then as well 🤔.

That does sound like a ballache. Seems abnormal that Rasbian doesn't do backports. Debs, Ubuntu, Mint? All do them, don't they?

inspector71, 2021-07-25:
> or the full KMS driver:
> 
> ```shell
> /boot/dietpi/func/dietpi-set_hardware rpi-opengl vc4-kms-v3d-pi4
> ```

No luck I'm afraid. Still running `ff-hevc (SW)` after this tweak.

inspector71, 2021-07-25:
> Yes indeed via X11 by default it is not accelerated. But probably when using the KMS DDX it will be 🤔. Try the following:
> 
> ```shell
> /boot/dietpi/func/dietpi-set_hardware rpi-opengl vc4-fkms-v3d
> ```

No luck here either. 

So seems like I could start again. 

What version of Kodi is available for DietPi 32 bit? 

Does the Kodi version in DietPi 32 bit support hardware acceleration for the RPi4?

Joulinar, 2021-07-25:
DietPi 32bit on RPi is using Raspberry OS 32bit and the Raspberry repository. There is no special DietPi version as DietPi is not an own OS. It's basically a diet Raspberry OS with some magic scripts.

```
root@DietPiProd:~# apt show kodi
Package: kodi
Version: 2:18.7-1~buster
```

MichaIng, 2021-07-25:
Yes, on Raspbian (Raspberry Pi OS 32-bit and hence our 32-bit RPi image) the Kodi build is a special RPi build with hardware acceleration.

inspector71, 2021-07-25:
> DietPi 32bit on RPi is using Raspberry OS 32bit and the Raspberry repository. There is no special DietPi version as DietPi is not an own OS. It's basically a diet Raspberry OS with some magic scripts.

I do like as many magic scripts as I can use, and the good people of DietPi land are goodly enough to write and maintain! :)

> ```
> root@DietPiProd:~# apt show kodi
> Package: kodi
> Version: 2:18.7-1~buster
> ```

Hmmm, slightly strange version numbe. Equivalent to Kodi 18.7 perhaps?

Either way, 32 bit seems the way to go.

inspector71, 2021-07-25:
> 
> 
> Yes, on Raspbian (Raspberry Pi OS 32-bit and hence our 32-bit RPi image) the Kodi build is a special RPi build with hardware acceleration.

Sounds like it's 32 bit all the way :)

MichaIng, 2021-07-25:
Its basically Kodi and Chromium where 64-bit builds with RPi GPU acceleration are missing. You can monitor it yourself:
- https://archive.raspberrypi.org/debian/pool/main/c/chromium-browser/?C=M;O=D
- https://archive.raspberrypi.org/debian/pool/main/k/kodi/?C=M;O=D

As fast as `*_arm64.deb` appear there, we have a GPU-accelerated build for the 64-bit image.

inspector71, 2021-07-25:
> Its basically Kodi and Chromium where 64-bit builds with RPi GPU acceleration are missing. You can monitor it yourself:
>
>     * [...]

Fair enough for sure.

Am currently looking into your automated install support :)

inspector71, 2021-07-26:
Wow, 32 bit not going much better. Got all excited about doing the dietpi.txt automated installation. Nope.

https://github.com/MichaIng/DietPi/issues/4587

Earlier, my Dietpi.txt automation did a lot but stopped ahead of installing the bulk of the software packages.

inspector71, 2021-08-08:
> Btw, there is a much newer Kodi available from backports: https://packages.debian.org/buster-backports/kodi
> Try this one:
> 
> ```shell
> apt install -t buster-backports kodi
> xinit /usr/bin/kodi --standalone
> ```

Think I misseed this initially but just trying it now, no luck:

Joulinar, 2021-08-08:
Actually you are on 32bit or 64bit?

inspector71, 2021-08-08:
> 
> 
> Actually you are on 32bit or 64bit?

`Device model : RPi 4 Model B (armv7l)`

That's 32bit, isn't it?

Joulinar, 2021-08-08:
Yes, 32bit did not have a buster-backports

inspector71, 2021-08-09:
> 
> 
> Yes, 32bit did not have a buster-backports

OK, well I'm getting a showstopper bug - Pulse CEC adaptor loading notification appears as normal but no CEC - with the current 32bit version anyway, so it's back to LibreELEC for something functional :(

Joulinar, 2021-08-09:
There was an issue with the Raspberry OS kernel. Could you check if you can update by running `apt update && apt upgrade`

Reference #4613

inspector71, 2021-08-09:
Can do, but for the moment I am trying an `aarch64` install again, hoping to go with Kodi `19.1` via `buster-backports`.

The 32 bit raspbian Kodi binary has poor decoding of x265 / HEVC in that it renders with hardware acceleration but too many videos I could play fine with LibreELEC 9.2.7 get picture break up that creates sync issues and sometimes freezes Kodi.

MichaIng, 2021-08-09:
It's a bit sad that the RPi build of Kodi was not updated for more than a year now: https://archive.raspberrypi.org/debian/pool/main/k/kodi/?C=M;O=D
Not sure what the current state of that project is, if there is any active maintainer or not. Something to check back in the Raspberry Pi forum, I suggest. LibreELEC has its own build, and yeah it is a whole OS only for Kodi, so it is supposed to work best 😄. Not sure which kernel and library base they use, but would be awesome if there was some collaboration between LibreELEC and the RPi Kodi maintainers for the hardware acceleration part of current Kodi versions, so that there is no doubled efforts on this.

inspector71, 2021-08-09:
> 
> 
> It's a bit sad that the RPi build of Kodi was not updated for more than a year now: https://archive.raspberrypi.org/debian/pool/main/k/kodi/?C=M;O=D

That does seem odd, however, might correlate with why OSMC is yet to publish support for RPi4. RASFo employed a video driver guy some time ago. Wonder if he's still working with them.

> Not sure what the current state of that project is, if there is any active maintainer or not.

[Kodi 19 Matrix involves the move to Python3 and elimination of proprietary rendering methods](https://libreelec.tv/2021/02/upcoming-changes/), so that may be a factor in why the Raspbian 32 Bit Kodi build is Kodi 18.7. After all, LibreELEC themselves are yet to release version 10, the Kodi 19 Matrix version, and openly states Raspberry Pi 4 will only get a: 

> _Stable "Alpha" build as the code is still very new"._

[Although LibreELEC 10.0 BETA5, actually based on Kodi Matrix 19.1, was released on June 27th/.](https://libreelec.tv/2021/06/libreelec-matrix-10-0-beta5/)

> Something to check back in the Raspberry Pi forum, I suggest. LibreELEC has its own build, and yeah it is a whole OS only for Kodi, so it is supposed to work best 😄. Not sure which kernel and [...]

Agreed, efforts would be best coordinated where human egos and so forth can handle it. LibreELEC and 'Team Kodi' seem to work together:

> _LibreELEC is helping Team Kodi push forwards towards their goal of eliminating proprietary decoding methods._ 

> _LibreELEC staff are working with Team Kodi to move all supported platforms onto modern (aka mainline) Linux kernels and a common display/decoding method, GBM/V4L2._

inspector71, 2021-08-10:
> ```shell
> /boot/dietpi/func/dietpi-set_hardware rpi-opengl vc4-fkms-v3d
> ```

LibreELEC has a `distroconfig.txt` file that is included in `config.txt`. Contains:

> dtoverlay=vc4-fkms-v3d

Tried a `buster-backports` 64 bit `aarch64` version of Kodi. Alas, hardware decoding did not work out of the box. If I just run:

> ```shell
> /boot/dietpi/func/dietpi-set_hardware vc4-fkms-v3d
> ```

... will that instruct DietPi to use the `vc4-fkms-v3d` driver that seems to be working well with LibreELEC?

Is a reboot required?

inspector71, 2021-08-11:
thanks for you great support Michal.

it looks like my options are exhausted.

OSMC night have been the best choice, no RPi4 support.

LibreELEC and dockSTARTer to install downloading apps but BusyBox doesn't
support `git` which dockSTARTer requires. Meanwhile the DeskPi Pro doesn't
support LibreELEC so no proper power down, fan control or functioning front
USB ports.

DietPi does everything, though I've not had it running with DeskPi Pro
support just yet. Everything it seems except keep up with LibreELEC / Kodi
... which as you've diligently explained is probably not feasible due to
the dedication required to reach LibreELEC's Kodi housing refinement.

thank you for all your exceptional support. maybe I'll come back to this
nightmare of I ever have more capability.

MichaIng, 2021-08-11:
Okay, I'll close the issue for now, as all questions should be answered. Feel free to reopen if required.

closed 2021-08-11 by MichaIng