MichaIng/DietPi #4582: First Boot on Nano Pi 2 fails.

Dtrieb, 2021-07-24:
First Boot on Nano Pi 2 fails.

Get it solved with:

`resize2fs /dev/mmcblk0p2`

[https://dietpi.com/phpbb/viewtopic.php?p=32324#p32324](https://dietpi.com/phpbb/viewtopic.php?p=32324#p32324)

Joulinar, 2021-07-24:
Hi,

Can you share following

```
/var/tmp/dietpi/logs/dietpi-firstboot.log
/var/tmp/dietpi/logs/dietpi-firstrun-setup.log
/var/tmp/dietpi/logs/fs_partition_resize.log
```

Dtrieb, 2021-07-24:
[fs_partition_resize.log](https://github.com/MichaIng/DietPi/files/6872643/fs_partition_resize.log)
[dietpi-firstrun-setup.log](https://github.com/MichaIng/DietPi/files/6872644/dietpi-firstrun-setup.log)
[dietpi-firstboot.log](https://github.com/MichaIng/DietPi/files/6872645/dietpi-firstboot.log)

MichaIng, 2021-07-24:
Hmm:
```
The partition table has been altered.
Error: Error informing the kernel about modifications to partition /dev/mmcblk0p2 -- Invalid argument.  This means Linux won't know about any changes you made to /dev/mmcblk0p2 until you reboot -- so you shouldn't mount it or use it in any way before rebooting.
Error: Partition(s) 2 on /dev/mmcblk0 have been written, but we have been unable to inform the kernel of the change, probably because it/they are in use.  As a result, the old partition(s) will remain in use.  You should reboot now before making further changes.
partx: /dev/mmcblk0: error updating partition 2
```
So `partprobe` and `partx` both failed. I think we should error-handle those and trigger a reboot when failed. Only problem, when  to do the filesystem resize then. When starting the same service, there is a change of a boot loop, which is not great either.

Probably we need to split those again, first partition resizing, if informing the kernel fails a reboot, then filesystem resizing. Or a flag file. I need to think about it for v7.5 🤔.

MichaIng, 2021-09-10:
Done with: #4724

If either `partprobe` or `partx` fails, the system is rebooted and a flag is set to skip the partition resize on next boot. The kernel loads the new partition table, the flag is removed and the filesystem is expanded. I have no system to replicate the issue, but I faked it here by changing one of these commands to an invalid one:
```
Removed /etc/systemd/system/local-fs.target.wants/dietpi-fs_partition_resize.service.
Disk /dev/sda: 8 GiB, 8589934592 bytes, 16777216 sectors
Disk model: VBOX HARDDISK
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x32713bac

Old situation:

Device     Boot Start      End  Sectors Size Id Type
/dev/sda1  *     2048 16777215 16775168   8G 83 Linux

/dev/sda1:
New situation:
Disklabel type: dos
Disk identifier: 0x32713bac

Device     Boot Start      End  Sectors Size Id Type
/dev/sda1  *     2048 16777215 16775168   8G 83 Linux

The partition table has been altered.
/var/lib/dietpi/services/fs_partition_resize.sh: line 59: partxb: command not found
Created symlink /etc/systemd/system/local-fs.target.wants/dietpi-fs_partition_resize.service → /etc/systemd/system/dietpi-fs_partition_resize.service.
[ INFO ] Rebooting to load the new partition table
Removed /etc/systemd/system/local-fs.target.wants/dietpi-fs_partition_resize.service.
resize2fs 1.46.2 (28-Feb-2021)
The filesystem is already 2096896 (4k) blocks long.  Nothing to do!
```

closed 2021-09-12 by MichaIng