MichaIng/DietPi #4585: questions: youtube-dl

maartenlangeveld, 2021-07-25:
Had youtube-dl binary installed manually with a cron job to update it weekly.

With DietPi v7.4.2. I have installed youtube-dl through dietpi-software. Have following questions:

1. Why is youtube-dl installed as script to run youtube-dl-py2 ?
2. Is youtube-dl updated regularly this way?

Thanks, Maarten

MichaIng, 2021-07-25:
Hi Maarten, thanks for your question.

We missed to have the docs ready for youtube-dl to explain this:
- `/usr/local/bin/youtube-dl` is a Python script with `python` shebang. So when calling it just like that, it will execute with Python 2, or, when not installed, fail. But it is Python 3 compatible.
- Since Python 2 is deprecated since beginning of 2020, we try to avoid using or even installing it whenever possible. So we store the `youtube-dl` script with a Python 2 suffix and create a Python 3 wrapper at the original location. So when calling `youtube-dl` from console, it will run with Python 3.

It's basically [this](https://github.com/ytdl-org/youtube-dl/issues/27649) we workaround.

maartenlangeveld, 2021-07-25:
@MichaIng,

I understand now, thanks a lot !!!

closed 2021-07-25 by maartenlangeveld