MichaIng/DietPi #4586: req: microk8s (optimized Kubernetes )

tomByrer, 2021-07-25:
### Creating a software request

#### Formal software information
- Software title | microk8s
- Short description | MicroK8s is a small, fast, single-package Kubernetes for developers, IoT and edge.
- Official URL (if available) | https://microk8s.io/
- GitHub URL (if available) | https://github.com/ubuntu/microk8s
- (Official) install documentation | 

#### Are there similar/alternative software titles available with DietPi-Software?
https://dietpi.com/docs/software/system_stats/#k3s

#### What makes your requested software better than the above solutions, if available?
Official ubuntu project, "Full enterprise support available, no license fees", directly supports RPi clusters

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?
unsure; untested, asking
- ...

Joulinar, 2021-07-25:
Does this requires Ubuntu? Because DietPi is based on Debian.

MichaIng, 2021-07-25:
It is shipped as Ubuntu snap container. Snap can be installed on Debian as well: https://packages.debian.org/buster/snapd

```sh
apt install snap
snap install microk8s --classic
```

So a container with a tool which runs containers 😄. Similar like with Docker containers, if there is not much reasonable we can do on top of the above two steps, there is not benefit of having this wrapped in `dietpi-software`.

silverraindog, 2022-02-02:
Not to sure how to go about this.
I would like to try add this feature to dietpi-software.

Although I wouldn't mind asking a few questions,

I see everything is done via dietpi-software shell script ? no other "directory" where the dietpi-software checks for plugin names or something ?

I am also assuming, I need to fork the repo, configure the setup in the appropriate section of dietpi-software (system Stats and Management), then create a pull request ?

(not new to creating software but new to public partisapation of software, this would be my first)

Joulinar, 2022-02-02:
let me link our wiki on how to add new software title. https://github.com/MichaIng/DietPi/wiki/How-to-add-a-new-software-title

All to be done inside `dietpi-software` script. Yes it is a huge script 🙄

MichaIng, 2022-02-02:
> I would like to try add this feature to dietpi-software.

That would be awesome.

Configuration files, services, assets or such can be also stored [here](https://github.com/MichaIng/DietPi/tree/dev/.conf), in `dps_<id>` subdirectory, then downloaded within install code via (as an example):
```sh
dps_index=$software_id Download_Install 'microk8s.conf' /mnt/dietpi_userdata/microk8s/microk8s.conf` 
```
Though actually I'm not too satisfied with this, since it breaks the possibility to do changes in these files which depend on the DietPi(-Software) version, as it is downloaded based on DietPi branch, not based on version. Long term it would be better to ship all these with DietPi locally but reduce `dietpi-software` code size by splitting individual software install/uninstall/config code into dedicated scripts which are then sourced on demand. Basically a plugin system like you mentioned, which would even allow to add custom software titles.

silverraindog, 2022-02-04:
Quick question, I want to add microk8s to the dietpi-software. On the page below, it says add the ID's of the dependencies. Microk8s requires snapd, which does not have a dependencies ID. Should one then create snapd like normal software, with an ID and then add it to the original application as an dependency or is there another way of doing it ?

I am assuming it's the "adding snapd to get a dependency id and then adding microk8s just want to make sure. Although on the forums Joulinar did mention to just make it install it. That got me thinking, about dependencies that are silent, Ie, you need for other programs but the user does not actually need them, like snapd. Is there a an extra script or part of dietpie-software that calls these extra dependencies?

MichaIng, 2022-02-04:
Lol, I just recognised it now, a container system which runs within a container system. So when you want to run an MicroK8s container, you run two container engines on top, that looks ridiculous on first view 😄. However, there are hopefully good reasons.

As long as it is not a dedicated software option, install Snap via `G_AGI snapd`. We may make an own software option out of it at a later time, makes somehow sense to have a more complete list of container system options listed.

silverraindog, 2022-02-06:
Microk8s Software: [PR](https://github.com/MichaIng/DietPi/pull/5239)
Microk8s Docs [PR](https://github.com/MichaIng/DietPi-Docs/pull/683)

MichaIng, 2022-03-02:
Marking this as closed. Topics regarding MicroK8s from now on deserve an own issue.

closed 2022-03-02 by MichaIng