MichaIng/DietPi #4587: [RPi4] Kodi installation bug: unable to install libcec4 libnfs12 kodi fbset

inspector71, 2021-07-26:
5adbef41-1671-4b70-be2e-5ac6cbb60b66

#### Details:
- Date           | Mon Jul 26 11:44:39 AEST 2021
- Bug report     | 5adbef41-1671-4b70-be2e-5ac6cbb60b66
- DietPi version | v7.4.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | `Linux DietPi 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | `apt-get -qq --allow-change-held-packages install libcec4 libnfs12 kodi fbset`
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Unable to correct problems, you have held broken packages.
```

inspector71, 2021-07-26:
Do not understand why but changing APT mirror appeared to fix this. Why the truck aren't APT mirrors synched better.

closed 2021-07-26 by inspector71

MichaIng, 2021-07-26:
I'm not sure either why such major APT servers are not setup a way to perform atomic updates:
- Update/sync to a 2nd/offline location, generate lists/signatures/checksums etc on that second location
- Switch to 2nd location via symlink, invalidate CDN cache, if any present, via API, or better have it disabled and purged before the update, re-enable it afterwards

Usually re-running apt update after some minutes helps as well.

inspector71, 2021-07-26:
I'll try more apps this time. Earlier, it was sqlite that caused a similar
error.

FWIW, have been trying to use local mirrors for speed. I remember when
`netselect` (as I now know it, but never knew it was called that at the
time, haha) was AFAIK a default package on Debbie ❤️s Ian (RIP) systems.
That was a very easy way to find the best performing mirrors. Last I saw
similar was `Mint` presumably using `netselect` as a library under their
very simple, but very effective, mirror picker GUI.

Regardless, not sure I can recall any of the local mirrors I've tried, in
the same way, over many years, being as faulty as they have been over
recent days.

Strange.

Maybe there's some DDoS botnet malarkey going on. Nefarious rascals.

MichaIng, 2021-07-26:
[netselect](https://packages.debian.org/buster/netselect) still exists and works fine. But it is not grat to estimate the best performing APT mirror. It checks only the ping time, and that can very quite a bit. But the ping is not relevant, but the bandwidth, which is not measured, and both correlate sometimes but often not.

Usually `deb.debian.org` and `raspbian.raspberrypi.org` should work just fine as mirror directors which use ping times, but also current load AFAIK. Check out how it works:
```
# curl -IL http://raspbian.raspberrypi.org/raspbian/pool/main/b/bash/bash_5.1-2_armhf.deb
HTTP/1.1 302 Found
Date: Mon, 26 Jul 2021 12:31:08 GMT
Server: Apache/2.4.25 (Debian)
X-MirrorBrain-Mirror: rwth-aachen.de
X-MirrorBrain-Realm: country
Link: <http://raspbian.raspberrypi.org/raspbian/pool/main/b/bash/bash_5.1-2_armhf.deb.meta4>; rel=describedby; type="application/metalink4+xml"
Link: <http://ftp.halifax.rwth-aachen.de/raspbian/raspbian/pool/main/b/bash/bash_5.1-2_armhf.deb>; rel=duplicate; pri=1; geo=de
Link: <http://mirror1.hs-esslingen.de/pub/Mirrors/archive.raspbian.org/raspbian/pool/main/b/bash/bash_5.1-2_armhf.deb>; rel=duplicate; pri=2; geo=de
Link: <http://ftp.agdsn.de/pub/mirrors/raspbian/raspbian/pool/main/b/bash/bash_5.1-2_armhf.deb>; rel=duplicate; pri=3; geo=de
Link: <http://mirror.de.leaseweb.net/raspbian/raspbian/pool/main/b/bash/bash_5.1-2_armhf.deb>; rel=duplicate; pri=4; geo=de
Link: <http://ftp.gwdg.de/pub/linux/debian/raspbian/raspbian/pool/main/b/bash/bash_5.1-2_armhf.deb>; rel=duplicate; pri=5; geo=de
Location: http://ftp.halifax.rwth-aachen.de/raspbian/raspbian/pool/main/b/bash/bash_5.1-2_armhf.deb
Connection: close
Content-Type: text/html; charset=iso-8859-1

HTTP/1.1 200 OK
Server: nginx/1.14.2
Date: Mon, 26 Jul 2021 12:31:09 GMT
Content-Type: application/octet-stream
Content-Length: 1295100
Last-Modified: Fri, 08 Jan 2021 10:08:15 GMT
Connection: keep-alive
Keep-Alive: timeout=30
Accept-Ranges: bytes
```