MichaIng/DietPi #4588: DietPi + PiHole/PiVPN connection permanently lost on router restart/ETH0 connection lost

GvY85, 2021-07-26:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7.4
- Distro version | Debian Buster (Backports)
- Kernel version | 5.10
- SBC model | x86_64 (UP Squared)
- Power supply used | n.a.
- SDcard used | n.a./eMMC

#### Additional Information (if applicable)
- Software title | DietPi general/PiHole+PiVPN
- Was the software title installed freshly or updated/migrated? no
- Can this issue be replicated on a fresh installation of DietPi? yes

#### Steps to reproduce
Remove ethernet cable from DietPi device to router and reconnect
and/or
Restart router

#### Expected behaviour
I would expect the ethernet connection to recover when you replug the ethernet cable or when the ethernet connection drops because (f.i.) the router restarts. 

#### Actual behaviour
I have a Fritz!BOX that connects to my ISP and acts as a switch. I have connected (via ethernet) a dedicated pc with DietPi and PiHole+PiVPN to it. What has been happening since the beginning (2 years ago) is that when for some reason (FritzBOX restart, connection drop) the ehternet connection drops it does not come back up.
Resulting in me having to walk to the closet and plug in a screen+keyboard or hotplug the psu of the DietPi device to restart it.
 This is not ideal and would be problematic if I am out of the home and the FRITZ!Box restarts for some reason.
I was under the impression Debian would solve this like Windows does but I assumed wrong. So I tried installing **ifplugd** and that seems to work for plugging out and in the ethernet cable for a short period of time and in the same port but does not work if I restart the FRITZ!Box. The LED's start doing stuff but the DietPi device is not reachable until I (forcefuly) restart it.

#### Extra details
I thought this could be related to the static IP in the DietPi device so I reset that to DHCP sine the FRITZ!Box is configured to give the device a fixed IP anyway but no change.
Putting a cheap unmanaged switch between FRITZ!Box and DietPi device seems to mitigate this issue but that is probably because the switch keeps the connection between itself and the DietPi device alive while the FRITZ!Box restarts.

My ifplugd config:

> /etc/ifplugd/ifplugd.conf:
> `INTERFACES="auto"
> HOTPLUG_INTERFACES=""
> ARGS="-q -f -u0 -d10 -w -I"
> SUSPEND_ACTION="stop"`

So in short: how do I get DietPi to automatically re-establish a connection when it gets dropped for some reason without having to restart the DietPi device?

Joulinar, 2021-07-26:
Usually Ethernet connection should be detected automatically. Kernel logs `dmesg` could give an indication if it's detected or not.

It's not fixing your issue but some time back we created a small user tutorial on how to use `ifplugd` https://dietpi.com/phpbb/viewtopic.php?f=15&t=7835

MichaIng, 2021-07-26:
`ifup eth0` or `ifup --force eth0` should do it. Actually I wonder why anything is required when you have a static IP. I have static IPs assigned here as well and when I unplug and replug the Ethernet cable, connections are just working as if nothing happened, stalled connections even resume by themselves (e.g. a running `apt update` trying to connect for a while) 🤔. Same when restarting the router. I tried that on a VM and with an RPi, can repeat with a notebook which is a closer match.

> the switch keeps the connection between itself

Well, as long there is not kind of download or established TCP connection alive, there is nothing to "keep". The only thing that may change is the carrier signal and the operation state "client-side":
```
cat /sys/class/net/eth0/carrier
cat /sys/class/net/eth0/operstate
```
In my case, when unplugging the Ethernet cable, both switches to `0`/`down`, when reattaching the cable, both switches back to `1`/`up`. Everything else stays as it was, including assigned IP addresses, routes etc.

Probably the Fritz!Box is behaving different when a carrier signal gets lost, but strange that a Fritz!Box restart causes the same issue, which indeed looks more like the "client" is the culprit.

Probably you can check whether IP addresses and routers stay active while the above changes like in my case:
```
ip a
ip r
```

GvY85, 2021-07-26:
Thanks for the replies.

The problem is also that when this happens I cant check anything because I dont have a keyboard + display in my "server room".

So just to be clear, in Debian a device with a static IP should resume working automatically when you unplug/replug in from its switch/router and or restarts its router/switch?
I don't understand why it now works with everything connected via a cheap TPLink 5 port switch connected to the Fritz!Box but not when directly connected to the Fritz!Box

Joulinar, 2021-07-26:
Well the switch will keep the physical connection active all the time even if internet connection is disrupted. On the other hand side, reboot your F!B will cause a drop of the physical connection while your router is rebooting.

GvY85, 2021-07-26:
Exactly. But if I manually disconnect a cable and reinsert in (thus dropping the physical connection) no problems. 
But when the F!B restarts it does not come back up.

GvY85, 2021-07-26:
It is very wierd.
I have 2 identical [UP Squared's](https://up-board.org/wp-content/uploads/2021/06/UP-Squared-Datasheet-v2.pdf) with Realtek 8111G chipsets.
Connecting them to a TPLink unmanaged switch I can unplug them and replug them all I want and no problems. 
But connecting them to the F!B it doenst work. Seems almost like a HW bug or some chipset incompatibilities

GvY85, 2021-07-26:
I think I can close this, its not DietPi related I think.
Fritz!BOX -> Switch -> Devices: no problems
Frirz!BOX -> Devices: unplug and replug cable no problem, remove power from Fritz!BOX no problem, select restart in Fritz!BOX UI: devices dont come back up.

Perhaps a bug in the 811G chip/fw/driver or something. I tried updating the drivers and firmware but no change.

closed 2021-07-26 by GvY85

MichaIng, 2021-07-26:
> Frirz!BOX -> Devices: unplug and replug cable no problem, remove power from Fritz!BOX no problem

Ah, in OP this seems to have failed as well?

If the Fritz!Box does not show helpful logs, doing the checks on the DietPi system suggested above + `dmesg` could give some indication why the interface does not come up, or the FBox does not pick it up anymore as suggested.

Probably, if it is really a driver or firmware issue, it got resolved on newer versions. On Debian backports, a newer kernel as well as newer Realtek firmware is available which you can try:
```sh
apt install -t buster-backports linux-image-amd64 firmware-realtek
reboot
# If all works fine, to remove the old obsolete kernel:
apt autopurge
```

The backports kernel btw can be seen as very stable. It has been tested thoroughly and will be released as stable in Debian Bullseye on August 14.

Joulinar, 2021-07-26:
Usually an issue could be open with AVM as well to have it checked.

GvY85, 2021-07-27:
Thanks for the help, so far it more or less work with the switch in between so I have to make some time to hook up a screen + keyboard