MichaIng/DietPi #4589: z-wave.me (RPI3 hat) doesn't work

katjo21, 2021-07-26:
### Creating a bug report/issue
z-wave.me (RPI3 hat) doesn't work
on Raspberry Pi 3B+
z-wave controller - RaZberry — a shield for Raspberry 

#### Required Information
- DietPi version
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=4
G_DIETPI_VERSION_RC=2
G_GITBRANCH='beta'
G_GITOWNER='MichaIng'

- Distro version: buster 10.10
- Kernel version: Linux dpi3 5.10.52-v8+ #1439 SMP PREEMPT Thu Jul 22 15:43:49 BST 2021 aarch64 GNU/Linux
- SBC model: RPi 3 Model B (aarch64)
- Power supply used 5V 2.5A
- SDcard used  SanDisk 32GB

#### Additional Information (if applicable)
- Software title: z-wave.me ( RaZberry )
- installed freshly by instrunctions https://z-wave.me/z-way/download-z-way/
- This issue be replicated: wget -q -O - https://storage.z-wave.me/RaspbianInstall | sudo bash
- Bug report ID: 34297980-405f-4bbf-b2f1-70df747b3a7d

#### Steps to reproduce

1.  Fresh installed dietpi
2.  wget -q -O - https://storage.z-wave.me/RaspbianInstall | sudo bash
3. Resilt:
Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'https://repo.z-wave.me/z-way/raspbian buster InRelease' doesn't support architecture 'arm64'

#### Expected behaviour
witouyt errors

#### Actual behaviour
dietpi@dietpidpi:~ $ wget -q -O - https://storage.z-wave.me/RaspbianInstall | sudo bash
Hit:1 https://deb.debian.org/debian buster InRelease
Hit:2 https://deb.debian.org/debian buster-updates InRelease
Hit:3 https://repo.z-wave.me/z-way/raspbian buster InRelease
Hit:4 https://deb.debian.org/debian-security buster/updates InRelease
Hit:5 https://deb.debian.org/debian buster-backports InRelease
Hit:6 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists... Done
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'https://repo.z-wave.me/z-way/raspbian buster InRelease' doesn't support architecture 'a                         rm64'
Reading package lists... Done
Building dependency tree
Reading state information... Done
apt-transport-https is already the newest version (1.8.2.3).
dirmngr is already the newest version (2.2.12-1+deb10u1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Executing: /tmp/apt-key-gpghome.U1lA2jqEvQ/gpg.1.sh --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 0x7E148E3C
gpg: key 2222290F7E148E3C: "Z-Wave.Me <info@z-wave.me>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
Hit:1 https://deb.debian.org/debian buster InRelease
Hit:2 https://deb.debian.org/debian buster-updates InRelease
Hit:3 https://deb.debian.org/debian-security buster/updates InRelease
Hit:4 https://archive.raspberrypi.org/debian buster InRelease
Hit:5 https://repo.z-wave.me/z-way/raspbian buster InRelease
Hit:6 https://deb.debian.org/debian buster-backports InRelease
Reading package lists... Done
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'https://repo.z-wave.me/z-way/raspbian buster InRelease' doesn't support architecture 'arm64'
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package z-way-full
E: Unable to locate package z-way-server
E: Unable to locate package zbw
E: Unable to locate package webif

# Extra
Installeted od Raspbian OS Lite without problem..
Software work perfect on the same hardware.

Joulinar, 2021-07-26:
Well this is not on issue of DietPi. You are running 64bit OS version while the software you try to install did not offer a 64bit version

MichaIng, 2021-07-26:
Right, you could use our 32-bit RPi image, which is based on Raspberry Pi OS Lite (32-bit), then this works. Otherwise you may want to ask z-wave to provide `arm64` packages via their repository. The Raspberry Pi OS Lite (64-bit) is still not officially released, but it will use Debian as primary repository, so it's `binary-arm64` builds missing here: https://repo.z-wave.me/z-way/debian/dists/buster/main/
Or better for Bullseye directly, which is released next month.

katjo21, 2021-07-26:
That's true..
Raspbian IS Lite was 32 bit.
I will ask for 64 bit version z-wave software.
Thanks
Jozef

Joulinar, 2021-07-26:
OK I'm going to close this issue as there is nothing we can do from our end 😉

closed 2021-07-26 by Joulinar

katjo21, 2021-07-28:
Can i ask you ?
- mongo is not in repository dietpi ?
Thanks
Jozef

Joulinar, 2021-07-28:
Correct, Mongo is not available as installation option. Furthermore Mongo themselves did not support `aarch64` https://github.com/MichaIng/DietPi/issues/4532

MichaIng, 2021-07-28:
You could pick up what I suggested on that issue to track down the dependency issue with the focal package. Chances are probably better with the Ubuntu bionic builds: http://repo.mongodb.org/apt/ubuntu/dists/bionic/mongodb-org/5.0/multiverse/binary-arm64/

katjo21, 2021-07-28:
Thank for your goodwill Michal
I will try
Jozef