MichaIng/DietPi #4595: password in clear text in logfile

timoteicampian, 2021-07-26:
Dear Support Team,

just notice the password is logged in clear text in /var/tmp/dietpi/logs/dietpi-firstrun-setup.log
shouldn't be better hashed or hidden?
(also, a small typo: Password has been **_ecrypted_** and saved to rootfs /boot/dietpi.txt)

thank you/best regards

```
Mode: Applying initial first run setup steps

[      ] DietPi-Software | sed -i /^[[:blank:]]*AUTO_SETUP_GLOBAL_PASSWORD=/c\#AUTO_SETUP_GLOBAL_PASSWORD= # Password has
[  OK  ] DietPi-Software | sed -i /^[[:blank:]]*AUTO_SETUP_GLOBAL_PASSWORD=/c\#AUTO_SETUP_GLOBAL_PASSWORD= # Password has been ecrypted and saved to rootfs /boot/dietpi.txt
[ SUB1 ] DietPi-Set_software > passwords (MYPassword)
[  OK  ] DietPi-Set_software | Comment in /boot/dietpi.txt converted to setting: AUTO_SETUP_GLOBAL_PASSWORD=Password has been encrypted and secured on rootFS
[  OK  ] DietPi-Set_software | mkdir -p /var/lib/dietpi/dietpi-software
[  OK  ] DietPi-Set_software | chown root:root /var/lib/dietpi/dietpi-software/.GLOBAL_PW.bin
[  OK  ] DietPi-Set_software | chmod 0600 /var/lib/dietpi/dietpi-software/.GLOBAL_PW.bin
[ INFO ] DietPi-Set_software | Global software password successfully changed
[ INFO ] DietPi-Set_software | "root" and "dietpi" login passwords successfully changed
[  OK  ] passwords MYPassword | Completed
```

MichaIng, 2021-07-26:
Many thanks for your report.

Indeed this should not be the case. Also dietpi.txt is written doubled, unnecessarily, as dietpi-software and dietpi-set_software both always replace the setting, both in a different way and regardless whether the setting has been nullified already or not.

First fix: https://github.com/MichaIng/DietPi/commit/8053cf8ed420fcf7804e1659c65565a0beb16588
Second fix: https://github.com/MichaIng/DietPi/commit/7d648985a9dd491fc9a149dde0c9c4cb0ed960c7

timoteicampian, 2021-07-27:
many thanks for the quick fix !!

closed 2021-07-27 by timoteicampian