MichaIng/DietPi #4596: NanoPC-T4 cpu can't up to 2.0g

keeno1982, 2021-07-27:
DietPi v7.4.2
Device model : NanoPi M4/T4/NEO4 (aarch64)

I set governor=performance, but cpu still only 1.8g max, how can I do ?

MichaIng, 2021-07-27:
Many thanks for your question.

The GPU governor has no effect on the maximum frequency, it only decides if/when to change it in the set min/max borders. I remember for the RK3399 SoCs there was a device tree overlay available on newest kernel to change the max frequency. Can you check:
```sh
ls -l /boot/dtb/rockchip/overlay
```

keeno1982, 2021-07-27:
@MichaIng
Thanks for your reply, info print below, and then what should I do next?
 
-rwxr-xr-x 1 root root 2921 3月   8 09:05 README.rockchip-overlays
-rwxr-xr-x 1 root root  272 3月   8 09:05 rockchip-dwc3-0-host.dtbo
-rwxr-xr-x 1 root root 2698 3月   8 09:05 rockchip-fixup.scr
-rwxr-xr-x 1 root root  218 3月   8 09:05 rockchip-i2c7.dtbo
-rwxr-xr-x 1 root root  218 3月   8 09:05 rockchip-i2c8.dtbo
-rwxr-xr-x 1 root root  267 3月   8 09:05 rockchip-pcie-gen2.dtbo
-rwxr-xr-x 1 root root  301 3月   8 09:05 rockchip-rk3328-opp-1.4ghz.dtbo
-rwxr-xr-x 1 root root  301 3月   8 09:05 rockchip-rk3328-opp-1.5ghz.dtbo
-rwxr-xr-x 1 root root  376 3月   8 09:05 rockchip-rk3399-opp-2ghz.dtbo
-rwxr-xr-x 1 root root 1314 3月   8 09:05 rockchip-spi-jedec-nor.dtbo
-rwxr-xr-x 1 root root 1266 3月   8 09:05 rockchip-spi-spidev.dtbo
-rwxr-xr-x 1 root root  384 3月   8 09:05 rockchip-uart4.dtbo
-rwxr-xr-x 1 root root  491 3月   8 09:05 rockchip-w1-gpio.dtbo

MichaIng, 2021-07-27:
Please try the following:
```sh
sed -i '/^overlays=/s/$/ rk3399-opp-2ghz/' /boot/armbianEnv.txt
```
This adds the RK3399 2 GHz overclocking device tree overlay. Switch the governor back to ondemand or better schedutil, so that it does not run at 2 GHz even on idle (performance governor).

keeno1982, 2021-07-28:
Thanks very much, it's worked on schedutil stable !

MichaIng, 2021-07-28:
Great 🙂, I'll mark the issue as closed then. Feel free to reopen if required.

keeno1982, 2022-05-09:
In the same way, overclocking not work on Bullseye : (

MichaIng, 2022-05-10:
Oh, that is nasty, I'll try to replicate on NanoPi R4S.

MichaIng, 2022-05-15:
I verified that overclocking doesn't work on NanoPi R4S either, so it seems to affect all RK3399 SBCs with recent kernel v5.15 😞.

MichaIng, 2023-08-27:
Solved with the Linux 6.1 upgrade applied with DietPi v8.21. At least on NanoPi R4S, overclocking works now well with these overlays 🙂.

closed 2023-08-27 by MichaIng