MichaIng/DietPi #4597: qBittorrent not working in v7.4

bt61, 2021-07-27:
I've been using DietPi for several years, qBittorrent has worked perfectly in all recent versions I can recall, including 7.3 for a few months until today.  I just upgraded to v7.4.2 and saw the web gui hadn't loaded. Checked htop and no qBittorrent in the processes.  

Tried:
- Rebooted, still no qBittorrent.  
- Uninstalled and reinstalled through dietpi-software, still not loading web gui or showing process in htop.  
- Tried 'service qbittorrent restart' and then quickly htop and I can see the qbittorrent process there for a few seconds before disappearing.

GvY85, 2021-07-27:
Maybe not of direct help but qBittorrent-nox is running fine here with v7.4.2

MichaIng, 2021-07-27:
Many thanks for your report.

Please paste the logs:
```sh
journalctl -u qbittorrent
tail -20 /var/log/qbittorrent/*
```

bt61, 2021-07-27:
@Michalng thank you for the reply
The "tail -20 /var/log/qbittorrent/*" line doesn't seem to do anything, here's the output from the first line:

```
root@DietPi:~# journalctl -u qbittorrent
-- Logs begin at Tue 2021-07-27 11:11:50 +08, end at Tue 2021-07-27 18:00:01 +08. --
Jul 27 11:12:39 DietPi systemd[1]: Started qBittorrent (DietPi).
Jul 27 11:12:44 DietPi qbittorrent-nox[1728]: *************************************************
************
Jul 27 11:12:44 DietPi qbittorrent-nox[1728]: Please file a bug report at http://bug.qbittorren
t.org and provide the following information:
Jul 27 11:12:44 DietPi qbittorrent-nox[1728]: qBittorrent version: v4.1.5
Jul 27 11:12:44 DietPi qbittorrent-nox[1728]: Caught signal: SIGSEGV
Jul 27 11:12:44 DietPi qbittorrent-nox[1728]: Stack trace:
Jul 27 11:12:44 DietPi qbittorrent-nox[1728]: There were no function names found in the stack t
race
Jul 27 11:12:44 DietPi qbittorrent-nox[1728]: .Seems like debug symbols are not installed, and 
the stack trace is useless.
Jul 27 11:12:44 DietPi systemd[1]: qbittorrent.service: Main process ex
ited, code=killed, status=11/SEGV
Jul 27 11:12:44 DietPi systemd[1]: qbittorrent.service: Failed with res
ult 'signal'.
Jul 27 11:20:13 DietPi systemd[1]: Started qBittorrent (DietPi).
Jul 27 11:20:17 DietPi qbittorrent-nox[4553]: *************************************************
************
Jul 27 11:20:17 DietPi qbittorrent-nox[4553]: Please file a bug report at http://bug.qbittorren
t.org and provide the following information:
Jul 27 11:20:17 DietPi qbittorrent-nox[4553]: qBittorrent version: v4.1.5
Jul 27 11:20:17 DietPi qbittorrent-nox[4553]: Caught signal: SIGSEGV
Jul 27 11:20:17 DietPi qbittorrent-nox[4553]: Stack trace:
Jul 27 11:20:17 DietPi qbittorrent-nox[4553]: There were no function names found in the stack t
race
Jul 27 11:20:17 DietPi qbittorrent-nox[4553]: .Seems like debug symbols are not installed, and 
the stack trace is useless.
Jul 27 11:20:17 DietPi systemd[1]: qbittorrent.service: Main process ex
ited, code=killed, status=11/SEGV
Jul 27 11:20:17 DietPi systemd[1]: qbittorrent.service: Failed with res
ult 'signal'.
Jul 27 11:20:40 DietPi systemd[1]: Started qBittorrent (DietPi).
Jul 27 11:20:44 DietPi qbittorrent-nox[4635]: *************************************************
************
Jul 27 11:20:44 DietPi qbittorrent-nox[4635]: Please file a bug report at http://bug.qbittorren
t.org and provide the following information:
Jul 27 11:20:44 DietPi qbittorrent-nox[4635]: qBittorrent version: v4.1.5
Jul 27 11:20:44 DietPi qbittorrent-nox[4635]: Caught signal: SIGSEGV
Jul 27 11:20:44 DietPi qbittorrent-nox[4635]: Stack trace:
Jul 27 11:20:44 DietPi qbittorrent-nox[4635]: There were no function names found in the stack t
race
Jul 27 11:20:44 DietPi qbittorrent-nox[4635]: .Seems like debug symbols are not installed, and 
the stack trace is useless.
Jul 27 11:20:44 DietPi systemd[1]: qbittorrent.service: Main process ex
ited, code=killed, status=11/SEGV
Jul 27 11:20:44 DietPi systemd[1]: qbittorrent.service: Failed with res
ult 'signal'.
Jul 27 11:21:36 DietPi systemd[1]: Started qBittorrent (DietPi).
Jul 27 11:21:39 DietPi qbittorrent-nox[4776]: *************************************************
************
Jul 27 11:21:39 DietPi qbittorrent-nox[4776]: Please file a bug report at http://bug.qbittorren
t.org and provide the following information:
Jul 27 11:21:39 DietPi qbittorrent-nox[4776]: qBittorrent version: v4.1.5
Jul 27 11:21:39 DietPi qbittorrent-nox[4776]: Caught signal: SIGSEGV
Jul 27 11:21:39 DietPi qbittorrent-nox[4776]: Stack trace:
Jul 27 11:21:39 DietPi qbittorrent-nox[4776]: There were no function names found in the stack t
race
Jul 27 11:21:39 DietPi qbittorrent-nox[4776]: .Seems like debug symbols are not installed, and 
the stack trace is useless.
Jul 27 11:21:39 DietPi systemd[1]: qbittorrent.service: Main process ex
ited, code=killed, status=11/SEGV
Jul 27 11:21:39 DietPi systemd[1]: qbittorrent.service: Failed with res
ult 'signal'.
```

MichaIng, 2021-07-27:
A SIGSEGV (segmentation fault), which means an issue with the binary itself 🤔.

Can you try to purge and reinstall it, and check if the kernel reports any error when doing that:
```sh
apt purge --autoremove -y qbittorrent-nox
apt install -y qbittorrent-nox
ls -l "$(which qbittorrent-nox)"
dmesg | tail -5
```

bt61, 2021-07-27:
That seems to have resolved it, thank you!  I have no idea the cause, I'm a novice and this is mostly greek to me but here's the result:
```
root@DietPi:~# apt purge --autoremove -y qbittorrent-nox
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be REMOVED:
  libboost-chrono1.67.0* libboost-random1.67.0* libboost-system1.67.0*
  libdouble-conversion1* libpcre2-16-0* libqt5core5a* libqt5dbus5* libqt5network5*
  libqt5xml5* libtorrent-rasterbar9* qbittorrent-nox*
0 upgraded, 0 newly installed, 11 to remove and 0 not upgraded.
After this operation, 22.0 MB disk space will be freed.
(Reading database ... 70967 files and directories currently installed.)
Removing qbittorrent-nox (4.1.5-1+deb10u1) ...
Removing libtorrent-rasterbar9 (1.1.11-2) ...
Removing libboost-chrono1.67.0:armhf (1.67.0-13+deb10u1) ...
Removing libboost-random1.67.0:armhf (1.67.0-13+deb10u1) ...
Removing libboost-system1.67.0:armhf (1.67.0-13+deb10u1) ...
Removing libqt5xml5:armhf (5.11.3+dfsg1-1+deb10u4) ...
Removing libqt5network5:armhf (5.11.3+dfsg1-1+deb10u4) ...
Removing libqt5dbus5:armhf (5.11.3+dfsg1-1+deb10u4) ...
Removing libqt5core5a:armhf (5.11.3+dfsg1-1+deb10u4) ...
Removing libdouble-conversion1:armhf (3.1.0-3) ...
Removing libpcre2-16-0:armhf (10.32-5) ...
Processing triggers for libc-bin (2.28-10) ...
(Reading database ... 70889 files and directories currently installed.)
Purging configuration files for libqt5core5a:armhf (5.11.3+dfsg1-1+deb10u4) ...
root@DietPi:~# apt install -y qbittorrent-nox
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libboost-chrono1.67.0 libboost-random1.67.0 libboost-system1.67.0
  libdouble-conversion1 libpcre2-16-0 libqt5core5a libqt5dbus5 libqt5network5 libqt5xml5
  libtorrent-rasterbar9
Suggested packages:
  libthai0 libtorrent-rasterbar-dbg qbittorrent-dbg
Recommended packages:
  qttranslations5-l10n
The following NEW packages will be installed:
  libboost-chrono1.67.0 libboost-random1.67.0 libboost-system1.67.0
  libdouble-conversion1 libpcre2-16-0 libqt5core5a libqt5dbus5 libqt5network5 libqt5xml5
  libtorrent-rasterbar9 qbittorrent-nox
0 upgraded, 11 newly installed, 0 to remove and 0 not upgraded.
Need to get 9,469 kB of archives.
After this operation, 22.0 MB of additional disk space will be used.
Get:1 https://deb.debian.org/debian buster/main armhf libboost-system1.67.0 armhf 1.67.0-13+deb10u1 [229 kB]
Get:2 https://deb.debian.org/debian buster/main armhf libboost-chrono1.67.0 armhf 1.67.0-13+deb10u1 [231 kB]
Get:3 https://deb.debian.org/debian buster/main armhf libboost-random1.67.0 armhf 1.67.0-13+deb10u1 [230 kB]
Get:4 https://deb.debian.org/debian buster/main armhf libdouble-conversion1 armhf 3.1.0-3 [39.0 kB]
Get:5 https://deb.debian.org/debian buster/main armhf libpcre2-16-0 armhf 10.32-5 [179 kB]
Get:6 https://deb.debian.org/debian buster/main armhf libqt5core5a armhf 5.11.3+dfsg1-1+deb10u4 [1,684 kB]
Get:7 https://deb.debian.org/debian buster/main armhf libqt5dbus5 armhf 5.11.3+dfsg1-1+deb10u4 [205 kB]
Get:8 https://deb.debian.org/debian buster/main armhf libqt5network5 armhf 5.11.3+dfsg1-1+deb10u4 [587 kB]
Get:9 https://deb.debian.org/debian buster/main armhf libqt5xml5 armhf 5.11.3+dfsg1-1+deb10u4 [115 kB]
Get:10 https://deb.debian.org/debian buster/main armhf libtorrent-rasterbar9 armhf 1.1.11-2 [1,168 kB]
Get:11 https://deb.debian.org/debian buster/main armhf qbittorrent-nox armhf 4.1.5-1+deb10u1 [4,804 kB]
Fetched 9,469 kB in 1min 5s (146 kB/s)                                                   
Selecting previously unselected package libboost-system1.67.0:armhf.
(Reading database ... 70889 files and directories currently installed.)
Preparing to unpack .../00-libboost-system1.67.0_1.67.0-13+deb10u1_armhf.deb ...
Unpacking libboost-system1.67.0:armhf (1.67.0-13+deb10u1) ...
Selecting previously unselected package libboost-chrono1.67.0:armhf.
Preparing to unpack .../01-libboost-chrono1.67.0_1.67.0-13+deb10u1_armhf.deb ...
Unpacking libboost-chrono1.67.0:armhf (1.67.0-13+deb10u1) ...
Selecting previously unselected package libboost-random1.67.0:armhf.
Preparing to unpack .../02-libboost-random1.67.0_1.67.0-13+deb10u1_armhf.deb ...
Unpacking libboost-random1.67.0:armhf (1.67.0-13+deb10u1) ...
Selecting previously unselected package libdouble-conversion1:armhf.
Preparing to unpack .../03-libdouble-conversion1_3.1.0-3_armhf.deb ...
[... 19 lines cut ...]
Selecting previously unselected package qbittorrent-nox.
Preparing to unpack .../10-qbittorrent-nox_4.1.5-1+deb10u1_armhf.deb ...
Unpacking qbittorrent-nox (4.1.5-1+deb10u1) ...
Setting up libpcre2-16-0:armhf (10.32-5) ...
Setting up libboost-system1.67.0:armhf (1.67.0-13+deb10u1) ...
Setting up libdouble-conversion1:armhf (3.1.0-3) ...
Setting up libboost-chrono1.67.0:armhf (1.67.0-13+deb10u1) ...
Setting up libqt5core5a:armhf (5.11.3+dfsg1-1+deb10u4) ...
Setting up libqt5dbus5:armhf (5.11.3+dfsg1-1+deb10u4) ...
Setting up libboost-random1.67.0:armhf (1.67.0-13+deb10u1) ...
Setting up libtorrent-rasterbar9 (1.1.11-2) ...
Setting up libqt5network5:armhf (5.11.3+dfsg1-1+deb10u4) ...
Setting up libqt5xml5:armhf (5.11.3+dfsg1-1+deb10u4) ...
Setting up qbittorrent-nox (4.1.5-1+deb10u1) ...
Processing triggers for libc-bin (2.28-10) ...
root@DietPi:~# ls -l "$(which qbittorrent-nox)"
-rwxr-xr-x 1 root root 5825488 Mar 30  2020 /usr/bin/qbittorrent-nox
root@DietPi:~# dmesg | tail -5
[   31.501265] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts: (null)
[  332.771730] EXT4-fs (sda1): error count since last fsck: 34
[  332.775945] EXT4-fs (sda1): initial error at time 1599232597: ext4_wait_block_bitmap:532
[  332.784018] EXT4-fs (sda1): last error at time 1599305712: ext4_journal_check_start:61: inode 28185589: block 112722271
[ 2199.454740] hrtimer: interrupt took 508209 ns
root@DietPi:~# 
```

MichaIng, 2021-07-27:
> error count since last fsck: 34

So there are at least filesystem errors, which may or may not be related. But you should fix them better earlier then later:
```sh
> /forcefsck
reboot
# after reboot, check fsck logs
cat /run/initramfs/fsck.log
```

bt61, 2021-07-27:
Thank you @MichaIng , I did as you said, this looks good now, right?
```
root@DietPi:~# cat /run/initramfs/fsck.log
Log of fsck -C -y -V -t ext4 /dev/mmcblk1p2 
Tue Jul 27 13:29:26 2021

fsck from util-linux 2.33.1
[/sbin/fsck.ext4 (1) -- /dev/mmcblk1p2] fsck.ext4 -y -C0 /dev/mmcblk1p2 
e2fsck 1.44.5 (15-Dec-2018)
rootfs: clean, 406914/3863552 files, 14544090/30902784 blocks

Tue Jul 27 13:29:27 2021
```

MichaIng, 2021-07-27:
Yes looks good. What does qbittorrent say now? For easier testing, when the service is not currently running, simply run it from console:
```sh
qbittorrent-nox
```

bt61, 2021-07-27:
All good with qbittorrent now, working as normal, thanks again1

MichaIng, 2021-07-27:
Great. I'll mark the issue as closed then. Feel free to reopen of required.

closed 2021-07-27 by MichaIng