MichaIng/DietPi #4598: No Portainer Documentation on https://dietpi.com/dietpi-software.html

psi5asp, 2021-07-27:
This may not be a bug, however since Portainer is not standardly install on Port 9000 for dietpi, wouldn't it be appropriate to state that in the https://dietpi.com/dietpi-software.html page?

Portainer isn't even listed in there....

Joulinar, 2021-07-27:
Hi,

It is clearly stated in our docs. https://dietpi.com/docs/software/programming/#portainer.

> The web interface is accessible via port 9002:

We even explain the why

> Logitech Media Server already listened to port 9000, and this is why Portainer has been configured to start using port 9002. For more details on the implementation Portainer in DietPi see the GitHub task: MichaIng/DietPi!3933

MichaIng, 2021-07-27:
There are many software titles missing on https://dietpi.com/dietpi-software.html. We're still not sure how to proceed with this page, whether to keep it or refer to the fully fledged software docs directly: https://dietpi.com/docs/software/

We are aware of the issue but so far noone found time to address it:
- https://github.com/MichaIng/DietPi-Website/issues/13
- https://github.com/MichaIng/DietPi-Website/issues/20

psi5asp, 2021-08-02:
I vote to keep the page - its a **_software list_**, with important hightlights (easy reference/feature page/guide) which has a quick summary of important info (similar to a cheat sheet). 

Anyone can skim through the list of available software, and decide what's available., rather than having to initially read all the details - details come later, and already covered in the **_docs_** page.