MichaIng/DietPi #4599: Can't Update Diet-Pi

ALittleOfEverything88, 2021-07-27:
#### Details:
- Date           | Tue Jul 27 15:39:24 EDT 2021
- Bug report     | 542039a9-6901-4234-aaae-041191e5b7dd
- DietPi version | v7.1.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi Zero W (armv6l) (ID=1)
- Kernel version | Linux PiHoleDR 5.10.17+ #1414 Fri Apr 30 13:16:27 BST 2021 armv6l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. Run dietpi-update
2. Attempt to update from 7.1.2 to 7.4.2
3. Below error received. Update does not complete successfully.
#### Expected behaviour:

- Device should update without the below error
#### Actual behaviour:

- Device does not update and throws the below error
#### Extra details:

- (None)
#### Additional logs:
```
dpkg-deb: error: tar subprocess was killed by signal (Segmentation fault)
dpkg: error processing archive /var/cache/apt/archives/base-files_10.3+rpi1+deb10u10_armhf.deb (--unpack):
 dpkg-deb --control subprocess returned error exit status 2
Errors were encountered while processing:
 /var/cache/apt/archives/base-files_10.3+rpi1+deb10u10_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Joulinar, 2021-07-27:
Hi,

You have a `Segmentation fault` during `apt upgrade` process. This could be some file system corruption. Can you try following

```
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade
```

ALittleOfEverything88, 2021-07-27:
Same error.

Joulinar, 2021-07-27:
Can you try following

```
> /forcefsck
reboot
# after reboot, check fsck logs
cat /run/initramfs/fsck.log
```

ALittleOfEverything88, 2021-07-27:
After some quick googling, I learned I needed to run:
`touch /forcefsck`
and reboot. Or so I thought. Because after the reboot I ran
`cat /run/initramfs/fsck.log`
and received
`cat: /run/initramfs/fsck.log: No such file or directory`

How do I run the requested command?

Joulinar, 2021-07-27:
What about `journalctl -t systemd-fsck
`

ALittleOfEverything88, 2021-07-27:
`-- Logs begin at Tue 2021-07-27 17:03:35 EDT, end at Tue 2021-07-27 17:10:05 EDT. --
Jul 27 17:03:36 PiHoleDR systemd-fsck[97]: Please pass 'fsck.mode=force' on the kernel command l
ine rather than creating /forcefsck on the root file system.
Jul 27 17:03:37 PiHoleDR systemd-fsck[97]: e2fsck 1.44.5 (15-Dec-2018)
Jul 27 17:03:37 PiHoleDR systemd-fsck[97]: Pass 1: Checking inodes, blocks, and sizes
Jul 27 17:04:12 PiHoleDR systemd-fsck[97]: Pass 2: Checking directory structure
Jul 27 17:04:16 PiHoleDR systemd-fsck[97]: Pass 3: Checking directory connectivity
Jul 27 17:04:16 PiHoleDR systemd-fsck[97]: Pass 4: Checking reference counts
Jul 27 17:04:18 PiHoleDR systemd-fsck[97]: Pass 5: Checking group summary information
Jul 27 17:04:21 PiHoleDR systemd-fsck[97]: rootfs: 28101/1799520 files (0.5% non-contiguous), 1278523/7526400 blocks
Jul 27 17:04:29 PiHoleDR systemd-fsck[184]: Please pass 'fsck.mode=force' on the kernel command
line rather than creating /forcefsck on the root file system.
Jul 27 17:04:30 PiHoleDR systemd-fsck[184]: fsck.fat 4.1 (2017-01-24)
Jul 27 17:04:30 PiHoleDR systemd-fsck[184]: /dev/mmcblk0p1: 327 files, 103414/516191 clusters`

MichaIng, 2021-07-27:
On RPi, since by default no initramfs is installed, it's `journalctl -t systemd-fsck`, on all other systems an initramfs is installed, so there it's `cat /run/initramfs/fsck.log` 🙂. The fsck run in the initramfs sets a flag file which prevents `systemd-fsck` from running, to avoid doubled fsck.

Does it fail as well when manually installing that package?
```sh
cd /tmp
curl -sSfLO http://raspbian.raspberrypi.org/raspbian/pool/main/b/base-files/base-files_10.3+rpi1+deb10u10_armhf.deb
dpkg -i base-files_10.3+rpi1+deb10u10_armhf.deb
rm base-files_10.3+rpi1+deb10u10_armhf.deb
```

ALittleOfEverything88, 2021-07-27:
`-i base-files_10.-bash: http://raspbian.raspberrypi.org/raspbian/pool/main/b/base-files/base-files_10.3+rpi1+deb10u10_armhf.deb: No such file or directory
3+rpi1+deb10u10_armhf.deb
rm base-files_10.3+rpi1+deb10u10_armhf.deb
curl: no URL specified!
curl: try 'curl --help' or 'curl --manual' for more information
root@PiHoleDR:/tmp# dpkg -i base-files_10.3+rpi1+deb10u10_armhf.deb
dpkg: error: cannot access archive 'base-files_10.3+rpi1+deb10u10_armhf.deb': No such file or directory
root@PiHoleDR:/tmp# rm base-files_10.3+rpi1+deb10u10_armhf.deb
rm: cannot remove 'base-files_10.3+rpi1+deb10u10_armhf.deb': No such file or directory`

ALittleOfEverything88, 2021-07-27:
Removed the extra `'s from the URL and the file downloaded successfully. Ran the dpkg command and got the following:

root@PiHoleDR:~# dpkg -i base-files_10.3+rpi1+deb10u10_armhf.deb
dpkg-deb: error: tar subprocess was killed by signal (Segmentation fault)
dpkg: error processing archive base-files_10.3+rpi1+deb10u10_armhf.deb (--install):
 dpkg-deb --control subprocess returned error exit status 2
Errors were encountered while processing:
 base-files_10.3+rpi1+deb10u10_armhf.deb

Joulinar, 2021-07-27:
Probably `dpkg` is faulty 🤔

ALittleOfEverything88, 2021-07-27:
I feel like I'm getting close to the point where this is probably an SD Card issue and I'll need to start again. yes?

Joulinar, 2021-07-27:
Probably. But let's wait on @MichaIng . Usually he has good ideas on such issues.

MichaIng, 2021-07-27:
> Removed the extra `'s from the URL

Jep sorry, wrong quote characters.

So it's the tarball extractor of dpkg. I guess it's `tar` itself. Let's try it out on a small tarball:
```sh
cd /tmp
# now with right single quote characters :)
curl -sSfLO 'http://raspbian.raspberrypi.org/raspbian/pool/main/t/tar/tar_1.30+dfsg-6.debian.tar.xz'
tar xf tar_1.30+dfsg-6.debian.tar.xz
rm tar_1.30+dfsg-6.debian.tar.xz
rm -R debian
```

ALittleOfEverything88, 2021-07-27:
root@PiHoleDR:/tmp# tar xf tar_1.30+dfsg-6.debian.tar.xz

Segmentation fault

MichaIng, 2021-07-28:
Okay, sadly `tar` is needed to extract the new tar binary 😄. Here an upload, extracted from http://raspbian.raspberrypi.org/raspbian/pool/main/t/tar/tar_1.30+dfsg-6_armhf.deb:
- [tar.zip](https://github.com/MichaIng/DietPi/files/6892094/tar.zip) (It is not a zip, but GitHub only allows certain file endings 😉)
```sh
curl -sSfL 'https://github.com/MichaIng/DietPi/files/6892094/tar.zip' -o /bin/tar
apt upgrade
```

Joulinar, 2021-07-28:
> `tar` is needed to extract the new tar

This sounds like a chicken and egg challenge. 🤣

ALittleOfEverything88, 2021-07-28:
Ok.

Ran those commands and the `apt update` completed successfully.

Reran the `dietpi-upgrade` and it appears everything upgraded successfully.

I'm going to make a backup of the card in case it's starting to die.

Thanks all!

Joulinar, 2021-07-28:
I'm going to close the issue. Feel free to reopen if needed.

closed 2021-07-28 by Joulinar