MichaIng/DietPi #4600: apt update fails with "Redirection from https is forbidden"

numanair, 2021-07-27:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=4
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

Was on v7.3.2

- Distro version | buster
- Kernel version | `Linux rock64 5.10.43-rockchip64 #21.05.4 SMP PREEMPT Wed Jun 16 08:02:12 UTC 2021 aarch64 GNU/Linux`
- SBC model | `ROCK64 (aarch64)`
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | apt
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

sudo apt update

#### Expected behaviour

sources update successfully

#### Actual behaviour

Fails with 

> Redirection from https to 'http://armbian.tnahosting.net/apt/dists/buster/Release' is forbidden [IP: 173.45.128.12 443]

#### Extra details

My other SBC, a pi3B+ appears to use different sources. Hence it does not have this issue.

### Fix:
To fix it, I changed /etc/apt/sources.list.d/armbian.list from https to http. Hopefully this is the correct fix?

> deb http://apt.armbian.com/ buster main buster-utils buster-desktop

Joulinar, 2021-07-27:
Hi,

The RPi is using Raspberry OS and therefore a different sources list compare to your Rock64, which is using Arabian OS. This might be a temp problem at Armbian side. You could switch back to https in a day or 2 to check if this is still present. But at least nothing we can do from DietPi side as we do not manage Armbian package server.

MichaIng, 2021-07-27:
It's the Armbian APT mirror service. Strange is that https://armbian.tnahosting.net/apt/ works, so that mirror supports HTTPS as well. Not sure why then the redirect goes to plain HTTP 🤔.

P4tzr1ot, 2021-07-29:
I dont know if this can help in your investigation or if it is linked but i think i have similar issue on raspberry pi 4 not on Rock64[image]

Let me know if its another issue i will create it. 

Thanks

Joulinar, 2021-07-29:
This is a complete different issue. As well, pls try to avoid screen prints and copy content from SSH terminal directly if possible.

P4tzr1ot, 2021-07-29:
ok no problem but you have to tell me something,
i get this error on first boot how i can reach this screen via ssh ?

Joulinar, 2021-07-29:
SSH server is active right from the beginning and should be reachable before first login from another computer using SSH.

MichaIng, 2021-07-30:
@P4tzr1ot 
You have a time sync issue, i.e. the system clock is in the past so that the HTTPS certificates are not "yet" valid, from the systems point of view. Please try to run:
```sh
/boot/dietpi/func/run_ntpd 1
```
from console. Not sure if you are asking how to use SSH in general, i.e. is it clear how to use e.g. [PuTTY](https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html) from Windows?
You can also exit this error prompt (and the next one asking you whether to repeat or exit the first run setup) to reach the local console.

MichaIng, 2021-07-30:
Found it on the Armbian forum: https://forum.armbian.com/topic/18711-update-in-buster-is-failing-with-redirection-from-https-to-httparmbiantnahostingnetaptdistsbusterrelease-is-forbidden/

I hope a redirect which preserves the initial scheme to allow HTTPS connections will be applied quickly.

MichaIng, 2021-08-04:
Issue solved: https://github.com/armbian/dl-router/pull/14

P4tzr1ot, 2021-08-04:
> @P4tzr1ot
>
> You have a time sync issue, i.e. the system clock is in the past so that the HTTPS certificates are not "yet" valid, from the systems point of view. Please try to run:
>
> ```sh [...]

It was NTP parameter 🤦🏻‍♂️ into dietpi.txt it works now. Probably I have deleted it by error. thanks for the clue 👍

swashy69, 2022-05-21:
Ok so I figured it out now.
This one is for the newbies
(Make sure you are root user)
Step 1:-cd /etc/apt/sources.list.d
Step-2:-ls -a     (this is going to show you the files present in the current directory)
Now note down the file name which has its extension as .list , for example:- there was a file named armbian.list in the directory, so I am going to use it.
Step 3:-sudo nano /etc/apt/sources.list.d/YOUR_FILE_NAME.list       (this is going to open up an editor)
Step 4:-rename the https to http
Step 5:- press ctrl+o then enter
Step 6:- press ctrl+x
Step 7:- reboot now
Step 8:- sudo apt update
Fixed.
:)

Joulinar, 2022-05-21:
Did you actually facing an issue with Armbian repository? Because the particular issue should have been fixed by Armbian guy's end of last year already.

MichaIng, 2022-05-21:
No it was never fixed, some strange sort of caching is preventing reliable request protocol detection. The Armbian repo cannot be used with HTTPS currently. However, we change it to HTTP via pre-patch, so with recent DietPi you shouldn't face this issue.

jace, 2025-07-31:
I've been affected by this with a fresh install of Armbian 25.5.1:

```apt
Err:9 https://mirror.albony.in/armbian bookworm Release
  Redirection from https to 'http://mirror.nag.albony.in/armbian/dists/bookworm/Release' is forbidden [IP: 82.180.145.196 443]
Reading package lists... Done
E: The repository 'https://apt.armbian.com bookworm Release' no longer has a Release file.
```

It took a while to realise `mirror.albony.in` was itself a redirect from `apt.armbian.com`. Replacing `https` with `http` in the scheme fixed it.

MichaIng, 2025-07-31:
One of the reasons we replaced the Armbian APT repo with our own. A dezentral distribution and mirror redirection is generally a good idea, but their partners are too unreliable. Our image builds needed to be retriggered multiple times, as 5-10% failed with some random error with an Armbian APT mirror doing a protocol downgrade or simply being offline.

For end users it is usually not such a big problem, may even remain unnoticed, if lists were successfully updated the `apt update` call before and after, also since Armbian keeps old package versions in repos for a long time. But we do error-handle these calls in our scripts, hence it constantly caused bug reports like this one as well.

dejavu21, 2025-12-29:
Types: deb
URIs: https://apt.armbian.com
Suites: trixie
Components: main
Signed-By: /usr/share/keyrings/armbian.gpg