MichaIng/DietPi #4602: Booting from USB causes load average to be 1.0. Fix proposed.

jvteleco, 2021-07-29:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`  
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=4
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='not applied'
- Distro version: buster
- Kernel version: Linux DietPiUK-pihole 5.10.17-v8+ #1421 SMP PREEMPT Thu May 27 14:01:37 BST 2021 aarch64 GNU/Linux
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | Official Raspberry USB-C
- SDcard used | No SD Card.  USB Stick Samsung FIT.

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud). Not related to any software. This is the first time I tried a USB Stick install and the 64 bit version.
- Was the software title installed freshly or updated/migrated? 78   LESP: Nginx    | SQLite  | PHP.   93   Pi-hole.    73   Fail2Ban
- Can this issue be replicated on a fresh installation of DietPi? Not tested. Will try to do it on another USB stick and the 32 bit version.

#### Steps to reproduce
1. Fresh install on a USB Stick, eg: Samsung FIT USB
2. I did install  Software 78, 93 and 73 before checking the "uptime/load average". 

#### Expected behaviour
- Having the raspberry on "idle", load average should be minimum (less than 0.1)

#### Actual behaviour
- uptime/htop shows a **load average of 1.0**. Searching on internet, seems it is a bug of raspbian, when SD card is missing, the kernel frequently tries to scan the SD card slot, which causes high CPU/IO usage.
- I did check the CPU spikes with htop -d 5, but they were none or very rare and short.

#### Extra details
- PROPOSED FIX: Seems to be fixed by adding `dtparam=sd_poll_once` on the last line of `/boot/config.txt`. Would like to propose that it is added on future releases (or at least investigate and consider it when you can).

WITHOUT FIX:
root@pihole:~# uptime
 18:44:27 up 18:27,  2 users,  load average: 1.04, 1.00, 1.00
root@pihole:~# systemctl enable Gunicorn_DashMachine.service

root@pihole:~# nano /boot/config.txt      $add at the EOF:   dtparam=sd_poll_once
root@pihole:~# sudo reboot

WITH FIX:
root@pihole:~# uptime
 19:10:05 up 10 min,  2 users,  load average: 0.02, 0.05, 0.05

Thanks.

Joulinar, 2021-07-29:
Indeed this is a default behaviour of Raspberry OS to constantly search for a SD card even if booted from USB. And DietPi are scripts on top of Raspberry OS, means it's behaving same. 

@MichaIng 
Not sure if this will have a downside if still booting from SD and if we should implement in our image.

MichaIng, 2021-07-30:
Totally makes sense. I wasn't aware that the induced load is THAT extreme and it looks like there must be something done firmware-wise. Especially, since I guess no one inserts an SD card while the Pi is running, so it doesn't make sense at all to continue polling that slot in any frequency near or inducing measurable CPU load.

Where did you find the fix, i.e. do you have some Raspberry Pi forum of GitHub link where this is discussed?

Joulinar, 2021-07-30:
this fit's the behaviour of the blinking green LED https://github.com/raspberrypi/rpi-eeprom/issues/147

I noticed similar on my RPi's booting from USB but simply turned of the LED 🤣

MichaIng, 2021-07-30:
I wonder if polling the SD card is even reasonable when booting from SD card. I mean one does not remove it from a running system, else havoc in any case. So probably we should ship this by default. I'll test if it works fine here when booting from SD card.

https://github.com/raspberrypi/firmware/blob/e83f413/boot/overlays/README#L223
> N.B. Using this option restricts the system to using a single card per boot (or none at all).

Not sure how multiple cards would be used on a Raspberry Pi with a single slot only? 🤣  
I'm pretty sure, when for any reason attaching a USB card reader, it is not affected by this option.

Joulinar, 2021-07-30:
Probably you could insert a SD card while the system is already running and booted from USB. Disabling the polling would make this unusable. But to be honest I never heard somebody was exchanging AD cards or even enter one during operation. Or are there SD cards who will take longer to become available and the polling will ensure them to be mounted correctly? Strange option 🤔

MichaIng, 2021-07-30:
> But to be honest I never heard somebody was exchanging AD cards or even enter one during operation

Same thought here. If you boot from USB already, why the hack would you continue using micro SD cards at all?

MichaIng, 2021-07-30:
Disabled by default from now on (new images): https://github.com/MichaIng/DietPi/commit/e662281151844153235fcb29962737be9765c7b5

Shall we check for any present SD card on next update and in case offer to have the option added? The following should be a reliable command to detect whether any `mmcblk` SD card device is present or not:
```sh
lsblk -nro NAME
```

jvteleco, 2021-07-30:
> Totally makes sense. I wasn't aware that the induced load is THAT extreme and it looks like there must be something done firmware-wise. Especially, since I guess no one inserts an SD card while the [...]

I just started googling High CPU on idle, then added the USB boot and found some threads relating Kworker CPU (https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=293974&p=1775887) 

I finally hit this page: https://jamesachambers.com/raspberry-pi-reclaim-1-cpu-usage-when-ssd-msd-booting/  which references the new overlay dtparam=sd_poll_once,  and the old way with dtoverlay=sdtweak,poll_once=on.

jvteleco, 2021-07-30:
Wow clearly you got it sorted. I hadn´t noticed that this also relates to the LED blinking (I have my rpi in a case). 
Thanks again.

Joulinar, 2021-07-30:
> Shall we check for any present SD card on next update and in case offer to have the option added?

Actually a good idea and an improved for our users.

Joulinar, 2021-08-16:
@MichaIng 
do you still like to include it into `dietpi-update` as option or just on new images?

MichaIng, 2021-08-16:
Done: https://github.com/MichaIng/DietPi/commit/addadeef518336d563370c460e707e96863b2efb

closed 2021-08-20 by MichaIng