MichaIng/DietPi #4603: Ping Failure Through Config, But Works Through Subshell?

seniorm0ment, 2021-07-29:
On a fresh install, Raspberry Pi 4 2GB, using DietPi 64, I installed.
I entered DietPi config. It showed internet failed.
I enabled ethernet, disabled wifi, and then when I went back, it still wasn't working. I tried changing from dhcp to static, still issues.

Note here, that all my other devices have zero issues, furthermore, this ethernet cord was plugged into another device, and that device had zero issues. 
So I figured it was a DNS issue, I entered into the subshell, and tried pinging gentoo.org, and it worked perfectly fine which I don't understand.

So I exit back to the dietpi config, and try changing the command it runs from quad.net9 to gentoo.org, and enter and it says temporary faikure in name resolution, alternative command execution exited with error.

My network does use Pihole, and it is running unbound. Nothing looks to be blocked, everything looks fine.
And again, it literally works when I enter the subshell. So I fail to understand why the DietPi config script is failing.

Joulinar, 2021-07-29:
Hi,

can you share the exact issue. Where does the issue appear? What is the error message. Do you have an issue to ran any software installation or initial setup?

seniorm0ment, 2021-07-29:
>Can you share the exact issue

I did? I said I entered the DietPi config (which is the initial setup). It showed internet failed. It was an icmp failure.

I then tried changing from dhcp to static, didn't fix.

I then opened the subshell, and tried pinging gentoo.org, and it worked.
So I went back to the initial setup config, and tried changing the command it runs to ping gentoo.org instead of the default quad9.
It failed stating temporary failure in name resolution, alternative command execution exited with error.

So it works when i'm in the subshell, it seems to be an issue with the inital setup script.

Not sure how I could be more clear.

Anyways, I don't need to configure network here in theory, however it seems like DietPi wants me to otherwise it won't let me finish setup.

If I just exit the setup, and type exit exit to try to logout, it says DietPi-Logon, root login required, to finish DietPi first run setup, root permissions are required. I click ok, it opens prompt as user dietpi, I sudo su root, then it shows it's testing network connectivity. It fails, and then reopens the DietPi inital setup to test network connectivity which is exactly where I was before.

Is there anyway to just skip this connectivity test seeing it's literally working in the subshell?

seniorm0ment, 2021-07-29:
Ok so, it looks like the loop was being made because autostart automatic login was enabled in config.

Now, if I go into cli and ping gentoo.org it works.
If I run apt update however, it fails.
It states certification failed, the cert is NOT trusted. The certificate chain uses not yet valid cert. Could not handshake: Error in the cert validation IP 151.101.118.132.443

Joulinar, 2021-07-29:
It would be helpful if you could share the exact error messages and the context around like copy the lines from SSH terminal. Just fragments might be misleading.

Usually initial installation will not start automatically and would need to be enabled inside `dietpi.txt`

To install DietPi, you need a valid internet connection, otherwise setup could not be completed. Connection is verified by a `ping` as well as a `DNS` test. Both test will target Quad9 global DNS server. This can be changed inside `dietpi.txt` before first boot, if you like to test against a different IP/DNS. It might be possible Quad9 is blocked by your ISP or something on the network.

The `apt` issue might be outside DietPi. I guess it's failing on global Debian apt repository which is out of our control. Maybe there was a new certificate applied not yet valid?

seniorm0ment, 2021-07-30:
I wasn't on ssh
anyways adjusting the dietpi.txt fixed it. thank you

closed 2021-07-30 by seniorm0ment

Joulinar, 2021-07-30:
Maybe you like to share what values you have set now.

MichaIng, 2021-07-30:
It looks like there were issues on two different levels:
- The ICMP failure with temporary name resolution does not involve any TLS certificate, but the nameserver is not reachable or fails for some reason. It is always helpful to repeat the exact command in subshell/console that is tried by the script (and shown in the error prompt) instead of trying a different one directly, i.e. `ping dns9.quad9.net` in this case. Obviously the previous `ping 9.9.9.9` succeeded (the connection test prior to the name resolution test), which is the default DNS server as well. If you use(d) a different DNS server, like your local Pi-hole, you'd need to check whether the requests from the failing DietPi are actually arriving at your Pi-hole and in case what the response is.
- The APT failure means that in this case, name resolution worked, but the shipped TLS certificate is not trued, e.g. often because the system time was not synced successfully, or because the domain is blocked, like by our Pi-hole or ISP, so that the actually contacted dummy IP does not ship any valid certificate for that domain.

The commands ran by the scripts, `ping`, `curl`, `apt-get` should work inside and outside the same way. There are not special environment variables or command flags set in our scripts, which would affect DNS resolution or certificate handling. If repeating the exact same command the script tries fails there (repeatedly, using "Retry" option) but succeeds in a subshell or any other console, then this would need to be deeper investigated, as this must not be.