MichaIng/DietPi #4604: FAIL2BAN default database location to :memory: to preven excessive writes

GvY85, 2021-07-30:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
While executing 'fatrace -c' I noticed a contieus stream of writes to the FAIL2BAN sqlite3 database. Since I have never looked inside that database I rather have it moved to '/var/log' ramdisk or to ':memory:'  like discribed in the config file

#### Describe the solution you'd like:
Modify the default SQLITE3 database location of FAIL2BAN to `:memory:` instead of the default `/var/lib/fail2ban/fail2ban.sqlite3` 
 to reduce writes to the filesystem

MichaIng, 2021-07-30:
Many thanks for your report.

> I noticed a contieus stream of writes

How continuous do you mean? It should only be written to when an actually failed login was recognised or a banned IP unbanned. So if that is very continuous, then I'd at least also have a look into who or what is so continuously trying to login our SSH or other monitored service.

The actual disk writes should be quite small and mostly written in an accumulated way as of the filesystem cache. You could move it to a tmpfs, like `/tmp/fail2ban.sqlite3` or `/run/fail2ban/fail2ban.sqlite3`, via `dbfile` option in e.g. a new file `/etc/fail2ban/fail2ban.d/99_local.conf`:
```sh
echo -e "[Definition]\ndbfile = /run/fail2ban/fail2ban.sqlite3" > /etc/fail2ban/fail2ban.d/99_local.conf
systemctl restart fail2ban
```
But you'd loose ban states on reboot and fail2ban needs to re-read all logs from the start. You can also disable it completely by using the value `None`, instead of a filename, in which case on a service restart, bans and log file parse states are lost. If it's for SSH only, that shouldn't be any issue indeed.

closed 2021-07-31 by GvY85

Joulinar, 2021-07-31:
@GvY85 
Do you like to share your final solution as you closed the issue without comment. Maybe there are other user having similar issue in future 😉