MichaIng/DietPi #4605: First run setup failed

timoteicampian, 2021-07-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
`buster`
- Kernel version | `uname -a`
`Linux DietPi 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`RPi 4 Model B (armv7l)`
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`
`a9fae084-628e-42fb-a88a-6fcb1cf02ff2`

#### Steps to reproduce

1. just boot the rpi with the changed /boot/dietpi.txt
2. ssh root@ip
3.  this popup appears after ssh login:
```
DietPi-Login 
[FAILED] Unknown install state/First run setup failed on another screen (PID=879). 
You can <Take over> the error handling on this screen and kill the other DietPi-Login session. 
Else you can finish the error handling on the other DietPi-Login session and <Retry> the login script on this session afterwards.
 <Take over> <Retry>
```
 
either  <Take over>  or <Retry> ends in a loop, which start the following popup each 5 seconds:
```
DietPi-Login 
[ INFO ] DietPi first run setup is currently running on another screen (PID=388).
Please resume setup on the active screen.
<Ok>
```
 
#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

Joulinar, 2021-07-30:
Hi,

I guess you activate automatic setup in `dietpi.txt`. Means the setup is running in background even without login into Basically you could disabled the automatic setup to have the setup started from SSH session. This way you should be able to see the error if still present. 

Next to this, can you share following plus

```
/var/tmp/dietpi/logs/dietpi-firstboot.log
/var/tmp/dietpi/logs/dietpi-firstrun-setup.log
/var/tmp/dietpi/logs/fs_partition_resize.log
```

timoteicampian, 2021-07-30:
many thanks for your reply.

[dietpi-firstboot.log](https://github.com/MichaIng/DietPi/files/6906273/dietpi-firstboot.log)
[dietpi-firstrun-setup.log](https://github.com/MichaIng/DietPi/files/6906274/dietpi-firstrun-setup.log)
[fs_partition_resize.log](https://github.com/MichaIng/DietPi/files/6906275/fs_partition_resize.log)

Joulinar, 2021-07-30:
looks like the initial setup was started automatically in background and paused on the question to change password. There a user input was expected but I guess you don't have screen keyboard attached. 

Btw what changes you have inside `dietpi.txt`?

Pls can you try to flash a new image and boot the system without changes done inside `dietpi.txt` done.

timoteicampian, 2021-07-30:
[dietpi.txt](https://github.com/MichaIng/DietPi/files/6907818/dietpi.txt)
this is my changed `dietpi.txt`

it worked without any problems in the previous release (7.3) - tested more than 7 times using the same `dietpi.txt`
now, in 7.4, 2 out of 3 trials went wrong

timoteicampian, 2021-07-30:
`Basically you could disabled the automatic setup to have the setup started from SSH session.`
how can this be done?

Joulinar, 2021-07-30:
Flash a fresh image and keep following on `0`. 

```
AUTO_SETUP_ACCEPT_LICENSE=0
AUTO_SETUP_AUTOMATED=0
```

You should be able to connect via SSH and start the installation right after login as user `root`.

MichaIng, 2021-07-31:
I think the loop when taking over the failed session is resolved with this commit (for future images): https://github.com/MichaIng/DietPi/commit/a8f291caee8f1760020984a385f4831b0c954327

The PID file is intended to be removed via an exit trap, but I think as we use `kill -9` (a hard SIGKILL), that might not always be executed anymore.

So yes, currently leaving `AUTO_SETUP_AUTOMATED=0` is the way to go as `AUTO_SETUP_AUTOMATED=1` will always do the autologin at local terminal, which you cannot see/interact without a screen/keyboard attached.

Joulinar, 2021-08-11:
Yes first setup would need to be done as user `root`

MichaIng, 2021-08-11:
With `AUTO_SETUP_AUTOMATED=1` it should all be done automatically, as autologin is enabled for `root` user then to perform everything. If it fails, we generally need the log file content to find out why:
```
/var/tmp/dietpi/logs/fs_partition_resize.log
/var/tmp/dietpi/logs/dietpi-firstboot.log
/var/tmp/dietpi/logs/dietpi-firstrun-setup.log
```

Taking over a failed first run setup via SSH has been solved with v7.5 but that is not yet true for the current (older images). So in this case do the following:
- Login via SSH **as root user**, if possible (network setup has succeeded)
- Hit `Take over`, then `Cancel` to get to the command prompt
- ```sh
   killall -SIGKILL -w 'dietpi-login'
   rm -f /tmp/.dietpi-login_firstrun_setup*
   /boot/dietpi/dietpi-login
   ```

MichaIng, 2021-09-01:
I'll mark this as closed. Feel free to reopen if the issue persists.

New RPi images should all contain the first run setup fix.

closed 2021-09-01 by MichaIng