MichaIng/DietPi #4606: 'Debian mirror redirection is forbidden' error loop when using PREP_SYSTEM_FOR_DIETPI.sh

wdirksen, 2021-07-30:
I recently was able to get DietPI master installed on a Libre Le Potato using the AML-S905 generic option. The switch from Armbian Buster to DietPi went smoothly except for getting stuck in an APT mirror status failure loop during DietPI-setup which is mandatory at first boot. All attempts to fix it in the network setup settings fail with a 'Debian mirror redirection is forbidden' error. It doesn't matter what is entered there. It was absolutely necessary to fix this by going out of the setup to an external terminal and changing the sources manually. Thanks for that 'bail out' option by the way!

After that experience it seemed to me that the installation experience would have been much improved if the pre-existing Debian sources list from the Armbian installation was purged during the DietPi installation process and replaced with a sane global mirror variant which will always be fully compatible with DietPi when you are finished. Then a user won't get stuck and will also be able to successfully change the mirror in the setup process.

I understand this isn't a bug per se but the setup script was impressive to behold and this issue seems both avoidable and easy to fix

Joulinar, 2021-07-30:
Can you share the source list creating the issues?

wdirksen, 2021-07-30:
It was necessary to fix /etc/sources.list.d/armbian.list to get it to work:

Original from Armbian installation:
deb http://apt.armbian.com buster main buster-utils buster-desktop

I edited it to normal Debian repo variant (in my region):
deb http://ftp.nz.debian.org/debian/ buster main

Joulinar, 2021-07-30:
Don't think this is correct as you are running armbian and I guess source list used for purposes. 

@MichaIng 
Can you have a look

wdirksen, 2021-07-30:
I wondered if /sources.list.d/armbian.list was even necessary to keep and considered removing it. This edit simply worked to get rid of the error and allow DietPi to refresh APT package list

wdirksen, 2021-07-30:
So I unpacked a new Le Potato Armbian image to another SD card then ran
`rm -Rf /etc/sources.list.d`

After that ran PREP_SYSTEM_FOR_DIETPI.sh with no mirror redirection errors !

Joulinar, 2021-07-30:
you simply removed Armbian source list which might be necessary on an Armbian based system. But I guess @MichaIng can answer this way better than I.

wdirksen, 2021-07-30:
> you simply removed Armbian source list which might be necessary on an Armbian based system. But I guess @MichaIng can answer this way better than I.

Yes I know but since it's about to become Diet PI, I assumed it isn't needed and it fixed the problem
Thanks for the great distro!

Joulinar, 2021-07-30:
DietPi is not an own OS. It's a set of shell scripts only. The underlying OS won't change it will be still Armbian. Hence Armbian source list might be needed still.

MichaIng, 2021-07-30:
The Armbian repo is mainly for kernel and firmware, while the Debian repo is required for everything else. So replacing Armbian by Debian doubles the already existing Debian repo entries in `/etc/apt/sources.list` and breaks the ability to receive kernel, bootloader and firmware upgrades.

DietPi-PREP overwrites/re-creates `/etc/apt/sources.list.d/armbian.list`: https://github.com/MichaIng/DietPi/blob/3160a50/PREP_SYSTEM_FOR_DIETPI.sh#L844
So removing it prior to running DietPi-PREP won't have any effect, unless, probably that step failed when the `/etc/apt/sources.list.d` directory was removed. I'll add an error-handling there.

The actual issue is a duplicate of #4600. I'll check the Armbian forum whether this has been reported already. Reasonably a redirect from HTTPS to plain HTTP is denied and should not be done.

MichaIng, 2021-08-04:
Issue solved: https://github.com/armbian/dl-router/pull/14

closed 2021-08-04 by MichaIng