MichaIng/DietPi #4607: Dietpi-Software Update: Lidarr 0.8.1.2135 (.net version)

psi5asp, 2021-07-30:
### Creating a software request

#### Formal software information
- Software title | Lidarr [0.8.1.2135](https://github.com/lidarr/Lidarr/releases/tag/v0.8.1.2135) 
- Short description | Lidarr is a music collection manager for Usenet and BitTorrent users.
- Official URL (if available) | https://lidarr.audio
- GitHub URL (if available) | https://github.com/Lidarr/Lidarr
- (Official) install documentation | https://wiki.servarr.com/lidarr/installation

#### Are there similar/alternative software titles available with DietPi-Software?
- Probably, but Lidarr is already part of the DietPi s/w suite.

#### What makes your requested software better than the above solutions, if available?

- No longer need to install mono version of Lidarr. Net version is also more responsive and .net core is already installed by default.

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- Will conform with dietpi install/data management rules if done via dietpi-software install, rather than the apt-get package install permissions and locations. Especially regarding `dietpi_userdata` and storage space.

#### Can you provide the installation steps that you would suggest DietPi-Software to do?
- Dietpi-Software already does this, just use the .net version package instead, mainly `wget --content-disposition 'http://lidarr.servarr.com/v1/update/master/updatefile?os=linux&runtime=netcore&arch=x64'`

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.
- not required.

#### Vote for this software on FeatHub: https://feathub.com/MichaIng/DietPi/

- Current Lidarr installs as follows:

> Fetched 98.6 kB in 3s (38.4 kB/s)
> Reading package lists...
> [  OK  ] DietPi-Software | APT update
> 
>  DietPi-Software
> ─────────────────────────────────────────────────────
>  Mode: Checking for prerequisite software
> 
> [ INFO ] DietPi-Software | SQLite will be reinstalled
> [ INFO ] DietPi-Software | Mono will be installed
> 
>  DietPi-Software
> ─────────────────────────────────────────────────────
>  Mode: Installing Mono: runtime libraries and repo
> 
> [  OK  ] DietPi-Software | apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 3FA7E0328081BFF6A14DA29AA6A19B38D3D831EF
> [  OK  ] DietPi-Software | eval echo 'deb https://download.mono-project.com/repo/debian/ buster main' > /etc/apt/sources.list.d/mono-xamarin.list
> [ INFO ] DietPi-Software | APT update, please wait...
> Hit:1 https://deb.debian.org/debian buster InRelease
> .......................
> 
> [  OK  ] DietPi-Software | APT install for: mono-runtime mono-complete
> 
>  DietPi-Software
> ─────────────────────────────────────────────────────
>  Mode: Installing SQLite: Persistent single-file database system
> 
> [ INFO ] DietPi-Software | APT install for: sqlite3 php7.3-sqlite3, please wait...
> [  OK  ] DietPi-Software | APT install for: sqlite3 php7.3-sqlite3
> 
>  DietPi-Software
> ─────────────────────────────────────────────────────
>  Mode: Installing Lidarr: automatically download music
> 
> [  OK  ] DietPi-Software | Checking URL: https://github.com/Lidarr/Lidarr/releases/download/v0.8.1.2135/Lidarr.master.0.8.1.2135.linux.tar.gz
> [  OK  ] DietPi-Software | cd /tmp/DietPi-Software
> [ INFO ] DietPi-Software | G_THREAD_START_0 | curl -sSfL https://github.com/Lidarr/Lidarr/releases/download/v0.8.1.2135/Lidarr.master.0.8.1.2135.linux.tar.gz -o Lidarr.master.0.8.1.2135.linux.tar.gz
> [ INFO ] DietPi-Software | APT install for: mediainfo, please wait...
> [  OK  ] DietPi-Software | APT install for: mediainfo
> [  OK  ] DietPi-Software | G_THREAD: All threads finished
> [  OK  ] DietPi-Software | tar xf Lidarr.master.0.8.1.2135.linux.tar.gz --one-top-level=/opt
> [  OK  ] DietPi-Software | rm Lidarr.master.0.8.1.2135.linux.tar.gz

MichaIng, 2021-07-30:
Many thanks for your suggestion.

Jep this totally makes sense. The logic from Radarr can be ported, even the download URL scheme matches.

Note that this won't work on ARMv6, which does not support Mono.

psi5asp, 2021-07-31:
Radarr no longer supports or needs Mono anyhow, and now Lidarr too......soon Sonarr will do the same, since they seem to be moving off mono and using .Net instead.

I had no issues with manually installing the .Net version of Lidarr on a Pi-4. I originally installed it via dietpi-software and found the mono pre-requisite, so I uninstalled both Mono and Lidarr to then just manually install Lidarr .net version.

For existing setups using Lidarr Mono version, I'm not sure what happens to the Lidarr settings/db/data/etc. when uninstalling it. If retained, then no issues to force a re-install to the .net version of Lidarr. If deleted, then perhaps auto-backup the settings/data/etc. first and copy-back once Lidarr .net version is installed?

psi5asp, 2021-07-31:
Currently, dietpi-software Lidarr (mono) install defaults as `/opt/Lidarr/` with user-group `lidarr dietpi`, data found under `/mnt/dietpi_userdata/lidarr/` with user-group `lidarr dietpi`.

To be consistent/mirror with Radarr, I'd suggest maybe as follows (although they can be install any-which-way)
- Radarr (mono) originally Installed as /opt/Radarr/
now Radarr (dotnet) via dietpi-software installs as `/opt/radarr/` with `/mnt/dietpi_userdata/radarr/` data with simlinked logs /var/log/radarr/....etc.

 - Lidarr (mono) via dietpi-software installs as `/opt/Lidarr` with `/mnt/dietpi_userdata/lidarr/` data with simlinked logs /var/log/lidarr/....etc.
with new Lidarr (dotnet) via dietpi-software install as `/opt/lidarr` (lowercase) and rename any existing `/mnt/dietpi_userdata/lidarr/` to `/mnt/dietpi_userdata/Lidarr/` (backup data), and newly created `/mnt/dietpi_userdata/lidarr/` with simlinked logs /var/log/lidarr/....etc.

lowercase installs should be default, since *nix file/folder case matters.

MichaIng, 2021-07-31:
Yes, I'd switch to lower-case `/opt/lidarr` then as well. Migration via `dietpi-update` patches wouldn't be done via uninstall+install but via reinstall, to preserve all data and configs. When succeeded, the old `/opt/Lidarr` is removed, everything else would be migrated by the `dietpi-software` install/config steps.

psi5asp, 2021-08-07:
Would it be prudent to standardize all dietpi-software that uses the `/mnt/dietpi_data/ `folder, to store app data in `/mnt/dietpi_data/.confg/` instead of just `/mnt/dieti_data/` ?

I believe ~/.config/ is the usual place for app data/configurations eg `/home/<user>/.confg/<appfolderdata>` so to keep it consistent `/mnt/dietpi_data/.config/ `would make more sense?

There are several dietpi-software installed packages that do not use .config folder...like the *arr's, etc. when the /dietpi_data/ is store location.

MichaIng, 2021-08-08:
You mean you want to have them e.g. hidden in explorer, respectively the parent `/mnt/dietpi_userdata` list cleaner? `/mnt/dietpi_userdata/lidarr` is the home directory of the `lidarr` user, hence, depending on how the software distributes configs and data within, `/mnt/dietpi_userdata/lidarr/.config/lidarr` might exist (not in this case). The possibility `/mnt/dietpi_userdata/.config/lidarr/.config/lidarr` is a little too nested, IMO.

As we speak of system services and not (login) user services, the usual way is to store configs in `/etc/lidarr` and data in `/var/lib/lidarr`. Merging it into `/mnt/dietpi_userdata/lidarr` was more a practical choice, as this directory can be moved easily to an external drive, to reduce system drive/SD card I/O, allow easier backups or migration to a new system etc, and some software titles simply take only a single directory as input.

Another practical issue is that some software titles always use their executables directory for configs and data. In such cases, usually `/mnt/dietpi_userdata/<name>` holds the whole installation, in a fashion it is otherwise often (FHS conform) used in `/opt/<name>`. Having a whole software installation provided within a `.config` subdirectory, is a bit strange. So as of different behaviour of different software, moving all `/mnt/dietpi_userdata/<name>` to `/mnt/dietpi_userdata/.config/<name>` is definitely not the right way, but I'm open to discuss a different approach, depending on the issue behind your suggestion.

psi5asp, 2021-08-08:
Sort of yea. I was under the impression that it was the dietpi user only folder, since the name implies  dietpi_**_user_**data , not any other users' services/apps running,etc.. Also the sub-folders appear to be similar to `/home/(user)/` structure (downloads, Music, Pictures, Videos eg users' own files & sub-folders(.....

If its just for the user application/service data, then the dietpi_userdata label is a little misleading - perhaps something for future consideration to rename it as `/mnt/dietpi_data/` ?

That way each user sub-folder can exist and settings stored as mentioned above. with its relevant permissions/ownership....ie:
`/mnt/dietpi_data/dietpi/`_(all apps/service data stored here, for only the 'dietpi' user running apps)_
`/mnt/dietpi_data/lidarr/`_(all apps/service data stored here, for only the 'lidarr' user running apps)_
`/mnt/dietpi_data/radarr/`_(all apps/service data stored here, for only the 'radarr' user running apps)_
etc.. for each user....if you get my meaning. Then browsing the data folder its so much cleaner and easier to locate any app data, and can selectively backup what one wants at whatever intervals required, etc.

Otherwise if more than just app/service settings/config/data, then just use the `/home/(user)/` structure on `/mnt/home/` as normal for easy backups, portability, etc.

a little regression, albiet a little related.....anyhow food for thought.

MichaIng, 2021-08-08:
> etc.. for each user....if you get my meaning. Then browsing the data folder its so much cleaner and easier to locate any app data

That is how it is done now, aside of the "dietpi" user, which has its home in `/home/dietpi` (like any login user automatically), and will be removed in future versions (the last software titles running with it will get their own user, creating a custom login user will be offered instead on first boot). I agree that `dietpi_userdata` is probably a bit misleading. I think originally it was designed to be the userdata directory indeed and using it for app data and config joined for mentioned reasons. Also some logic is there to have dedicated download/media directories there, as many software titles (downloaders, file servers, media players and streaming servers) require non-exclusive access to music, videos, downloads, etc: downloader must be able to download to "downloads", Sonarr must be able to import from "downloads" to "videos", Samba must be able to R/W "videos" and Kodi at least requires read access, write access as well when metadata is to be changed.

It's basically two conflicting concepts of running software and storing variable data in general: running everything with the unprivileged login user, storing everything in its home directory, versus running every daemon/service as dedicated service user, storing everything in a system directory not associated to any specific login user, providing access via group memberships. Debian packages follow the second approach mostly, while many other software titles we install from (Git) sources directly are intended to be used with the first approach. Setting up both so that they work together the challenge.

psi5asp, 2021-08-13:
Ok that mostly makes sense - the name does however give the wrong impression (it's not intuitive).

Also the dietpi (user) home folder is missing when SMB'd to the pi, instead showing /mnt/dietpi_data/ folder content (with default permissions, r access to sub-folders, etc.). Admins can ofc manually change this but by default its a little mixed up and a security concern?

MichaIng, 2021-08-13:
> the name does however give the wrong impression (it's not intuitive).

Agreed. It's however a little difficult to change such things just on a regular update, as this path is used in so many created service files, configs etc. I think a change can only be done when keeping the old patch valid via symlink, which causes even more confusion for existing systems.

That SMB and other cloud/network drive solutions use this path instead of the `dietpi` login user's home directory, is a security feature, not an issue: There may be private information, SSH keys and other crypto keys etc stored in users' home directories, which must never be part of a public network share or cloud share solution. The `/mnt/dietpi_userdata` directory is explicitly not a home directory of a login user, but a separate "global" media directory to be used by multiple services/system users.

> Also SSH'd with dietpi access to the other user folders (/mnt/dietpi_data/, etc.)

It is key here that this directory is not any users directory, but a global system directory. The setup is meant to be used by a single real person, or multiple persons using/sharing the same software/services and media, like a family. I think that covers 99% of the cases, at least if an SBC is used. I never heard of an SBC that is used by multiple people with their own login user account, own software and such. If this is the case, then of course using `/mnt/dietpi_userdata` at all may be not applicable, aside of keeping it as server-software data store, for databases, or e.g. Nextcloud data and such, which are naturally accessible only to the service user, not any login user (aside of root, of course).

MichaIng, 2021-08-20:
PR up: #4665

closed 2021-08-20 by MichaIng