MichaIng/DietPi #4608: cifs not supported after software upgrade :(

k1r0s, 2021-07-30:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=4
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
```
10.10
```

- Kernel version | `uname -a`
```
Linux DietPi 5.10.17-v8+ #1421 SMP PREEMPT Thu May 27 14:01:37 BST 2021 aarch64 GNU/Linux

```

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPi 4 B
- Power supply used | (EG: 5V 1A RAVpower)
PSU 5v 3A
- SDcard used | (EG: SanDisk ultra)
usb
#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### brief
I've been using a smb network share for months, in the recent updates my share won't automatically mount so I tried to manually mount it using the dietpi-drive_manager and it says that cifs isn't supported on this fs. 

I tried to debug a bit looking for packages without success: 

```
dietpi@DietPi:~$ sudo apt install cifs-utils
Reading package lists... Done
Building dependency tree       
Reading state information... Done
cifs-utils is already the newest version (2:6.8-2).

```

#### Steps to reproduce

```
dietpi@DietPi:~$ lsmod | grep cifs
dietpi@DietPi:~$ modprobe cifs
modprobe: ERROR: ../libkmod/libkmod.c:586 kmod_search_moddep() could not open moddep file '/lib/modules/5.10.17-v8+/modules.dep.bin'
modprobe: FATAL: Module cifs not found in directory /lib/modules/5.10.17-v8+

```

I need some help here, thanks in advance!

Joulinar, 2021-07-30:
Hiwndoesn

k1r0s, 2021-07-30:
Hi, thanks for the quick response. 

> How does it behaves running following as root.

it gives the same outcome

MichaIng, 2021-07-30:
I guess you applied the latest kernel upgrade and did not yet reboot. In `/lib/modules/` you should see that Linux `5.10.52` is installed while currently `5.10.17` is loaded, for which no modules are present anymore and hence cannot be loaded:
```sh
reboot
```

closed 2021-07-30 by k1r0s