MichaIng/DietPi #4609: 4tb Toshiba drive

TreahRevol, 2021-08-01:
#### Details:
- Date           | Sun Aug  1 17:39:35 SAST 2021
- DietPi version | v7.4.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite (64-bit)
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | `Linux DietPi 5.10.52-v8+ #1440 SMP PREEMPT Tue Jul 27 09:56:40 BST 2021 aarch64 GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | `umount /mnt/mass`
- Exit code      | 32
- Software title | DietPi-Drive_Manager
#### Steps to reproduce:

1. I am not sure. mass is a 4tb external
2. I can't get any write permissions so wanted to format
#### Expected behaviour:

- a format I guess?
#### Actual behaviour:

- Can't format or even unmount the drive
#### Extra details:

- it also states that it can't assign a UUID to the drive:
- [FAILED] DietPi-Drive_Manager | The following drive seems to have no UUID, skipping fstab entry:
SOURCE    TARGET    FSTYPE UUID
/dev/sdb1 /mnt/mass ext4
#### Additional logs:
```
umount: /mnt/mass: target is busy.
```

Joulinar, 2021-08-01:
Hi,

To force the unmount you could try

`umount -l /dev/sdb1`

Or to check what process is using the disk

```
lsof /dev/sdb1
fuser -m /dev/sdb1
```

TreahRevol, 2021-08-01:
Thank you so much for the quick reply.

That got it unmounted, what is the best way to format the drive? As I have tried Partition as well as drive on ext4 and ntfs but both gives me no write permission even though I assign 7777 to the share?

Joulinar, 2021-08-01:
Usually you should be able to use DietPi drive manager to format the drive. Using ext4 would be preferred on Linux based systems. 

What user you tried to write to the disk?

TreahRevol, 2021-08-01:
I use dietpi-Drive_Manager for the format and it would be the normal dietpi user. I have another 1tb drive working perfectly fine with dietpi on perm 7777

Joulinar, 2021-08-01:
Can you share `ls -la /mnt`

TreahRevol, 2021-08-01:
For sure

root@DietPi:~# ls -la /mnt
total 36
drwxr-xr-x  9 root   root   4096 Aug  1 17:32 .
drwxr-xr-x 21 root   root   4096 Apr 21 23:58 ..
drwxrwxr-x 12 dietpi dietpi 4096 Jul 30 20:07 dietpi_userdata
drwxrwxr-x  2 dietpi dietpi 4096 Jan 21  2021 ftp_client
drwxr-xr-x  3 root   root   4096 Aug  1 18:32 mass
drwxr-xr-x  2 root   root   4096 Jul 30 15:45 Mass1
drwxrwxrwx  1 root   root   4096 Jul 16 07:29 media
drwxrwxr-x  2 dietpi dietpi 4096 Jan 21  2021 nfs_client
drwxrwxr-x  2 dietpi dietpi 4096 Jan 21  2021 samba

TreahRevol, 2021-08-01:
I tried the following

root@DietPi:~# chmod -R 777 /mnt/mass 
chmod: changing permissions of '/mnt/mass': Read-only file system
chmod: cannot read directory '/mnt/mass': Input/output error

Any idea on how to fix?

TreahRevol, 2021-08-01:
Seems it does not like the name change. when I left it on the uuid name it works...

TreahRevol, 2021-08-01:
> Seems it does not like the name change. when I left it on the uuid name it works...

Nevemind, worked for a few minutes and now I get an I/O error - https://i.imgur.com/9xpO12n.png

Joulinar, 2021-08-01:
the directory `/mnt/mass` is writable for user `root `only. Can you try to create files/folder as user `root`?

as well can you share kernel error messages `dmesg -l err,crit,alert,emerg`

TreahRevol, 2021-08-01:
> the directory `/mnt/mass` is writable for user `root `only. Can you try to create files/folder as user `root`?
> 
> as well can you share kernel error messages `dmesg -l err,crit,alert,emerg`

I can for some reason not (unsure of why it still shows mass as I formated and left the name as the UUID):

root@DietPi:~# ls -la /mnt                  
total 40
drwxr-xr-x 10 root   root   4096 Aug  1 19:17 .
drwxr-xr-x 21 root   root   4096 Apr 21 23:58 ..
drwxr-xr-x  3 root   root   4096 Aug  1 19:16 aa404886-bbaa-4479-a946-db66e2549e98
drwxrwxr-x 12 dietpi dietpi 4096 Jul 30 20:07 dietpi_userdata
drwxrwxr-x  2 dietpi dietpi 4096 Jan 21  2021 ftp_client
drwxr-xr-x  2 root   root   4096 Aug  1 18:46 mass
drwxr-xr-x  2 root   root   4096 Jul 30 15:45 Mass1
drwxrwxrwx  1 root   root   4096 Jul 16 07:29 media
drwxrwxr-x  2 dietpi dietpi 4096 Jan 21  2021 nfs_client
drwxrwxr-x  2 dietpi dietpi 4096 Jan 21  2021 samba
root@DietPi:~# sudo nano /etc/samba/smb.conf
root@DietPi:~# sudo service smbd restart    
root@DietPi:~# cd /mnt/aa404886-bbaa-4479-a946-db66e2549e98
root@DietPi:/mnt/aa404886-bbaa-4479-a946-db66e2549e98# ls
ls: reading directory '.': Input/output error

As for the kernel messages: https://pastebin.com/Je4wPbAr

Joulinar, 2021-08-01:
your drive has some issues and would need to be checked.

pls try to unmount first and do `fsck -a /dev/sdb1`

TreahRevol, 2021-08-01:
root@DietPi:/# fsck -a /dev/sdb1
fsck from util-linux 2.33.1
/dev/sdb1: clean, 11/244195328 files, 15616779/976754176 blocks
root@DietPi:/# 

Using DietPi-Drive_Manager: https://i.imgur.com/dk586AV.png

psi5asp, 2021-08-02:
You can also check the SMART info of the drive: `smartctl -a /dev/sdb1 | grep Health` as there may be some indications there.

MichaIng, 2021-08-02:
And forced fsck (not trusting the "clean" flag) via:
```sh
fsck -f /dev/sdb1
```

MichaIng, 2021-09-01:
I'll mark this as closed. Feel free to reopen if the issue persists.

closed 2021-09-01 by MichaIng