MichaIng/DietPi #4612: InfluxDB Fails to Start after Update

breadman100, 2021-08-03:
I submitted bug report with info.

breadman100, 2021-08-03:
influxdb.service - InfluxDB is an open-source, distributed, time series database                       │
│    Loaded: loaded (/lib/systemd/system/influxdb.service; disabled; vendor preset: enabled)               │
│   Drop-In: /etc/systemd/system/influxdb.service.d                                                        │
│            └─dietpi-services_edit.conf                                                                   │
│    Active: failed (Result: exit-code) since Tue 2021-08-03 01:14:23 BST; 3s ago                          │
│      Docs: https://docs.influxdata.com/influxdb/                                                         │
│   Process: 18792 ExecStart=/usr/lib/influxdb/scripts/influxd-systemd-start.sh (code=exited,              │
│ status=203/EXEC)                                                                                         │
│                                                                                                          │
│ Aug 03 01:14:23 DietPi-10 systemd[1]: influxdb.service: Control process exited, code=exited status=203   │
│ Aug 03 01:14:23 DietPi-10 systemd[1]: Failed to start InfluxDB is an open-source, distributed, time      │
│ series database.                                                                                         │
│ Aug 03 01:14:23 DietPi-10 systemd[1]: influxdb.service: Unit entered failed state.                       │
│ Aug 03 01:14:23 DietPi-10 systemd[1]: influxdb.service: Failed with result 'exit-code'.                  │
│ Aug 03 01:14:23 DietPi-10 systemd[1]: influxdb.service: Service hold-off time over, scheduling restart.  │
│ Aug 03 01:14:23 DietPi-10 systemd[1]: Stopped InfluxDB is an open-source, distributed, time series       │
│ database.                                                                                                │
│ Aug 03 01:14:23 DietPi-10 systemd[1]: influxdb.service: Start request repeated too quickly.              │
│ Aug 03 01:14:23 DietPi-10 systemd[1]: Failed to start InfluxDB is an open-source, distributed, time      │
│ series database.                                                                                         │
│ Aug 03 01:14:23 DietPi-10 systemd[1]: influxdb.service: Unit entered failed state.                       │
│ Aug 03 01:14:23 DietPi-10 systemd[1]: influxdb.service: Failed with result 'exit-code'.

Joulinar, 2021-08-03:
Hi,

Pla can you share `journalctl -u influxdb.service`

MichaIng, 2021-08-03:
Yes, please. I checked the bug report bug the logs are tailed as above while we require the ones before the restart failure loop started.

breadman100, 2021-08-03:
journalctl -u influxdb.service
-- Logs begin at Tue 2021-08-03 01:34:50 BST, end at Tue 2021-08-03 16:35:17 BST. --
Aug 03 01:35:24 DietPi-10 systemd[1]: Starting InfluxDB is an open-source, distributed, time series database...
Aug 03 01:35:24 DietPi-10 systemd[1]: influxdb.service: Control process exited, code=exited status=203
Aug 03 01:35:24 DietPi-10 systemd[1]: Failed to start InfluxDB is an open-source, distributed, time series database.
Aug 03 01:35:24 DietPi-10 systemd[1]: influxdb.service: Unit entered failed state.
Aug 03 01:35:24 DietPi-10 systemd[1]: influxdb.service: Failed with result 'exit-code'.
Aug 03 01:35:24 DietPi-10 systemd[1]: influxdb.service: Service hold-off time over, scheduling restart.
Aug 03 01:35:24 DietPi-10 systemd[1]: Stopped InfluxDB is an open-source, distributed, time series database.
Aug 03 01:35:24 DietPi-10 systemd[1]: Starting InfluxDB is an open-source, distributed, time series database...
Aug 03 01:35:24 DietPi-10 systemd[791]: influxdb.service: Failed at step EXEC spawning /usr/lib/influxdb/scripts/influxd-systemd
-start.sh: Permission denied
Aug 03 01:35:24 DietPi-10 systemd[1]: influxdb.service: Control process exited, code=exited status=203
Aug 03 01:35:24 DietPi-10 systemd[1]: Failed to start InfluxDB is an open-source, distributed, time series database.
Aug 03 01:35:24 DietPi-10 systemd[1]: influxdb.service: Unit entered failed state.
Aug 03 01:35:24 DietPi-10 systemd[1]: influxdb.service: Failed with result 'exit-code'.
Aug 03 01:35:25 DietPi-10 systemd[1]: influxdb.service: Service hold-off time over, scheduling restart.
Aug 03 01:35:25 DietPi-10 systemd[1]: Stopped InfluxDB is an open-source, distributed, time series database.
Aug 03 01:35:25 DietPi-10 systemd[1]: Starting InfluxDB is an open-source, distributed, time series database...
Aug 03 01:35:25 DietPi-10 systemd[806]: influxdb.service: Failed at step EXEC spawning /usr/lib/influxdb/scripts/influxd-systemd
-start.sh: Permission denied
Aug 03 01:35:25 DietPi-10 systemd[1]: influxdb.service: Control process exited, code=exited status=203
Aug 03 01:35:25 DietPi-10 systemd[1]: Failed to start InfluxDB is an open-source, distributed, time series database.
Aug 03 01:35:25 DietPi-10 systemd[1]: influxdb.service: Unit entered failed state.
Aug 03 01:35:25 DietPi-10 systemd[1]: influxdb.service: Failed with result 'exit-code'.
Aug 03 01:35:25 DietPi-10 systemd[1]: influxdb.service: Service hold-off time over, scheduling restart.
Aug 03 01:35:25 DietPi-10 systemd[1]: Stopped InfluxDB is an open-source, distributed, time series database.
Aug 03 01:35:25 DietPi-10 systemd[1]: Starting InfluxDB is an open-source, distributed, time series database...
Aug 03 01:35:25 DietPi-10 systemd[818]: influxdb.service: Failed at step EXEC spawning /usr/lib/influxdb/scripts/influxd-systemd
-start.sh: Permission denied
Aug 03 01:35:25 DietPi-10 systemd[1]: influxdb.service: Control process exited, code=exited status=203
Aug 03 01:35:25 DietPi-10 systemd[1]: Failed to start InfluxDB is an open-source, distributed, time series database.
Aug 03 01:35:25 DietPi-10 systemd[1]: influxdb.service: Unit entered failed state.
Aug 03 01:35:25 DietPi-10 systemd[1]: influxdb.service: Failed with result 'exit-code'.
Aug 03 01:35:25 DietPi-10 systemd[1]: influxdb.service: Service hold-off time over, scheduling restart.
Aug 03 01:35:25 DietPi-10 systemd[1]: Stopped InfluxDB is an open-source, distributed, time series database.
Aug 03 01:35:25 DietPi-10 systemd[1]: Starting InfluxDB is an open-source, distributed, time series database...
Aug 03 01:35:25 DietPi-10 systemd[848]: influxdb.service: Failed at step EXEC spawning /usr/lib/influxdb/scripts/influxd-systemd
-start.sh: Permission denied
Aug 03 01:35:25 DietPi-10 systemd[1]: influxdb.service: Control process exited, code=exited status=203
Aug 03 01:35:25 DietPi-10 systemd[1]: Failed to start InfluxDB is an open-source, distributed, time series database.
Aug 03 01:35:25 DietPi-10 systemd[1]: influxdb.service: Unit entered failed state.
Aug 03 01:35:25 DietPi-10 systemd[1]: influxdb.service: Failed with result 'exit-code'.
Aug 03 01:35:26 DietPi-10 systemd[1]: influxdb.service: Service hold-off time over, scheduling restart.
Aug 03 01:35:26 DietPi-10 systemd[1]: Stopped InfluxDB is an open-source, distributed, time series database.
Aug 03 01:35:26 DietPi-10 systemd[1]: influxdb.service: Start request repeated too quickly.
Aug 03 01:35:26 DietPi-10 systemd[1]: Failed to start InfluxDB is an open-source, distributed, time series database.
Aug 03 01:35:26 DietPi-10 systemd[1]: influxdb.service: Unit entered failed state.
Aug 03 01:35:26 DietPi-10 systemd[1]: influxdb.service: Failed with result 'exit-code'.

Joulinar, 2021-08-03:
There seems to be a permission issue

`Failed at step EXEC spawning /usr/lib/influxdb/scripts/influxd-systemd
-start.sh: Permission denied`

Can you share following

`ls -la /usr/lib/influxdb/scripts/`

breadman100, 2021-08-03:
ls -la /usr/lib/influxdb/scripts/
total 24
drwxr-xr-x 2 root root 4096 Aug  1 23:17 .
drwxr-xr-x 3 root root 4096 Oct 11  2019 ..
-rw-r--r-- 1 root root  533 Jul 21 20:55 influxdb.service
-rwxr--r-- 1 root root  520 Jul 21 20:55 influxd-systemd-start.sh
-rw-r--r-- 1 root root 5702 Jul 21 20:55 init.sh

Joulinar, 2021-08-03:
Ok you are missing execute permission. Pls can you try `chmod +x usr/lib/influxdb/scripts/influxd-systemd-start.sh` ans restart the service afterwards.

Joulinar, 2021-08-03:
@MichaIng 
I could replicate the behaviour on RPi 32bit as well as 64bit. Looks like it is related to latest InfluxDB release (not DietPi release) which has been released on 27th of July https://repos.influxdata.com/debian/pool/stable/i/influxdb/

MichaIng, 2021-08-03:
Ah lol I faced exactly the same on our server. Totally forgot about that. The latest InfluxDB package simply contains this file with wrong permissions, which were correct before. I though there will be lots of reports within hours, so that a hotfix release is done quickly after, but that obviously is not the case 🤔.

I fail to find their packaging sources currently to send a fix 🤔.

Joulinar, 2021-08-03:
their GitHub is full of reports for this issue. Fix is on the way https://github.com/influxdata/influxdb/pull/21987

MichaIng, 2021-08-03:
Ah great, thanks for linking. I was already about to patch a `chmod` into the [postinst script](https://github.com/influxdata/influxdb/blob/master/scripts/post-install.sh#L15-L18), but much better to have the file inside the package with correct permissions in the first place 👍.

breadman100, 2021-08-03:
Thanks everyone, I verified this workaround works for me for now:
 
chmod +x /usr/lib/influxdb/scripts/influxd-systemd-start.sh

Joulinar, 2021-08-09:
New version of InfluxDB has been released and issue should be fixed now. Going to close the issue. Feel free to reopen if needed.

closed 2021-08-09 by Joulinar