MichaIng/DietPi #4613: HDMI CEC on kodi

satinebm, 2021-08-03:
### Creating a bug report/issue

#### Required Information
- DietPi version | `7.4.2`
- Distro version | `Debian Buster`
- Kernel version | `Linux DietPiv2 5.10.52-v7l+ #1440 SMP Tue Jul 27 09:55:21 BST 2021 armv7l GNU/Linux`
- SBC model | `RPi 4 Model B (armv7l)
- Power supply used | 5V 3A
- SDcard used | SanDisk ultra

#### Additional Information (if applicable)
- Software title | Kodi 18.7 
- Was the software title installed freshly or updated/migrated? The software was installed a long time ago
- Can this issue be replicated on a fresh installation of DietPi? I don't no but in 7.3 no problem and with backup no problem

#### Steps to reproduce

1. Update to 7.3 to 7.4.2
2. Restart with kodi on autostart
3. Kodi start on TV
4. The TV does not detect the rpi4 as HDMI CEC and the remote control does not control the Kodi interface. 

#### Expected behaviour

- The remote control should control the interface of kodi
- Before the update: OK
- After update : no detection
- Restore backup  to 7.3 Ok The TV detect the rpi4 as HDMI CEC and the remote control control the Kodi interface.

Joulinar, 2021-08-03:
Hi,

Just to check if it might be related to some `apt` packages or kerne update, can you run `apt update && apt upgrade` while on 7.3 pls.

ConallB, 2021-08-03:
Same issue here. Kodi CEC no longer works following dietpi-update from 7.3 to 7.4.2.

MichaIng, 2021-08-03:
Please both try the kernel upgrade suggested by Joulinar. Since the DietPi update did not imply anything that would affect CEC support, the recent kernel release from a few days ago are likely the reason. If so, we'd need to forward this to the RPi developers.

ConallB, 2021-08-03:
Would love to but backup of my 7.3 install failed so I'm stuck on 7.4.2

Device itself runs several services that I cannot do without. 

Will try a reinstall of Kodi to see if that is the issue.

ConallB, 2021-08-03:
> Would love to but backup of my 7.3 install failed so I'm stuck on 7.4.2
> 
> Device itself runs several services that I cannot do without.
> 
> Will try a reinstall of Kodi to see if that is the issue.

No luck. Happy to post logs etc. if needed.

MichaIng, 2021-08-03:
Does `cec-client -l` actually list the TV as CEC device? Probably you need to install that CLI client first:
```sh
apt install cec-utils
```

Please try to downgrade the kernel by manually downloading and installing the older packages:
```sh
cd /tmp
curl -sSfLO https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi-bin_1.20210527-1_armhf.deb
curl -sSfLO https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi0_1.20210527-1_armhf.deb
curl -sSfLO https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-bootloader_1.20210527-1_armhf.deb
curl -sSfLO https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20210527-1_armhf.deb
apt install ./libraspberrypi-bin_1.20210527-1_armhf.deb ./libraspberrypi0_1.20210527-1_armhf.deb ./raspberrypi-bootloader_1.20210527-1_armhf.deb ./raspberrypi-kernel_1.20210527-1_armhf.deb
rm ./libraspberrypi-bin_1.20210527-1_armhf.deb ./libraspberrypi0_1.20210527-1_armhf.deb ./raspberrypi-bootloader_1.20210527-1_armhf.deb ./raspberrypi-kernel_1.20210527-1_armhf.deb
```

ConallB, 2021-08-03:
> Does `cec-client -l` actually list the TV as CEC device? 

libCEC version: 4.0.7, compiled on Linux-5.10.17-v8+ ... , features: P8_USB, DRM, P8_detect, randr, RPi, Exynos, Linux, AOCEC
Found devices: 1

device:              1
com port:            RPI
vendor id:           2708
product id:          1001
firmware version:    1
type:                Raspberry Pi

> Please try to downgrade the kernel by manually downloading and installing the older packages:
> 
> ```shell
> cd /tmp
> curl -sSfLO https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi-bin_1.20210527-1_armhf.deb
> curl -sSfLO https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/libraspberrypi0_1.20210527-1_armhf.deb
> curl -sSfLO https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-bootloader_1.20210527-1_armhf.deb
> curl -sSfLO https://archive.raspberrypi.org/debian/pool/main/r/raspberrypi-firmware/raspberrypi-kernel_1.20210527-1_armhf.deb
> apt install ./libraspberrypi-bin_1.20210527-1_armhf.deb ./libraspberrypi0_1.20210527-1_armhf.deb ./raspberrypi-bootloader_1.20210527-1_armhf.deb ./raspberrypi-kernel_1.20210527-1_armhf.deb
> rm ./libraspberrypi-bin_1.20210527-1_armhf.deb ./libraspberrypi0_1.20210527-1_armhf.deb ./raspberrypi-bootloader_1.20210527-1_armhf.deb ./raspberrypi-kernel_1.20210527-1_armhf.deb
> ```

ConallB, 2021-08-03:
> libCEC version: 4.0.7, compiled on Linux-5.10.17-v8+ ... , features: P8_USB, DRM, P8_detect, randr, RPi, Exynos, Linux, AOCEC
> Found devices: 1
>
> device: 1
> com port: RPI
> vendor id: 2708
> product id: [...]

Downgraded the kernel as requested. CEC now working again.

ConallB, 2021-08-03:
Thanks for the quick response @MichaIng 

Can confirm system is working as before. Many thanks. Will wait for update before attempting further upgrade.

MichaIng, 2021-08-03:
I forgot to verify that you're running the 32-bit image, but since the packages got installed, it seems you do 😅. Which RPi model is it, RPi4 as well?

satinebm, 2021-08-03:
Hello,

Yes, 32bit version and Rpi4

Tanks

MichaIng, 2021-08-03:
Okay, it was reported already on a different repository: https://github.com/Hexxeh/rpi-firmware/issues/267
So next kernel release should fix it. If you want to test it as well, you can install the latest development version via `rpi-update`:
```sh
apt install rpi-update
rpi-update
```

Joulinar, 2021-08-03:
@MichaIng 
You could have downgrade the kernel as well using `rpi-update` instead of downloading the different kernel packages :wink:

MichaIng, 2021-08-03:
Generally I prefer to stay on packages, so it can be easily upgraded to current stable release again and no `rpi-update` is required. But indeed in this case it would have been easier, especially since a `dietpi-update` could have been preformed without breaking CEC again, since APT/DPkg do not know about the `rpi-update` file overwrites 😃.

satinebm, 2021-08-03:
After update and upgrade :

libgssapi-krb5-2 libk5crypto3 libkrb5-3 libkrb5support0 
libraspberrypi-bin libraspberrypi0 libsndfile1 linux-libc-dev 
raspberrypi-bootloader raspberrypi-kernel

On reboot, no HDMI-SEC

So I'm testing the new kernel and ... CEC is OK, fine.

After updating from 7.3 to 7.4.2, everything works perfectly.

Thanks

Joulinar, 2021-08-03:
thx for confirmation. Keep in mind to go back to stable kernel once it has been released by RPi devs

MichaIng, 2021-08-06:
Solved in https://github.com/Hexxeh/rpi-firmware/issues/267 and new package released today 🚀. I mark this as closed.

closed 2021-08-06 by MichaIng