MichaIng/DietPi #4614: Forum registration issue in Firefox 78.12.0esr ?

inspector71, 2021-08-04:
Irony: trying to post questions to the forum instead of making them "issues" here. Despite, honestly, this being such a better medium for it!

However, cannot register. I had an account in the past but forgot email doesn't seem to do the job.

[image]

Error: `The solution you provided was incorrect`.

I did not get a chance to provide a solution :)

Tried:

1. Deleting cookies (bottom right link)
2. forget you password (email does not seem to match an account)
3. Disabling Firefox's Tracking Protection
4. EFF Privacy Badger is blocking `google.com`. Captchya Google's version? Allow `google.com`. No difference.
5. NoScript is enabling all three domains. No difference.
6. uBlock disabled, including cosmetic filtering. No difference.

Joulinar, 2021-08-04:
Hi, 

Could you share your forum user name and could have a look.

inspector71, 2021-08-04:
I think it was the same as my github username. Possibly not though

MichaIng, 2021-08-04:
I checked back with reCAPTCHA and generally it works, so it seems it gives you a low reputation, too low for the registration to pass. reCAPTCHAv3 does not have the option anymore to solve a puzzle, so either the score is too low or not, which is not great, I agree. And also phpBB does not show this great, while I remember there has an enhancement been merged for next release already.

Do you have a Google account? If so can you try to login at some Google service and then try again to register with the same device (IP)? Else, try to do 1-2 random Google searches and then try to register.

What I see in the stats is that the number of "high risk" (score below 0.5, in 0 - 1) significantly increased with the beginning of June, while the number of "low risk" registrations (0.5 or higher, where registration passes) stood more or less the same. Not sure if there are many more real users trying to register and Google made reCAPTCHA stricter, or, and that is what I think, a new group of spam bots is trying to register and is successfully blocked by reCAPTCHA:
[image]

inspector71, 2021-08-05:
> I checked back with reCAPTCHA and generally it works, so it seems it gives you a low reputation, ...

Understandable. Nothing less than I deserve.

Wow Michal, exceptional detail :)

Appears it might have been a VPN issue whilst indeed, I had no idea `recaptchya v3` was invisible. 

Sorry to be a PITA. I'll try to reduce my level of bothering the hard-working DietPi legends.

closed 2021-08-05 by inspector71

inspector71, 2021-08-06:
On Thu., 5 Aug. 2021, 9:38 pm MichaIng, ***@***.***> wrote:

> Nothing less than I deserve.
>
> 😄, not to confuse the reputation that Google gives the particular IP
> address you are currently connecting with, with any important and actual
> reputation of your person.
>
yeah I knew / know what it meant, I just couldn't cognitively avoid going
with the other association of the concept. was having a difficult time

speaking of time, sorry to waste yours like this

> It's some automated algorithm which, AFAIK, mainly checks whether the IP
> address has been used anywhere in combination with real user inputs, or
> not. Since Google's services are directly or indirectly involved vastly
> across the internet, this seems to generally work fine to separate bots
> from real users, only based on the IP address. But yeah, not 100%,
> obviously.
>
> Ah yes, a VPN, and I guess Tor and such may be a reason. It's good to have
> such reports collected, and if they become too much, we might switch back
> to reCAPTCHAv2, where one can hit this "I'm not a bot" button and in case
> solve a puzzle. Will be interesting to track whether the amount of spam
> bots then increases significantly, or not, e.g. as our second SFS layer
> filters them good enough 😎.
>

that sounds like a good approach. I'm very surprised they are trying to do
an invisible solution ... as much as I know how people hate filling out the
not a robot games. kinda feels a bit more of a skullduggery approach the
likes of which goowellian does enough of already ... just when some
transparency (annoying as the notices were/are) regarding hidden spyware
like cookies was being given appropriate regulatory attention. oh well.
with botnets and the like, I do not envy anybody who hosts websites these
days, let alone interactive services. Is cloudfront / flare a solid
solution easily thrown at the problem of nefarious access attempts (DDoS
etc), if a host / published has the cash to implement it of course?

SFS

Secondary * Security ?

MichaIng, 2021-08-06:
> speaking of time, sorry to waste yours like this

Not at all, as I said its good to have some feedback on this topic, as forum registrations must work reliable.

> kinda feels a bit more of a skullduggery approach the likes of which goowellian does enough of already

I would call it a brave approach and, as it seems to work for the majority of cases, a prove of how much data goowellian is collecting to be able to rate IP addresses that way. Nothing new, after all...
It is not completely hidden btw, as you see the reCAPTCHA icon at the bottom right of the page, like with other reCAPTCHA versions. Also v2 was mostly non-interactive, only in case of a too low rating, one was given the ability to raise it by pressing the button and then in case solving the puzzle. That is gone with v3 and instead the rating is returned to the website to proceed with. In case of phpBB, by default a rating of >=0.5 is required to register, but it can be chosen, and in theory websites could implement their own puzzles to allow passing for lower ratings. That such puzzles are mostly hacked by bots, and hence do not work reliable anymore, is I think another reason why Google stopped using them at all. There is now [hCAPTCHA](https://www.hcaptcha.com/) as alternative, which works very similar to reCPATCHAv2 and is used more and more (not yet supported by phpBB), would be interesting to have some experience/data from websites about how effective this method still works 🤔.

> Secondary * Security ?

SFS = [Stop Forum Spam](https://stopforumspam.com/)

ravenclaw900, 2021-08-06:
When using Tor, hCAPTCHA is what Cloudflare puts on pretty much every website.  I personally always found it much more annoying and harder than reCAPTCHA, but that's just my experience.

MichaIng, 2021-08-06:
I forgot about Cloudflare, indeed it switched to hCAPTCHA. We also use it, but I lowered the security level to minimum, so nearly no one should be annoyed by an hCAPTCHA on regular website access. The problem with Cloudflare is that it does not and cannot differentiate between common homepage access and a login form. While I do not care about bots and hackers scraping and accessing every bit of our website, but want every user, including VPN and Tor users to browse our website with least possible disturbance, I do care about who is able to register to our forum, and, as one brute-force protection element, who is able to try logins on our blog. Hence a second layer exclusively for registration and login forms is required, which is reCAPTCHA currently.

What Cloudflare at least may help with, is DDoS attacks, although one last year was not blocked at all by Cloudflare, for some reason, while our VPS provider was brought to its knees. But the main reason for us using it is the edge cache, to reduce traffic and load on the actual server and speed up downloads and page loads for users, by having things served from a closer and faster Cloudflare edge cache server.