MichaIng/DietPi #4615: [Bazarr] Installation broken

psi5asp, 2021-08-04:
### Creating a bug report/issue

#### Required Information
- DietPi version | `G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=4
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
G_LIVE_PATCH_STATUS[0]='applied'`

- Distro version | `buster 10.10`
- Kernel version | `Linux Pi-1 5.10.52-v8+ #1440 SMP PREEMPT Tue Jul 27 09:56:40 BST 2021 aarch64 GNU/Linux`
- SBC model | `RPi 4 Model B (aarch64)`
- Power supply used | 5V 3A (Official)
- SDcard used | SanDisk ultra 400

#### Additional Information (if applicable)
- Software title | Bazarr
- Was the software title installed freshly or updated/migrated? Freshly
- Can this issue be replicated on a fresh installation of DietPi? Unknown

- Bug report ID | `5b241bb0-0380-4659-82d6-a134e82b2e1b`

#### Steps to reproduce

1. Use dietpi-software and install Bazarr
2. Check <ip>: 6767 for the page load error

#### Expected behaviour

- Once Bazarr is installed, it should be accessible via the IP webpage and port 6767.

#### Actual behaviour

- Getting page load error. It appears that Bazarr is running (ps -aux) and port 6767 is accessible (a service is running), but no page comes up. It is possibly a webserver issue, except that the systemctl logs indicate its a Bazarr signal error to Radarr - which should not be set/specified in the Bazarr *install*. 
 
> Internal Server Error
> The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

#### Extra details

- Already had the following installed:
- Radarr (current version) - I noticed Bazarr appeared to config for it, however I won't be using Bazarr with Radarr (on the same machine)
- Webserver Preference : Nginx
- systemctl shows the following:

> root:/# systemctl status bazarr
> ● bazarr.service - Bazarr Daemon (DietPi)
>    Loaded: loaded (/etc/systemd/system/bazarr.service; disabled; vendor preset: enabled)
>    Active: active (running) since *** ; 56min ago
>  Main PID: 1085 (python3)
>     Tasks: 3 (limit: 4915)
>    CGroup: /system.slice/bazarr.service
>            ├─1085 /usr/bin/python3 /opt/bazarr/bazarr.py -c /mnt/dietpi_userdata/bazarr
>            └─1274 /usr/bin/python3 -u /opt/bazarr/bazarr/main.py -c /mnt/dietpi_userdata/bazarr
> 
> Bazarr[1085]: *** - app   (7fa790f6d8) :  ERROR (app:1891) - Exception on / [GET]
> Bazarr[1085]: *** - app   (7fa790f6d8) :  ERROR (app:1891) - Exception on /favicon.ico [GET]
> Bazarr[1085]: *** - root  (7fa79f1e80) :  ERROR (signalr_client:143) - BAZARR SignalR client for Radarr connection as been lost. Trying to reconnect...
> Bazarr[1085]: *** - root  (7fa790f908) :  INFO (signalr_client:124) - BAZARR SignalR client for Radarr is connected and waiting for events.
> Bazarr[1085]: *** - root  (7fa790f4a8) :  INFO (signalr_client:124) - BAZARR SignalR client for Radarr is connected and waiting for events.
> Bazarr[1085]: *** - root  (7fa790fc50) :  ERROR (get_movies:545) - BAZARR Error trying to get movies from Radarr. Timeout Error.
> Bazarr[1085]: *** - root  (7fa790f5c0) :  ERROR (get_movies:545) - BAZARR Error trying to get movies from Radarr. Timeout Error.
> Bazarr[1085]: *** - root  (7fa790f390) :  ERROR (get_movies:545) - BAZARR Error trying to get movies from Radarr. Timeout Error.
> Bazarr[1085]: *** - app   (7fa79f1e80) :  ERROR (app:1891) - Exception on / [GET]
> Bazarr[1085]: *** - app   (7fa79f1e80) :  ERROR (app:1891) - Exception on /favicon.ico [GET]

- app seems to have launched:

> root:/# ps -aux | grep bazarr
> bazarr   25672  0.0  0.1  17844  9952 ?        Ss   16:05   0:00 /usr/bin/python3 /opt/bazarr/bazarr.py -c /mnt/dietpi_userdata/bazarr
> bazarr   25673  1.9  1.5 246276 118252 ?       Sl   16:05   0:15 /usr/bin/python3 -u /opt/bazarr/bazarr/main.py -c /mnt/dietpi_userdata/bazarr

Joulinar, 2021-08-04:
Hi,

Short question. This happen right after fresh installation of Bazarr?

psi5asp, 2021-08-04:
Yes

Joulinar, 2021-08-04:
@MichaIng 
can you have a look. Looks like we can't install Bazarr via Git anymore https://github.com/morpheus65535/bazarr/issues/1476#issuecomment-883320661

Probably we need to adjust our install path https://wiki.bazarr.media/Getting-Started/Installation/Linux/linux/

Looks like it has been introduced on version v0.9.6 2 weeks back

psi5asp, 2021-08-04:
I disabled Radarr option in the Bazarr config to try to isolate the issue.

- on restart (before accessing the webpage) - it seems to launch fine:

> root@:/mnt/dietpi_userdata/bazarr/config# systemctl status bazarr
> ● bazarr.service - Bazarr Daemon (DietPi)
>    Loaded: loaded (/etc/systemd/system/bazarr.service; disabled; vendor preset: enabled)
>    Active: active (running) since Wed 2021-08-04 16:56:19 ACST; 6s ago
>  Main PID: 28696 (python3)
>     Tasks: 2 (limit: 4915)
>    CGroup: /system.slice/bazarr.service
>            ├─28696 /usr/bin/python3 /opt/bazarr/bazarr.py -c /mnt/dietpi_userdata/bazarr
>            └─28720 /usr/bin/python3 -u /opt/bazarr/bazarr/main.py -c /mnt/dietpi_userdata/bazarr
> 
> systemd[1]: Started Bazarr Daemon (DietPi).
> root@:/mnt/dietpi_userdata/bazarr/config#

- However when trying to access the webpage (page error) - systemctrl shows these issues:

> root@:/mnt/dietpi_userdata/bazarr/config# systemctl status bazarr
> ● bazarr.service - Bazarr Daemon (DietPi)
>    Loaded: loaded (/etc/systemd/system/bazarr.service; disabled; vendor preset: enabled)
>    Active: active (running) since ***; 11min ago
>  Main PID: 28696 (python3)
>     Tasks: 3 (limit: 4915)
>    CGroup: /system.slice/bazarr.service
>            ├─28696 /usr/bin/python3 /opt/bazarr/bazarr.py -c /mnt/dietpi_userdata/bazarr
>            └─28720 /usr/bin/python3 -u /opt/bazarr/bazarr/main.py -c /mnt/dietpi_userdata/bazarr
> 
> *** systemd[1]: Started Bazarr Daemon (DietPi).
> *** Bazarr[28696]: *** - root  (7fa590b308) :  INFO (driver:121) - Generating grammar tables from /usr/lib/python3.7/lib2to3/Grammar.txt
> *** Bazarr[28696]: *** - root  (7fa590b308) :  INFO (driver:121) - Generating grammar tables from /usr/lib/python3.7/lib2to3/PatternGrammar.txt
> *** Bazarr[28696]: *** - root  (7fa590b308) :  INFO (server:39) - BAZARR is started and waiting for request on http://0.0.0.0:6767
> *** Bazarr[28696]: *** - app   (7f9e61cc50) :  ERROR (app:1891) - Exception on / [GET]
> *** Bazarr[28696]: *** - app   (7f9e61cc50) :  ERROR (app:1891) - Exception on /favicon.ico [GET]
> *** Bazarr[28696]: *** - app   (7f9e61cc50) :  ERROR (app:1891) - Exception on / [GET]
> *** Bazarr[28696]: *** - app   (7f9e61cc50) :  ERROR (app:1891) - Exception on /favicon.ico [GET]

- stopping the service seems fine

> root@:/mnt/dietpi_userdata/bazarr/config# systemctl stop bazarr
> root@:/mnt/dietpi_userdata/bazarr/config# systemctl status bazarr
> ● bazarr.service - Bazarr Daemon (DietPi)
>    Loaded: loaded (/etc/systemd/system/bazarr.service; disabled; vendor preset: enabled)
>    Active: inactive (dead)
> 
> *** Bazarr[28696]: *** - app  (7f9e61cc50) :  ERROR (app:1891) - Exception on /favicon.ico [GET]
> *** Bazarr[28696]: *** - app  (7f9e61cc50) :  ERROR (app:1891) - Exception on / [GET]
> *** Bazarr[28696]: *** - app  (7f9e61cc50) :  ERROR (app:1891) - Exception on /favicon.ico [GET]
> *** Bazarr[28696]: *** - app  (7f9e61cc50) :  ERROR (app:1891) - Exception on / [GET]
> *** Bazarr[28696]: *** - app  (7f9e61cc50) :  ERROR (app:1891) - Exception on /favicon.ico [GET]
> *** systemd[1]: Stopping Bazarr Daemon (DietPi)...
> *** Bazarr[28696]: Bazarr starting...
> *** Bazarr[28696]: Bazarr exited.
> *** systemd[1]: bazarr.service: Succeeded.
> *** systemd[1]: Stopped Bazarr Daemon (DietPi).

- restarting it initially is fine:

> root@:/mnt/dietpi_userdata/bazarr/config# systemctl start bazarr
> root@:/mnt/dietpi_userdata/bazarr/config# systemctl status bazarr
> ● bazarr.service - Bazarr Daemon (DietPi)
>    Loaded: loaded (/etc/systemd/system/bazarr.service; disabled; vendor preset: enabled)
>    Active: active (running) since ***; 1s ago
>  Main PID: 15906 (python3)
>     Tasks: 2 (limit: 4915)
>    CGroup: /system.slice/bazarr.service
>            ├─15906 /usr/bin/python3 /opt/bazarr/bazarr.py -c /mnt/dietpi_userdata/bazarr
>            └─15925 /usr/bin/python3 -u /opt/bazarr/bazarr/main.py -c /mnt/dietpi_userdata/bazarr
> 
[... 24 lines cut ...]
> *** Bazarr[15906]: *** - peewee.sqliteq  (7fa5f215c0) :  DEBUG (sqliteq:176) - received query UPDATE "table_settings_languages" SET "name" = ? WHERE ("table_settings_languages"."code3" = ?)
> *** Bazarr[15906]: *** - peewee          (7fa5f215c0) :  DEBUG (peewee:3132) - ('UPDATE "table_settings_languages" SET "name" = ? WHERE ("table_settings_languages"."code3" = ?)', ['Chinese Simplified', 'zho'])
> 
> root@:/mnt/dietpi_userdata/bazarr/config# systemctl status bazarr
> ● bazarr.service - Bazarr Daemon (DietPi)
>    Loaded: loaded (/etc/systemd/system/bazarr.service; disabled; vendor preset: enabled)
>    Active: active (running) since ***; 34s ago
>  Main PID: 15906 (python3)
>     Tasks: 3 (limit: 4915)
>    CGroup: /system.slice/bazarr.service
>            ├─15906 /usr/bin/python3 /opt/bazarr/bazarr.py -c /mnt/dietpi_userdata/bazarr
>            └─15925 /usr/bin/python3 -u /opt/bazarr/bazarr/main.py -c /mnt/dietpi_userdata/bazarr
> 
> *** Bazarr[15906]:     return self._load_template(name, self.make_globals(globals))
> *** Bazarr[15906]:   File "/opt/bazarr/bazarr/../libs/jinja2/environment.py", line 804, in _load_template
> *** Bazarr[15906]:     template = self.loader.load(self, name, globals)
> *** Bazarr[15906]:   File "/opt/bazarr/bazarr/../libs/jinja2/loaders.py", line 113, in load
> *** Bazarr[15906]:     source, filename, uptodate = self.get_source(environment, name)
> *** Bazarr[15906]:   File "/opt/bazarr/bazarr/../libs/flask/templating.py", line 60, in get_source
> *** Bazarr[15906]:     return self._get_source_fast(environment, template)
> *** Bazarr[15906]:   File "/opt/bazarr/bazarr/../libs/flask/templating.py", line 89, in _get_source_fast
> *** Bazarr[15906]:     raise TemplateNotFound(template)
> *** Bazarr[15906]: jinja2.exceptions.TemplateNotFound: index.html
> 
> root@:/mnt/dietpi_userdata/bazarr/config#

Joulinar, 2021-08-04:
@psi5asp 
there seems to be an issue how Bazarr is installed at the moment as there was a change introduced on latest version. No need to continue testing from your end.

psi5asp, 2021-08-04:
- ininstalled (not sure if clean, but seems to have been):

> root@:/mnt/dietpi_userdata# dietpi-software
> [  OK  ] DietPi-Software | Initialised database
> [  OK  ] DietPi-Software | Reading database
> 
>  DietPi-Software
> ─────────────────────────────────────────────────────
>  Mode: Uninstalling Bazarr: automatically download subtitles
> 
> 
>  DietPi-Software
> ─────────────────────────────────────────────────────
>  Mode: Finalise uninstall
> 
> [ INFO ] DietPi-Software | APT autopurge, please wait...
> [  OK  ] DietPi-Software | APT autopurge
> [  OK  ] DietPi-Software | systemctl daemon-reload
>

Joulinar, 2021-08-04:
One more hint. Next time use the `code` tag to insert logs and error message information, instead of the `quote` option. For now I have corrected all your comments.

Joulinar, 2021-08-04:
as workaround you could do following in meantime until we adjusted our install procedure

```
cd /opt
git clone --depth 1 --branch v0.9.5 https://github.com/morpheus65535/bazarr.git
dietpi-software install 180
```

closed 2021-08-20 by MichaIng