MichaIng/DietPi #4616: How to access DLNA devices in the home network with Wireguard

GvY85, 2021-08-04:
This is probably very simple to fix but I am a bit of a networking/iptables n00b so I cant seemt to figure this one out.
Have been using 2 DietPi devices for years.
One has Pi-hole + Unbound + Wireguard (PiVPN) server and number two has Samba + Wireguard client (to Mullvad) + Qbittorrent-nox + MiniDLNA.
This is all behind a Fritz!Box router with a open port to device One so that all my mobile devices have a Wireguard tunnel to the home network and use Pi-hole. So far so good and this has been working for a few years.
However, when I try to access device Two that is running MiniDLNA via the Wireguard tunnel that does not seem to work.
So, I cannot watch anything via DLNA (with VLC) on device Two when connected via Wireguard to device One. Everything works fine while I am in the home Wifi network and disconnected from Wireguard.

My assumption is that the problem lies in the fact that my home network has a 192.168.178.0/24 network range and WG devices get a 10.0.0./24(?) range?
So it seems WG clients can connect to the internet via device One but not to all the devices in the home network. Wierd thing is that I can access the qBittorrent webinterface running on device Two and the Admin page of the Fritz!Box but not MiniDLNA on device Two so I can access more devices than just the internet and device One
Is WG by default only allowing some ports or?

Joulinar, 2021-08-04:
Hi,

Wireguard server device 1 did not block any ports and you should be able to reach every device in your local network. Issue might be with the 2nd device connected to a public VPN provider. Could you try to disconnect VPN to Mullvad and try again if you are able to reach the 2nd device from extern.

GvY85, 2021-08-04:
Thanks for the quick reply.

```
# wg-quick down mullvad-nl4
[#] ip -4 rule delete table 51820
[#] ip -4 rule delete table main suppress_prefixlength 0
[#] ip -6 rule delete table 51820
[#] ip -6 rule delete table main suppress_prefixlength 0
[#] ip link delete dev mullvad-nl4
[#] resolvconf -d mullvad-nl4 -f
[#] iptables-restore -n
[#] ip6tables-restore -n
```

Now I can find it indeed. 

Enabling it again it doesnt work anymore:

```
# wg-quick down mullvad-nl4
[#] ip -4 rule delete table 51820
[#] ip -4 rule delete table main suppress_prefixlength 0
[#] ip -6 rule delete table 51820
[#] ip -6 rule delete table main suppress_prefixlength 0
[#] ip link delete dev mullvad-nl4
[#] resolvconf -d mullvad-nl4 -f
[#] iptables-restore -n
[#] ip6tables-restore -n
root@DownloadBox:~# wg-quick up  mullvad-nl4
[#] ip link add mullvad-nl4 type wireguard
[#] wg setconf mullvad-nl4 /dev/fd/63
[#] ip -4 address add 10.67.158.86/32 dev mullvad-nl4
[#] ip -6 address add fc00:bbbb:bbbb:bb01::4:9e55/128 dev mullvad-nl4
[#] ip link set mtu 1420 up dev mullvad-nl4
[#] resolvconf -a mullvad-nl4 -m 0 -x
[#] wg set mullvad-nl4 fwmark 51820
[#] ip -6 route add ::/0 dev mullvad-nl4 table 51820
[#] ip -6 rule add not fwmark 51820 table 51820
[#] ip -6 rule add table main suppress_prefixlength 0
[#] ip6tables-restore -n
[#] ip -4 route add 0.0.0.0/0 dev mullvad-nl4 table 51820
[#] ip -4 rule add not fwmark 51820 table 51820
[#] ip -4 rule add table main suppress_prefixlength 0
[#] sysctl -q net.ipv4.conf.all.src_valid_mark=1
[#] iptables-restore -n

```

GvY85, 2021-08-04:
Hmm, nevermind, if I do a `service minidlna restart` it pops up instantly, wierd stuff

Joulinar, 2021-08-04:
I guess Mullvad will set `iptables` routes to forward all outgoing traffic to the Mullvad interface. This would be a different way than the incoming traffic via your 1st device. But strange restarting the service fixed it.

GvY85, 2021-08-04:
Have to test it on antother device later but on the home ip adress range it works (on my tv) when mullvad is on and off so mullvad rerouting all outgoing traffic is not the case then right?

Joulinar, 2021-08-04:
I'm not an expert on network routing but maybe it make a difference if you access the device from local network or via VPN from extern

GvY85, 2021-08-04:
Solved it: the `notify_interval=` in `/etc/minidlna.conf` was set to `895 `meaning time between anounces is 15 minutes. Set it to 60 (1 min) and now it always finds the server witin 1 min.
Thanks for the help

GvY85, 2021-08-04:
Hmm, did not solve it. It works on an Android device with VLC but on iOS I still need to be connected to the home wifi and disconnect the vpn to home for VLC to detect the DLNA (upnp) server

GvY85, 2021-08-16:
Does anyone have a clue how to solve this?
Basically I have the following use cases:

A home network with 192.168.178.0/24 range and some devices:
A. My Dietpi Pihole + PiVPN device that acts as a Wireguard server and gives out 10.6.0.0/24 adresses to my mobile devices.
B. A Dietpi Qbittorrent + MiniDLNA server with a Wireguard client connected to a VPN service
C. A TV used to watch content from the MiniDLNA server via WiFi.
D. Mobile devices (Android + Apple) used to access MiniDLNA server over wifi or mobile via Wireguard

So, while device B is connected to its VPN device C can access the content from MiniDLNA over Wifi, no problem there.
So it seems the problem lies not in the fact that device B is connected to a VPN.
When connected wo the home network via Wifi I can access device B with Wireguard enabled on devices D
As soon as I am connected via mobile network and wireguard to the home network I cannot access MiniDLNA on device B but I still can access the Qbittorrent Admin page on device B and Pihole page on device A.

So it seems that for some reason DLNA gets blocked when accessing via mobile network + wireguard?

MichaIng, 2021-08-16:
Probably MiniDLNA is bound to/strictly listens only to the `192.168.178.0/24` network? You can check where is listens to via:
```sh
ss -tulpn
```

GvY85, 2021-08-16:
It listens to 0.0.0.0:8200

MichaIng, 2021-08-16:
Does it actually make a difference whether the VPN client is active on B?

Finally `tcpdump` could be used to catch all incoming and outgoing packets to see whether the requests from the mobile phone arrive at B or not, and if/how they are answered.

GvY85, 2021-08-16:
No, that does not make a difference unfortinately.

MichaIng, 2021-08-16:
The HTTP interface is listening on TCP port 8200 is one, but it listens also to UDP port 1900, isn't it? Does it allow all remote addresses there as well? `0.0.0.0:1900`

GvY85, 2021-08-16:
Hmm, on `1900` it listens to `239.255.255.250:1900`

MichaIng, 2021-08-16:
Partly I understand. Can you verify first the multicast is by default disabled on the `wg0` interface:
```sh
ip l show wg0
```
If there is no `MULTICAST` in the output, enable it:
```sh
ip l set wg0 multicast on
```

And then you can check whether the TTL of those packets is really 1 (hop) only:
```sh
apt install tcpdump
tcpdump -v 'udp port 1900'
```
Then do some DLNA access while tcpdump will print detailed information about every packet it sends or receives on UDP port 1900.

GvY85, 2021-08-16:
`ip l show wg0`:

```
 wg0: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1420 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/none
```

`ip l set wg0 multicast on`
`ip l show wg0`:

```
wg0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1420 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/none
```

`tcpdump -v 'udp port 1900'`:
from the mobile network or Wifi guest network: nothing
from the home network but via pivpn/wireguard:
`tcpdump: listening on eth0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
18:33:10.794149 IP (tos 0x0, ttl 4, id 3700, offset 0, flags [DF], proto UDP (17), length 314)
    DownloadBox.fritz.box.37722 > 239.255.255.250.1900: UDP, length 286`

So TTL seems to be 4. 
Really wierd it still does not work, even with Multicast on

MichaIng, 2021-08-16:
This is an outgoing packet. Is the TTL the same for incoming packets? That should be sufficient then.

The last part is the routing, where I have no idea how this works in case of multicast. The guide you found uses some tool set to manage routing groups, but I don't fully understand that yet. Can you check if there are any special routes present, e.g. created by MiniDLNA on startup or so:
```sh
ip r
```

GvY85, 2021-08-16:
`ip r` on the device with MiniDLNA:
`default via 192.168.178.1 dev eth0 onlink
192.168.178.0/24 dev eth0 proto kernel scope link src 192.168.178.4`

on the device with pivpn:
`default via 192.168.178.1 dev eth0 onlink
10.6.0.0/24 dev wg0 proto kernel scope link src 10.6.0.1
192.168.178.0/24 dev eth0 proto kernel scope link src 192.168.178.3`

MichaIng, 2021-08-16:
Ah it is on a different routing table. Can you check on the MiniDLNA device:
```sh
ip r table all
```

GvY85, 2021-08-16:
`ip r table all`
does not work: `Command "table" is unknown, try "ip route help".
`

MichaIng, 2021-08-16:
Ah sorry, it's:
```sh
ip r l table all
```

GvY85, 2021-08-16:
`ip route show table all` ?

`default dev mullvad-nl4 table 51820 scope link
default via 192.168.178.1 dev eth0 onlink
192.168.178.0/24 dev eth0 proto kernel scope link src 192.168.178.4
local 10.67.158.86 dev mullvad-nl4 table local proto kernel scope host src 10.67.158.86
broadcast 127.0.0.0 dev lo table local proto kernel scope link src 127.0.0.1
local 127.0.0.0/8 dev lo table local proto kernel scope host src 127.0.0.1
local 127.0.0.1 dev lo table local proto kernel scope host src 127.0.0.1
broadcast 127.255.255.255 dev lo table local proto kernel scope link src 127.0.0.1
broadcast 192.168.178.0 dev eth0 table local proto kernel scope link src 192.168.178.4
local 192.168.178.4 dev eth0 table local proto kernel scope host src 192.168.178.4
broadcast 192.168.178.255 dev eth0 table local proto kernel scope link src 192.168.178.4
default dev mullvad-nl4 table 51820 metric 1024 pref medium
::1 dev lo proto kernel metric 256 pref medium
_(ipv6)_b:bbbb:bb01::4:9e55 dev mullvad-nl4 proto kernel metric 256 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
local ::1 dev lo table local proto kernel metric 0 pref medium
local _(ipv6)_bb:bb01::4:9e55 dev mullvad-nl4 table local proto kernel metric 0 pref medium
local _(ipv6)_:c428 dev eth0 table local proto kernel metric 0 pref medium
multicast ff00::/8 dev mullvad-nl4 table local proto kernel metric 256 pref medium
multicast ff00::/8 dev eth0 table local proto kernel metric 256 pref medium
`

MichaIng, 2021-08-16:
Okay, so there are two broadcast routes for every interface, one with the subnet internal bits filled with `0`, like when defining a network range, the other one with those filled with `255`, which is the broadcast target address. I was now thinking to add two new such rules for the WireGuard subnet, but that server doesn't know about it.

You use NAT/masquerading on the WireGuard server for LAN access through the tunnel, so that for the MiniDLNA server it looks like the WireGuard server machine did the access, right? So then those broadcast routes need to be checked and in case applied on the WireGuard server.

And now, thinking about the TTL, when packet goes DLNA => Router => VPN => Client, is a TTL of 4 then assured to be sufficient? Not sure if from one machine to the next it is definitely only one hope, hence a TTL of 4 is assured to be able to pass the four links from system 1 to system 5.

GvY85, 2021-08-16:
Tried increasing the TTL on the MiniDLNA device with `iptables -t mangle -A PREROUTING -i eth0 -d 239.255.255.250 -j TTL --ttl-inc 8` but no difference.

MichaIng, 2021-08-16:
Yes its above my current knowledge as well, to be true 😄. Broadcast routes for the WireGuard interface are there, so it seems, and somehow this makes sense, that the system does not forward broadcast packets through NAT and we need to tell it to do so, which is obviously the other step in the linked guide as well. While the guide could be followed exactly, I'll do some searching tomorrow about how to achieve this with low-level `ip` tool.

GvY85, 2021-08-16:
Thinking about "the system does not forward broadcast packets through NAT and we need to tell it to do so", is that really the case?
Because there are more or less 3 situations:
`A. Phone (Wifi Home Network 192.168.178.0/24) -> Wireguard (10.6.0.1) -> PiVPN device (192.168.178.3) -> MiniDNLA device (192.168.178.4): `**works** 

`B. Phone (Wifi **Guest** Network 192.168.17**9**.0/24) -> Wireguard (10.6.0.1) -> PiVPN device (192.168.178.3) -> MiniDNLA device (192.168.178.4): `**does not work** 

`C. Phone (4G/Mobile Network IP?) -> Wireguard (10.6.0.1) -> PiVPN device (192.168.178.3) -> MiniDNLA device (192.168.178.4): `**does not work** 

So what is the difference, they all connect via Wireguard so why would it matter what kind of network/ip range is on the left side of the Wireguard connection?

MichaIng, 2021-08-16:
But in the first case, when the Phone is connected to the home WiFi, it does not tunnel the requests through the VPN, as the MiniDLNA server is in the same network, so no NATing involved.

On the other two cases, the Phone is connected to a different network, so in both those cases the VPN must be used to access the MiniDLNA network, hence NATing involved and no broadcast forwarding. So far at least the theory makes sense.

GvY85, 2021-08-17:
Hmm, that does make sense. I simply thought that with Wireguard on everything (if not told explicitly) would go trough the tunnel, to the internet and then via your ISP IP back inside the router? So in that case it would matter on what network you  would be becasue all traffic would come in via your ISP's public, outside IP into the router even if you where within the homenetwork to begin with. 
Also because as long as I keep Wireguard on I keep my 10.1.x.x address on my phone (shown in Pihole) regardless of network connection.
So my understanding of it as a tunnel that everything passes trough and always comes back via the public IP is wrong?

GvY85, 2021-08-17:
Btw, tried the guide on [Making DLNA through site-to-site VPN work](https://an0n-r0.medium.com/making-dlna-through-site-to-site-vpn-work-f393629f4ce0) but that did not work, even with TTL at 8

MichaIng, 2021-08-17:
Now that you say it, probably you are right. With OpenVPN, local/LAN routes override the VPN routes, but WireGuard handles this different, with a dedicated routing table, I forgot about that.

But most routers do not route requests to www when the target IP points to themselves, but being smart enough to route them directly to the target inside the LAN, when port forwarding is set accordingly. Otherwise I'm not sure whether the TTL need to be larger, or whether it doesn't play a role for hops of the VPN wrapping connection, of if those don't play a role since the VPN peers only see the single hop through the tunnel.

#### WARNING: Below much guessing/theory, nothing validated 😄
So then, if the above is true in your case, the difference is not NAT in general, but that the router is a second wrapping NAT in the  two failing cases. But the router should only see regular encrypted packets from the peer to the other, while it shouldn't even know that it's a VPN connection and broadcast packets within, so it shouldn't play a role.

... It may still be that the VPN server does not route broadcast packets it receives further through the tunnel. I mean this makes sense: They don't have an explicit target and are naturally only meant to be relevant through the single network segment. The broadcast routes which are currently present are most likely only for broadcast messages from the VPN server machine itself, but do not route received broadcast messages further. This would break the whole idea, as every peer has those routes and then broadcast messages would be routed in a global loop until the TTL is reached. I guess, within the same network, communication works as the DLNA server can send broadcast messages to the mobile phone directly (outside of the tunnel, as it is not connected to the VPN server in your LAN). So probably the phone talks to MiniDLNA through the VPN, but MiniDLNA answers via broadcast directly through the LAN 😄.

GvY85, 2021-08-17:
Lol, this is getting complicated. What can be so hard about connecting to your home network over a VPN and accessing a DLNA device within that network right 🤣 ?

MichaIng, 2021-11-07:
I'll mark this as closed. We seem to be unable to fully answer the question here, without intense research 😅. This should be asked/discussed somewhere more targeted, where it's about DLNA or WireGuard, or at some [Stack Exchange site](https://stackexchange.com/sites).

closed 2021-11-07 by MichaIng