MichaIng/DietPi #4623: Upgrade DietPi version v6.34.3 to v7.4.2

olivamauricio, 2021-08-05:
#### Details:
- Date           | Thu Aug  5 19:13:41 -03 2021
- DietPi version | v6.34.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite 64-bit
- Hardware       | RPi 3 Model B (aarch64) (ID=3)
- Kernel version | `Linux DietPi 5.10.52-v8+ #1441 SMP PREEMPT Tue Aug 3 18:14:03 BST 2021 aarch64 GNU/Linux`
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | `apt-get -qq upgrade`
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1.  Use dietpi-update
2.  apt-get -qq upgrade
#### Expected behaviour:

- update to v7.4.2
#### Actual behaviour:

- E: Packages were downgraded and -y was used without --allow-downgrades.
#### Extra details:

- apt-get update - Ok
- apt-get upgrade - no ok
#### Additional logs:
[ SUB2 ] DietPi-Set_software > apt-mirror (http://ftp.us.debian.org/debian/)
[  OK  ] DietPi-Set_software | Setting in /boot/dietpi.txt adjusted: CONFIG_APT_DEBIAN_MIRROR=http://ftp.us.debian.org/debian/
[  OK  ] apt-mirror http://ftp.us.debian.org/debian/ | Completed
[ INFO ] DietPi-Config | APT update, please wait...
Hit:1 https://deb.debian.org/debian-security buster/updates InRelease
Hit:2 http://packages.azlux.fr/debian buster InRelease
Hit:3 https://archive.raspberrypi.org/debian buster InRelease
Get:4 http://ftp.us.debian.org/debian buster InRelease [122 kB]
Get:5 http://ftp.us.debian.org/debian buster-updates InRelease [51.9 kB]
Get:6 http://ftp.us.debian.org/debian buster-backports InRelease [46.7 kB]
Get:7 http://ftp.us.debian.org/debian buster/main armhf Packages [7698 kB]
Get:8 http://ftp.us.debian.org/debian buster/non-free armhf Packages [62.1 kB]
Get:9 http://ftp.us.debian.org/debian buster/main arm64 Packages [7735 kB]
Get:10 http://ftp.us.debian.org/debian buster/contrib armhf Packages [40.1 kB]
Get:11 http://ftp.us.debian.org/debian buster/non-free arm64 Packages [53.8 kB]
Get:12 http://ftp.us.debian.org/debian buster/contrib arm64 Packages [38.4 kB]
Get:13 http://ftp.us.debian.org/debian buster-updates/main arm64 Packages [14.5 kB]
Get:14 http://ftp.us.debian.org/debian buster-updates/main armhf Packages [14.4 kB]
Get:15 http://ftp.us.debian.org/debian buster-backports/main armhf Packages [473 kB]
Get:16 http://ftp.us.debian.org/debian buster-backports/non-free armhf Packages [12.9 kB]
Get:17 http://ftp.us.debian.org/debian buster-backports/contrib armhf Packages [7724 B]
Get:18 http://ftp.us.debian.org/debian buster-backports/contrib arm64 Packages [8424 B]
Get:19 http://ftp.us.debian.org/debian buster-backports/non-free arm64 Packages [21.7 kB]
Get:20 http://ftp.us.debian.org/debian buster-backports/main arm64 Packages [478 kB]
Fetched 16.9 MB in 48s (348 kB/s)
Reading package lists...
[  OK  ] DietPi-Config | APT update
[ INFO ] DietPi-Update | APT upgrade, please wait...
E: Packages were downgraded and -y was used without --allow-downgrades.
[FAILED] DietPi-Update | APT upgrade
 - Command: apt-get -qq upgrade

E: Packages were downgraded and -y was used without --allow-downgrades.
```

Joulinar, 2021-08-05:
Hi,

pls can you run `apt update && apt upgrade` manually to see which packages should be downgraded

olivamauricio, 2021-08-06:
root@DietPi:~# apt update && apt upgrade
Get:1 https://deb.debian.org/debian-security buster/updates InRelease [65.4 kB]
Hit:2 http://ftp.us.debian.org/debian buster InRelease
Hit:3 http://ftp.us.debian.org/debian buster-updates InRelease
Hit:4 http://ftp.us.debian.org/debian buster-backports InRelease
Get:5 https://archive.raspberrypi.org/debian buster InRelease [32.6 kB]
Hit:6 http://packages.azlux.fr/debian buster InRelease
Get:7 https://archive.raspberrypi.org/debian buster/main arm64 Packages [287 kB]
Get:8 https://archive.raspberrypi.org/debian buster/main armhf Packages [378 kB]
Fetched 763 kB in 12s (62.0 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be DOWNGRADED:
  cmake cmake-data
0 upgraded, 0 newly installed, 2 downgraded, 0 to remove and 0 not upgraded.
Need to get 4644 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n]

Joulinar, 2021-08-06:
It's basically same as this #4385

Workaround would be to purge `cmake` package

MichaIng, 2021-08-06:
Reported [here](https://github.com/RPi-Distro/repo/issues/245) but sadly no traction yet.

olivamauricio, 2021-08-06:
Ok, thank you. 

Everything worked. Removed the package and updated.

Joulinar, 2021-08-06:
I'm going to close the issue as we can keep track on the original report.

closed 2021-08-06 by Joulinar