MichaIng/DietPi #4624: Cannot wget dps_6/99-dietpi-dpms_off.conf when install a desktop software(LXDE or XFCE) through dietpi-software

xiayang0521, 2021-08-06:
#### Details:
- Date           | Fri Aug  6 15:21:45 CST 2021
- DietPi version | v6.32.0 (MichaIng/dev)
- Image creator  | StephanStS
- Pre-image      | Armbian
- Hardware       | NanoPi K2 (aarch64) (ID=54)
- Kernel version | Linux DietPi 5.10.43-meson64 #21.05.4 SMP PREEMPT Wed Jun 16 08:12:48 UTC 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | wget --spider -t 1 -T 10 https://raw.githubusercontent.com/MichaIng/DietPi/dev/.conf/dps_6/99-dietpi-dpms_off.conf
- Exit code      | 8
- Software title | DietPi-Software
#### Steps to reproduce:

1. I am trying to install a desktop software(LXDE or XFCE) through dietpi-software
2. The command "wget --spider -t 1 -T 10 https://raw.githubusercontent.com/MichaIng/DietPi/dev/.conf/dps_6/99-dietpi-dpms_off.conf" fails  to download the file.

#### Expected behaviour:

- installing a desktop software(LXDE or XFCE) through dietpi-software successfully
#### Actual behaviour:

- The command "wget --spider -t 1 -T 10 https://raw.githubusercontent.com/MichaIng/DietPi/dev/.conf/dps_6/99-dietpi-dpms_off.conf" fails  to download the file.
- installing a desktop software(LXDE or XFCE) through dietpi-software unsuccessfully

#### Additional logs:
```
Spider mode enabled. Check if remote file exists.
--2021-08-06 15:21:34--  https://raw.githubusercontent.com/MichaIng/DietPi/dev/.conf/dps_6/99-dietpi-dpms_off.conf
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.199.108.133, 185.199.109.133, 185.199.110.133, ...
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.108.133|:443... failed: Connection timed out.
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.199.109.133|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
Remote file does not exist -- broken link!!!

Joulinar, 2021-08-06:
Hi,

Pls update your DietPi version first as yours is pretty old.

MichaIng, 2021-08-06:
Yes exactly, while it was not intended to break backwards-compatibility, an upgrade is anyway often required to keep install options functional. As long as you don't have a strong reason to stay on the old DietPi version, please upgrade first. If you have a strong reason, less try to solve this to allow the update 🙂.