MichaIng/DietPi #4627: DietPi ARMv8 RPi not booting/working

ravenclaw900, 2021-08-08:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
Should be 7.4?
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
buster
- Kernel version | `uname -a`
Don't know
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
RPI4 4GB
- Power supply used | (EG: 5V 1A RAVpower)
Canakit
- SDcard used | (EG: SanDisk ultra)
SanDisk Ultra (brand new)

#### Steps to reproduce

1. Try to boot a SD card with DietPi

#### Expected behaviour

- DietPi boots and works fine

#### Actual behaviour

- 1st time booting: Flashing cursor on screen for about a minute, before skipping straight to login screen
- 2nd time booting: Doesn't boot, fails with `Bad cluster number 0`

#### Extra details

- Tested with both BalenaEtcher and dd
- Raspbian ARMv8 worked fine

MichaIng, 2021-08-08:
Sounds like something broken with the FAT/boot partition.

Please try the following on first boot to enable additional screen output:
```sh
sed -i 's/ quiet//' /boot/cmdline.txt
```

Another approach, try to fully upgrade the Raspbian ARMv8 (most importantly the kernel upgrade that is available):
```sh
apt update
apt full-upgrade
```
And see if that causes the same issue.

ravenclaw900, 2021-08-08:
Here's something (hopefully) better than a picture:
```
dietpi@DietPi:~ $ dmesg 
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.10.17-v8+ (dom@buildbot) (aarch64-linux-gnu-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.2
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003e5fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000] Zeroed struct page in unavailable ranges: 512 pages
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000000fbffffff]
[    0.000000] On node 0 totalpages: 1025536
[    0.000000]   DMA zone: 4096 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 255488 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 12032 pages used for memmap
[    0.000000]   DMA32 zone: 770048 pages, LIFO batch:63
[    0.000000] percpu: Embedded 32 pages/cpu s91224 r8192 d31656 u131072
[    0.000000] pcpu-alloc: s91224 r8192 d31656 u131072 alloc=32*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1009408
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:7C:1B:C3 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  console=tty1 root=PARTUUID=8f4dbd00-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait net.ifnames=0 console=ttyS0,115200
[    0.000000] Kernel parameter elevator= does not have any effect anymore.
               Please use sysfs to set IO scheduler for individual devices.
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x000000003a600000-0x000000003e600000] (64MB)
[    0.000000] Memory: 3868344K/4102144K available (10944K kernel code, 1918K rwdata, 3912K rodata, 3648K init, 1240K bss, 168264K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 36199 entries in 142 pages
[    0.000000] ftrace: allocated 142 pages with 4 groups
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x3b8/0x578 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000003] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000123] Console: colour dummy device 80x25
[    0.000152] printk: console [tty1] enabled
[    0.000179] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000191] pid_max: default: 32768 minimum: 301
[    0.000257] LSM: Security Framework initializing
[    0.000375] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.000414] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.001084] Disabling memory control group subsystem
[    0.002049] rcu: Hierarchical SRCU implementation.
[    0.002532] EFI services will not be available.
[    0.002764] smp: Bringing up secondary CPUs ...
[    0.003209] Detected PIPT I-cache on CPU1
[... 226 lines cut ...]
[    2.527717] [vc_sm_connected_init]: failed to initialize shared memory service
[    2.543566] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[    2.557700] bcm2835_isp: module is from the staging directory, the quality is unknown, you have been warned.
[    2.559594] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[... cut ...]
[   23.602124] random: 7 urandom warning(s) missed due to ratelimiting
[   28.051910] usbcore: deregistering interface driver brcmfmac
[   28.570401] bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
[   28.570654] bcmgenet fd580000.ethernet eth0: Link is Down
[   32.671219] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   32.671270] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
```
All in all, everything seems to be working, except it doesn't show anything on screen (with `quiet` enabled).

MichaIng, 2021-08-09:
I was just thinking to remove `quiet` from the default `cmdline.txt`. But while quiet mutes kernel messages, the resizing process should actually be visible, as this is a systemd service which should print to the main screen on tty1.

ravenclaw900, 2021-08-14:
The ARMv8 Bullseye images work fine.

MichaIng, 2021-08-14:
Thanks for testing it. Today the default download on the website will switch to the Bullseye images (for Raspberry Pi), so let's not waste time to debug the Buster image then 🙂. I'll mark the issue as closed then.

closed 2021-08-14 by MichaIng