MichaIng/DietPi #4628: Hotspot limited to one Device

chtugha, 2021-08-09:
### Creating a bug report/issue

#### Required Information
- DietPi version | v7.4.2
- Distro version | buster
- Kernel version | Linux DietPi 5.10.52-v8+
- SBC model | RPi 4 Model B (aarch64)
- Power supply used | Strompi v3
- SDcard used | SanDisk ultra 32gb

#### Additional Information (if applicable)
- Software title | Hotspot / Hostapd
- fresh installation
- Can this issue be replicated on a fresh installation of DietPi? yes

- Bug report ID | aa6cdf00-fa32-4f41-a2c9-903ee91058b6

#### Steps to reproduce

1. Onboard Wifi Off
2. Insert Usb-Wifi-Dongle
3. Activate Hotspot on Dongle
4. Onboard Wifi On
5. Onboard Wifi is not configurable anymore via dietpi-config (i.e. Ip-Address etc..) b/c it is wlan1 now maybe?

#### Expected behaviour

- Onboard Wifi should also be fully configurable!

#### Actual behaviour

- Don't know why the Rest (wpa_supplicant etc.) is not configurable, but: 
- Hostapd is configured as systemd hostapd.service - this causes several Problems
- First: it would be much better if the onboard device (and any usb-wifi) would get a persistent udev-rule (i.e. onboard Wifi is always wlan99, and the dongle is always wlanXY)
- Second: Hostapd should NOT be running as hostapd.service but instead as hostapd@wlan99.service
- -> This is already implemented (if hostapd.conf is moved to let's say wlan0.conf) and working, and should be the standard configuration, so adding more hotspots on different adapters is easier.

#### Extra details

- There seems to be a routing/firewall/networking/whatever Problem if hostapd.conf is moved to wlan1.conf, or wlan2.conf or anything else than wlan0.conf.
- I tried a lot to get it fixed...the wifi part is working flawless. I can connect to the wifi, but the networking part is blocked.
- The Service was unmasked & enabled.
- Iptables Forwarding was set
- RFKill shows nothing is blocked
- The Adapters are both working (tested by swapping the udev) as wlan0
- I even tried to bridge both Adapters
- No Chance :P ...I must have something missed - anyway. I believe swapping from hostapd.service to hostapd@wlanX.service would make things easier for everybody.

Greetings

JM

MichaIng, 2021-08-09:
Many thanks for your report. This is a known limitation, tracked here: #2923

What you can do is configuring the second WiFi manually with a drop-in configuration like `/etc/network/interfaces.d/onboard_wifi.conf`. The onbaord WiFi will btw always be wlan0 and the external one wlan1, so while fixed naming is good, it is not really required:
```
iface wlan0 inet dhcp
gateway 192.168.1.1
dns-nameservers 192.168.1.1
wireless-power off
wpa-conf /etc/wpa_supplicant/wpa_supplicant.conf
```
_replace `192.168.1.1` with router IP_

While multiple hotspots are nice, `/etc/hostapd/hostapd.conf` defines the interface it listens on clearly, so I'd not bother with renaming that but try first to get things working with the minimal required changes. If you can connect to the WiFi hotspot, then the DHCP server is already configured fine and running as it should, I assume?

Assure that forwarding is generally enabled:
```sh
sysctl net.ipv4.ip_forward net.ipv6.conf.all.forwarding net.ipv6.conf.default.forwarding
```
should all report `1`.

And now for accessing the network through the WiFi hotspot, you want clients connected to the hotspot to use the onboard WiFi as shared network/internet adapter?

Then the iptables rules should be that:
```sh
iptables -F # flush all filter rules
iptables -t nat -F # flush all nat rules
iptables -t nat -A POSTROUTING -o wlan0 -j MASQUERADE # enable NAT on wlan0
iptables -A FORWARD -i wlan0 -o wlan1 -m state --state RELATED,ESTABLISHED -j ACCEPT # allow forwarding of incoming packets from established connections
iptables -A FORWARD -i wlan1 -o wlan0 -j ACCEPT # allow forwarding of outgoing packets
iptables-save > /etc/iptables.ipv4.nat # store rules to be applied by ifupdown
```
Repeat the same with `ip6tables`, if required. That part is currently missing in our implementation, which may be even the issue in your case, when clients try to reach a public host via IPv6. I'll implement this short-term, but let's see first if it works for you.

chtugha, 2021-08-09:
Hi,

Thanks for the fast reply.

1st - wlan0 seems not to be always the onboard wifi. As I stated, installing hotspot with onboard-wifi off will give wlan0 to the external dongle.

2nd - /etc/network/interfaces.d/ is empty. Turning onboard-wifi on an off vi dietpi-config does not change that. I had to configure everything manually via /etc/network/interfaces.

3rd - Everything runs fine with only one hotspot, I am talking about adding another hotspot to the existing configuration.

4th - Therefore: the iptables rules were already set, and did not help with the issue :) and the rules you offer seem to expect wlan0 to be in some kind of WISP-configuration, while what I want to do is actually run several hotspots (at least one on 2,4 and one on 5ghz) on wlan0, wlan1 etc... So the Forwarding rules should point towards eth0.
You offer a solution for the issue [#2923](https://github.com/MichaIng/DietPi/issues/2923) an that is very different from the problem of adding another hotspot.

5th - the momentary solution with hostapd.service will NOT work with a second adapter. Therefore changing the configuration to hostapd@wlan0.service would be a very elegant solution to that...

MichaIng, 2021-08-09:
> installing hotspot with onboard-wifi off will give wlan0 to the external dongle.

Yes of course, if no onboard WiFi is present (disabled firmware-wise), then the external dongle is the first adapter found, hence wlan0. Okay but now I get your issue, when you want to configure the external one as hotspot, which is not possible without disabling onboard WiFi first. Then when enabling onboard WiFi, the interface name changes, so the hotspot config is active on the wrong adapter. Well, of course you could simply stick with it and use the onboard adapter as hotspot and the external one for outgoing access.

> /etc/network/interfaces.d/ is empty

Yes, it is exactly there for custom adapter configurations, as a workaround for the network setup limitations of `dietpi-config`. So the idea is to use `dietpi-config` for the base setup, where applicable, and use custom configurations, like the suggested one, to configure further interfaces.

> Everything runs fine with only one hotspot, I am talking about adding another hotspot to the existing configuration.

So you do actually want to run two WiFi hotspots on one Raspberry Pi? And then the onboard Ethernet for outgoing connections? If so, then sorry for my misunderstanding, else, lets stick with solving your particular need for now. Making `dietpi-config` more flexible, and integrating the WiFi+Ethernet hotspots/APs on a freely choosable interface is planned (see attached project).

> 4th ... 5th

Okay, again sorry for my misunderstanding then. Jep that requires a different configuration then. Shall the hotspots be able to communicate with each other all only with the network at the Ethernet adapter?

Then it makes sense (or is required) to run the instantiated hostapd services. Also you need to add the additional interfaces to `/etc/default/isc-dhcp-server` to enable the DHCP server on them. It also makes sense then to create individual iptables rules then, like:
```
cat << '_EOF_' > /etc/network/iptables.wlan0.nat
*nat
-A POSTROUTING -o eth0 -j MASQUERADE
COMMIT
*filter
-A FORWARD -i eth0 -o wlan0 -m state --state RELATED,ESTABLISHED -j ACCEPT
-A FORWARD -i wlan0 -o eth0 -j ACCEPT
COMMIT
_EOF_
cat << '_EOF_' > /etc/network/interfaces.d/wlan0.conf
allow-hotplug wlan0
iface wlan0 inet static
address 192.168.42.1/24
wireless-power off
up iptables-restore -n < /etc/network/iptables.wlan0.nat
up ip6tables-restore -n < /etc/network/iptables.wlan0.nat
_EOF_
```
The `-n` option is required here to not flush the rules applied for other interfaces.

Now you say you want to run multiple hotspots for multiple to cover multiple frequencies. I guess this means same network (IP range) and same SSID, right? Indeed this may involve routing difficulties, as there are conflicting link routes for e.g. `192.168.1.0/24` then, and I'm not sure currently and answer is automatically routed through the correct WiFi adapter, or may go to the wrong one. `ip rule`'s or routing tables may be a solution. E.g. new connections could get a mark, indicating the origin adapter. And depending on the mark, the route for incoming packets can then be forced. Simple would be probably to use different network ranges. To allow clients communicating with each others then, an additional set of iptables rules needs to allow forwarding between the WiFi interfaces then, but all routing should be correct automatically.