MichaIng/DietPi #4629: lighttpd index.html

Axellarator, 2021-08-09:
### Lighttpd index.html shows this: Bullet point 2

<p>The DocumentRoot, which is the directory under which all your HTML files should exist, is set to /var/www/html.</p>

#### I suggest to change this **/html** addressing in Dietpi

<p>Either to delete it or to change the referring documentation, so we talk about the same directory for HTML files.</p>

Joulinar, 2021-08-09:
The document root on DietPi is `/var/www`

Axellarator, 2021-08-09:
I know that honey. Index.html shows the wrong text. It points to /var/www/html as in Lighttpd documentation..

MichaIng, 2021-08-09:
Thanks for the hint. Fixed with: https://github.com/MichaIng/DietPi/commit/9af5454353d782613a4eb57f792ccf001c0c4618

Joulinar, 2021-08-09:
@MichaIng 
Something to add to our docs as well?

MichaIng, 2021-08-09:
We could add to all webserver docs that, other than the default on Debian, the webroot is `/var/www`. Just checked, and we have listed "Local directory = `/var/www`" in most stacks docs: https://dietpi.com/docs/software/webserver_stack/#lamp-web-stack
For the Nginx stacks, that is somehow missing.

We could probably also mention this as a note/warning box before the individual stack headers start?

Joulinar, 2021-08-09:
Not sure if an info box is needed. Maybe to many boxes for the user to click. What about giving a short info message during installation?

Axellarator, 2021-08-10:
I suggest simply remove the "/html" part during the lighttpd installation or adapt the lighttpd.conf with server.document-root = "/var/www/html", Benefit here it is consistent with the lighttpd docs cruising around and future updates. Apache is using this directory, too. But Dietpi and Lighttpd are working either way with Let's Encrypt and Duckdns.

Joulinar, 2021-08-10:
As already stated above, there will be a cosmetic adjustment on next release. As well we will add this to our docs for each individual web server. Btw, web server root is the same for all web server stacks. Doesn't matter if this is Apache, Nginx or Lighttpd. It's always `/var/www/`.

closed 2021-08-20 by MichaIng