MichaIng/DietPi #4630: Error running apt-get -q update

carlosgalarce, 2021-08-09:
#### Details:
- Date           | Mon Aug  9 16:05:01 CDT 2021
- Bug report     | ff2289cc-a171-4099-8d3f-cd634fe9f57b
- DietPi version | v7.1.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -q update
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Hit:1 https://downloads.plex.tv/repo/deb public InRelease
Hit:2 https://packages.grafana.com/oss/deb stable InRelease
Hit:3 https://download.docker.com/linux/raspbian buster InRelease
Hit:4 http://raspbian.raspberrypi.org/raspbian buster InRelease
Err:5 https://ookla.bintray.com/debian generic InRelease
  403  Forbidden [IP: 52.37.64.70 443]
Hit:6 https://archive.raspberrypi.org/debian buster InRelease
Reading package lists...
E: Failed to fetch https://ookla.bintray.com/debian/dists/generic/InRelease  403  Forbidden [IP: 52.37.64.70 443]
E: The repository 'https://ookla.bintray.com/debian generic InRelease' is no longer signed.
```

Joulinar, 2021-08-09:
Hi,

you have a 3rd party repository from `ookla` added to your source list and this doesn't seems to be valid. Did you installed any speed test software?

MichaIng, 2021-11-07:
I'll mark this as closed. Feel free to reopen when required.

closed 2021-11-07 by MichaIng