MichaIng/DietPi #4631: youtube-dl not working correctly when installed by dietpi-software

maartenlangeveld, 2021-08-10:
Have re-installed youtube-dl using dietpi-software.
However O!MPD won't play added Youtube streams anymore. When installed manually everything works fine.

MichaIng, 2021-08-10:
Thanks for reporting.

Does it work when you manually use it? And can you try to set the youtube path setting (https://github.com/ArturSierzant/OMPD/blob/master/include/config.inc.php#L258) to `/usr/local/bin/youtube-dl` explicitly?

And how did you install manually? Manually downloaded the Python script, or via `pip`, or different?

maartenlangeveld, 2021-08-10:
Hi MichalIng,

Yes it does when installed manually
```
wget https://yt-dl.org/downloads/latest/youtube-dl -O /usr/local/bin/youtube-dl >/dev/null 2>&1
chmod a+rx /usr/local/bin/youtube-dl
```
Paths defined as follows in include/config.local.inc.php : 
```
# paths
$cfg['python_path']							= '/usr/bin/python3';
$cfg['youtube-dl_path']						= '/usr/local/bin/youtube-dl';
```

maartenlangeveld, 2021-08-10:
May be I should change ownership of the batch file to www-data ?

MichaIng, 2021-08-10:
Please do not set the Python path, as the executable we install is a shell wrapper which includes the Python call. Then it should work. This is so that you can run `youtube-dl` as is from console without having it starting with Python 2 or failing when it's not installed.

MichaIng, 2021-08-10:
> May be I should change ownership of the batch file to www-data ?

No definitely not, the file is world-executable and that is all what's required. Only the set Python path setting is the issue, I think.

maartenlangeveld, 2021-08-10:
Hi MichalIng,

You are fabulous, I have disabled Python path setting and now it works...

Thank a lot for your great support!

MichaIng, 2021-08-10:
You're welcome 🙂. I think this is something we should document, just in case uses have those settings still set from previous installs or when being used to doing like that to invoke Python 3.

closed 2021-08-20 by MichaIng