MichaIng/DietPi #4632: Java not installing (fresh DietPi install)

NuclearManD, 2021-08-11:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
```
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=4
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `buster`
- Kernel version | `Linux DietPi 5.10.17+ #1421 Thu May 27 13:58:02 BST 2021 armv6l GNU/Linux`
- SBC model | RPi Zero W (armv6l)
- Power supply used | USB power going through my modded Sparkfun 5V/1A LiPo Charger/Booster, 3 large Schottky diodes added to supply power from USB
- SDcard used | SanDisk Ultra, 512 GB

#### Additional Information (if applicable)
- Software title | Java
- Was the software title installed freshly or updated/migrated? - No, it was being installed
- Can this issue be replicated on a fresh installation of DietPi? - Yes

`[ INFO ] DietPi-Bugreport | Reference code: 3a00f207-e889-41fa-82c3-e1ebef447cec`

#### Steps to reproduce
1. Install DietPi, without automatic headless setup
2. Login via SSH and follow the setup prompts
3. Request Java and Python 3 to be installed when DietPi-Software automatically opens (Python probably isn't related, but this is what I did)
4. Execute install
5. Wait for the below image to display

#### Expected behaviour
We should have Java successfully install

#### Actual behaviour
[image]

#### Extra details

- ...

Joulinar, 2021-08-11:
If I'm not mistaken, there will be a fix on upcoming version 7.5

@MichaIng pls correct me if I'm wrong

MichaIng, 2021-08-11:
Yes, fix is done here: https://github.com/MichaIng/DietPi/commit/38651957dc0c71be215ab8cedab37eb2df3a0d52

ARMv6 simply does not support the VM engine used since Java 9, so Java 8 needs to be installed. Workaround:
```sh
sed -i 's/version=11/version=8' /boot/dietpi/dietpi-software
```

What do you want to use Java for? Since Java 8 is quite outdated, it may not be supported for everywhere anymore. I'll see if there is some newer build that uses a different engine and hence runs on ARMv6: #4505
- https://github.com/adoptium/temurin17-binaries/releases
- https://github.com/AdoptOpenJDK/openjdk17-binaries/releases
- http://static.azul.com/zulu/bin/

maartenlangeveld, 2021-08-16:
Azul provides armv6 comaptible Open-JDK 11  !! Have installed it and works fine.

MichaIng, 2021-08-16:
Yes, this is also the only one I found so far. There is the alternative OpenJ9 engine, but no ARM 32-bit builds are shipped with it at all.

The problem is that Minecraft v1.17 requires Java 16 at least. Unless there is an explicit software title with runs with Java 11 but not with Java 8, I'd keep the current workaround which installs Java 8 on ARMv6 RPi models. At least even Raspbian Bullseye still ships it (other than Debian Buster/Bullseye).

maartenlangeveld, 2021-08-16:
Ok, thanks.
BTW, I have installed java 8 manually (apt install  openjdk-8-jre-headless) since i did not see in software list.

MichaIng, 2021-08-16:
Jep, that should do it. The fix and JDK/JRE separation has been implemented with v7.5, so it's in the current beta but not yet stable release.

maartenlangeveld, 2021-08-16:
Ok, that's fine! Thanks again.

closed 2021-08-20 by MichaIng