MichaIng/DietPi #4633: Add ability for custom script alongside automatic login

scorgn, 2021-08-12:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:
I want to be able to autostart with cli-visualizer instead of C.A.V.A.

#### Describe the solution you'd like:
I would like an option for a custom script autostart to be executed in shell after automatic login.

#### Describe alternatives you've considered:
An alternative would be to just select automatic login and then add conditionals similar to [those in dietpi-login](https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-login#L241-L247) to my .bashrc before running the script. However, I thought it would be better implemented as a native option.

#### Additional context
It wasn't immediately clear that some options would go alongside the automatic login option, so I was a bit confused when my custom script wasn't starting cli-visualizer the same way the cava autostart option was starting cava. It took a lot of digging to find out that some of the autostart options would automatically login before executing, while others wouldn't - and that the custom script was included in those that wouldn't.

I think that there are a lot of use cases for wanting a custom script to be executed after automatic login.

Joulinar, 2021-08-12:
Hi,

Did you tried auto login option 14?

`14 : Custom - /var/lib/dietpi/dietpi-autostart/custom.sh`

MichaIng, 2021-08-12:
Hmm, actually if you do autologin already, why not using `/etc/profile.d/` or `/etc/bashrc,d/` instead? If it doesn't matter which user is executing the script, also `/var/lib/dietpi/postboot.d/` can be used, as your implementation also executes the script in foreground.

The custom.sh autostart option is only useful in these situations:
- It doesn't matter who executes the script, respectively it must be the root user only, once at boot.
- And it is a long-running service/daemon process, which you want to be able to check status, start/stop/restart.

For oneshot tasks there exist these options already:
- Any script/executable inside `/var/lib/dietpi/postboot.d/` will be executed once at boot with root user, as part of the `dietpi-postboot.service`, with output printed to the main console. So you'll see it along with dietpi-service starts and before the login prompt banner (before actual login).
- Any bourne shell script inside `/etc/profile.d` will be sourced on any interactive login shell session, means you login on any console into an interactive shell, executed by user you login with, of course.
- Any shell script inside `/etc/bashrc.d` will be sourced on any interactive bash shell, means this is also triggered when you open a terminal emulator from a desktop, or open an interactive subshell, like running `bash` on a console etc, as well of course executed by the user you are currently logged in with.
- As all above options are sourced/executed in foreground, they delay the login prompt, respectively the command prompt for the 2. and 3. option by the time the scripts are running, and output is printed to the foreground console. `&` can be used to send tasks into background, but for long-term running tasks a systemd unit is the more elegant solution, so it can be transparently controlled via `systemctl`.

Now our custom.sh option is such a systemd unit, the way it is implemented currently. Your PR just executes the script in foreground, which should be covered by the other three options above? To be true I'm not sure why custom.sh is bound to autologin vs non-autologin at all, as it starts/runs completely independent from any login session. So it makes more sense to have it as an option which is indent from the `dietpi-autostart` choice. But this means to not add another autostart ID, but remove the custom.sh autostart ID and instead add it as a separate menu switch, so that the autostart index stays the same while the custom.sh can be created/enabled/disabled. Finally we could allow, either via menu options or via text editor, to customise the systemd unit further, when it shall start, whether it shall print to foreground console or not etc.

scorgn, 2021-08-12:
I suppose that the issue is not whether there is automatic login before execution, but whether it is ran in the foreground or the background. I suppose that with the current custom.sh script, automatic login is not relevant because it is just added as a systemd.service unit. I believe that automatic login occurs for everything that it is executed in the foreground.

The idea is that I did want the script to execute in the foreground, similar to how C.A.V.A. autostart option is currently working. I did look into postboot.d directory for services and am using that as well for things that run in the background, but I wanted to have programs open in the foreground only of the main TTY1 on boot without having to connect any keyboard or mouse. This way it would be ran in the foreground without needing to connect an input device to the pi or log in (similar to C.A.V.A. autostart option).

For adding it in `/etc/profile.d` or `/etc/bashrc.d`, it is not something that I want to execute on every shell. I only want to execute the script on the main shell (TTY1). It's the same type of execution that C.A.V.A. has. When the pi starts up, it automatically loads C.A.V.A. in the foreground (after automatic login). But when I ssh in, C.A.V.A. would not start in my shell. I could have added `[[ $(readlink -f /proc/$$/fd/1) == '/dev/tty1' ]] || break` (from [dietpi-login script](https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-login#L245)) to my .bashrc but it might not be immediately obvious how to do that for other users.

If the C.A.V.A autostart option did not exist, selecting this new option and adding just `cava` to the `custom.sh` would have the same exact effect as the C.A.V.A. autostart option. For me, the cli-visualizer autostart option does not exist.

I agree it would make sense to have it as a separate option from the autostart, or even possibly a sub-option of the "Custom" option (Do you want it to execute in the foreground or background?).

Adding menu options for how the systemd.service unit executes and where the output goes would also be a good idea, although I wonder if anyone who would want that granularity (and understand how that works) would have the knowledge to create their own systemd.service unit. Is it possible to have a systemd.service unit start in the foreground of the main TTY on boot?

MichaIng, 2021-08-13:
Thanks for your clarifications, now I understand better the aim.

`dietpi-login` itself is started from `/etc/bashrc.d/dietpi.bash`, but it does some checks first, most importantly the one you adopted: https://github.com/MichaIng/DietPi/blob/328adcb/dietpi/dietpi-login#L239-L247

Okay that sounds reasonable, a custom foreground script that executes after (auto)login on the main TTY1 only.

Btw, I checked the history about the current custom.sh implementation, and originally it was exactly what you propose: executed after autologin as foreground script: https://github.com/MichaIng/DietPi/commit/a8cf0fd7b9e33aa2bad8457976e5dd2a4d3316c7

This changed after an issue was found on OrangePi Zero: https://github.com/MichaIng/DietPi/commit/ed594f03ab46ea581b93b61a7b5cadda6ef7c952
Strange is, that on all OrangePi models supported that time, autostart options were skipped on any local console anyway and only ran for logins at the 1st SSH pseudo-terminal `/dev/pts/0`: https://github.com/MichaIng/DietPi/blob/ed594f0/dietpi/login#L57-L61
The OrangePi Zero has no HDMI/display port, so it makes sense that GUI applications cannot run at all and must be skipped. To try running them in an SSH session on the other hand is doomed to fail as well 😄. For a custom script on the other hand, which is not necessarily a GUI application. I also see the custom.sh service currently prints to the foreground console, so if a screen is attached, one will see its output there, around the login prompt. So it can still be used for GUI applications, but also for other things on headless devices, independently of a login.

I'd suggest the following, to separate the two options more clearly from each other:
- Change the current systemd version to not print to console anymore, so that it is a real background process, aimed for long running tasks to run once/permanent at boot, independently from any login. Preventing it from printing to TTY enables the ability to use `journalctl -u dietpi-autostart_custom` to reads logs/output, instead of having them on the main console only. Better for headless devices. And I agree that adding too much granularity does not make much sense, as experienced users can write their own systemd units.
- This makes your additional option a clearer separated choice for foreground processes, like it was when the feature was implemented the first time, e.g. for GUI applications, running on the main TTY1, hence a physical screen only, also with STDIN (keyboard) attached to it.

scorgn, 2021-08-22:
That all makes sense to me. Do you think that it should be implemented as a separate startup option, or as a sub-option of the current custom startup script? For example, when you choose custom script, it asks whether you want it to run in the foreground or the backend.

scorgn, 2021-08-22:
The more I think about it the more it makes sense to only have one custom script autostart option. 

Having a script autostart in the background is not mutually exclusive with any of the other startup options. It could be something that is always available (you just have to put code in a file), or it could be something you have to enable manually.

In this scenario I think using a @reboot crontab would make more sense. As a sidenote (related to crontab) I would like to put some time into making a crontab/command scheduler made with whiptail for dietpi, if you have any thoughts on that as well I'd love to hear them.

MichaIng, 2021-08-22:
As a first step, an additional option would be fine, but I agree in a second step the current once shouldn't be an exclusive option, i.e. it should be possible to add a custom systemd unit while still booting into Kodi kiosk mode or so. Only foreground autostart options need to be exclusive among each others, of course.

scorgn, 2021-08-25:
Okay sounds good. I've updated the MR and tested just to make sure it all works.

Also, I see now that the service was correctly outputting to the main console. I suppose the issue was that when I tried putting `cava` in the custom.sh script, there was no `$HOME` set so it couldn't find the configuration file.

MichaIng, 2021-08-25:
Another reason why it makes sense to run foreground processes after login from shell instead of via service 🙂. So the `cava` executable itself looks for the `$HOME` variable to determine the users home directory? I mean systemd units do not set the `$HOME` variable (it can be forced via `Environment=HOME=/root`), but of course there is a calling user and ways to determine its home directory without relying on the environment having this variable set.

Looks good to me with the latest change. I'll do further tests and polish after having the VNC rework finished.

closed 2021-09-16 by MichaIng