MichaIng/DietPi #4635: dietpi cannot install python3

nikobearrr, 2021-08-12:
#### Details:
- Date           | Thu Aug 12 14:57:56 BST 2021
- DietPi version | v7.4.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | FriendlyARM
- Hardware       | NanoPi K1 Plus (aarch64) (ID=67)
- Kernel version | `Linux pi 4.16.0-rc6 #19 SMP Fri Apr 20 14:25:42 CST 2018 aarch64 GNU/Linux`
- Distro         | stretch (ID=4)
- Command        | `apt-get -q update`
- Exit code      | 255
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
Get:1 https://packages.sury.org/php stretch InRelease [6839 B]
Err:1 https://packages.sury.org/php stretch InRelease
  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
Ign:2 https://deb.debian.org/debian stretch InRelease
Hit:3 https://download.docker.com/linux/debian stretch InRelease
Hit:4 https://deb.debian.org/debian stretch-updates InRelease
Hit:5 https://deb.debian.org/debian-security stretch/updates InRelease
Hit:6 https://deb.debian.org/debian stretch-backports InRelease
Hit:7 https://deb.debian.org/debian stretch Release
Fetched 6839 B in 3s (2137 B/s)
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://packages.sury.org/php stretch InRelease: The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Failed to fetch https://packages.sury.org/php/dists/stretch/InRelease  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

Joulinar, 2021-08-12:
Hi,

this is basically know. Can you try

```
apt-key del '95BD4743'
curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
```

nikobearrr, 2021-08-12:
Hi @Joulinar, that helped. Here is the output:

```
root@pi:~# apt-key del '95BD4743'
OK
root@pi:~# curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg
root@pi:~# /boot/dietpi/func/dietpi-set_software apt-cache clean

 DietPi-Set_software
─────────────────────────────────────────────────────
 Mode: apt-cache (clean)

[  OK  ] apt-cache clean | Completed
root@pi:~# apt update
Get:1 https://packages.sury.org/php stretch InRelease [6,839 B]
Get:2 https://packages.sury.org/php stretch/main arm64 Packages [320 kB]
Ign:3 https://deb.debian.org/debian stretch InRelease
Get:4 https://download.docker.com/linux/debian stretch InRelease [44.8 kB]
Get:5 https://deb.debian.org/debian stretch-updates InRelease [93.6 kB]
Get:6 https://deb.debian.org/debian-security stretch/updates InRelease [53.0 kB]
Get:7 https://deb.debian.org/debian stretch-backports InRelease [91.8 kB]
Get:8 https://download.docker.com/linux/debian stretch/stable arm64 Packages [14.4 kB]
Get:9 https://deb.debian.org/debian stretch Release [118 kB]
Get:10 https://deb.debian.org/debian stretch Release.gpg [2,410 B]
Get:11 https://deb.debian.org/debian-security stretch/updates/non-free arm64 Packages [5,320 B]
Get:12 https://deb.debian.org/debian-security stretch/updates/contrib arm64 Packages [1,760 B]
Get:13 https://deb.debian.org/debian-security stretch/updates/main arm64 Packages [686 kB]
Get:14 https://deb.debian.org/debian stretch-backports/non-free arm64 Packages [6,588 B]
Get:15 https://deb.debian.org/debian stretch-backports/main arm64 Packages [598 kB]
Get:16 https://deb.debian.org/debian stretch-backports/contrib arm64 Packages [7,828 B]
Get:17 https://deb.debian.org/debian stretch/contrib arm64 Packages [39.6 kB]
Get:18 https://deb.debian.org/debian stretch/main arm64 Packages [6,921 kB]
Get:19 https://deb.debian.org/debian stretch/non-free arm64 Packages [50.9 kB]
Fetched 9,062 kB in 8s (1,049 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
16 packages can be upgraded. Run 'apt list --upgradable' to see them.```

I see you have put Known issue and I am sorry I couldn't find that. Is there a place for such common/known/faq things that "could" go wrong?

Joulinar, 2021-08-12:
Basically we had a leading issue created for this some time back #4219

It's closed in meantime. I thought DietPi update would correct this. @MichaIng maybe I'm wrong.

Anyway, you could always search our GitHub for reference words. In your case I used `sury.org` as this was the repository failing.

nikobearrr, 2021-08-12:
Thanks for the information @Joulinar.

I went for your advice and tried to upgrade it. Complete failure 😅. However as I said, I didn't run anything but pihole so I just re-downloaded the latest and working on it. 

I think we can close the issue. Thanks for all the help

closed 2021-08-12 by nikobearrr

Joulinar, 2021-08-12:
You could have ask :lol: we have quite a good description on how to move from Stretch to Buster. But indeed, just running Pihole is quite an easy thing to set it up again :wink:

MichaIng, 2021-08-13:
Pretty strange as indeed the key should have been renewed automatically via pre-patch during the DietPi update: https://github.com/MichaIng/DietPi/blob/14998ff/.update/pre-patches#L27-L37

Would have been interesting, whether the following shows the expired key on your system or not, not too late to test now, I guess 😄:
```sh
apt-key list '95BD4743'
```