MichaIng/DietPi #4642: USB NAND Drive Bricked x2

Fireman1224, 2021-08-15:
Here is the result of `lsblk` after booting from the microSD with the USB drive installed.

```
root@PiHole-2:~# lsblk
NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
mmcblk0     179:0    0 14.9G  0 disk
├─mmcblk0p1 179:1    0 41.8M  0 part /boot
└─mmcblk0p2 179:2    0 14.8G  0 part /
root@PiHole-2:~#
```

The USB is not listed, only the 16gb microSD.

Should we move the conversation to a different thread?

Thanks
The USB drive is not listed, only the 16gb microSD.

Should we move the conversation to a different thread?

Thanks

_Originally posted by @Fireman1224 in https://github.com/MichaIng/DietPi/issues/3708#issuecomment-899042357_

Fireman1224, 2021-08-15:
@MichaIng @Joulinar 
`
root@PiHole-2:~# dmesg

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.66-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1253 SMP Thu Aug 15 11:49:46 BST 2019
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3dc00000
[    0.000000] On node 0 totalpages: 255488
[    0.000000]   Normal zone: 2246 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 255488 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xac/0x4b4 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s39488 r8192 d21952 u69632
[    0.000000] pcpu-alloc: s39488 r8192 d21952 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 253242
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=720 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=PARTUUID=27504eef-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 989820K/1021952K available (8192K kernel code, 629K rwdata, 2176K rodata, 1024K init, 821K bss, 23940K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbe800000 - 0xff800000   (1040 MB)
                   lowmem  : 0x80000000 - 0xbe600000   ( 998 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 630 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 822 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 26394 entries in 78 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000023] Switching to timer-based delay loop, resolution 52ns
[    0.000291] Console: colour dummy device 80x30
[    0.000916] console [tty1] enabled
[    0.000978] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.001023] pid_max: default: 32768 minimum: 301
[    0.001379] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001415] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.002348] CPU: Testing write buffer coherency: ok
[    0.002830] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.003496] Setting up static identity map for 0x100000 - 0x10003c
[    0.003664] rcu: Hierarchical SRCU implementation.
[    0.004478] smp: Bringing up secondary CPUs ...
[    0.005338] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.006268] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.007130] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.007250] smp: Brought up 1 node, 4 CPUs
[    0.007331] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.007355] CPU: All CPU(s) started in HYP mode.
[    0.007375] CPU: Virtualization extensions available.
[    0.008374] devtmpfs: initialized
[    0.021429] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.021703] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.021752] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.022358] pinctrl core: initialized pinctrl subsystem
[    0.023225] NET: Registered protocol family 16
[    0.026190] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.031897] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.031933] hw-breakpoint: maximum watchpoint size is 8 bytes.
[... 164 lines cut ...]
[    3.252235] lan78xx 1-1.1.1:1.0 (unnamed net_device) (uninitialized): int urb period 64
[    3.417023] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    3.667042] systemd-journald[119]: Received request to flush runtime journal from PID 1
[    3.910608] Adding 1097724k swap on /var/swap.  Priority:-2 extents:7 across:1171452k SSFS
[    3.995107] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[    4.003846] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[... cut ...]
[    8.821984] usb 1-1.1.2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[    8.822430] usb-storage 1-1.1.2:1.0: USB Mass Storage device detected
[    8.828540] scsi host0: usb-storage 1-1.1.2:1.0
[    9.153651] usbcore: registered new interface driver uas
[    9.832209] scsi 0:0:0:0: Direct-Access     NAND     USB2DISK         0.00 PQ: 0 ANSI: 4
[    9.833976] sd 0:0:0:0: [sda] Attached SCSI removable disk
[   10.196077] sd 0:0:0:0: Attached scsi generic sg0 type 0
[  540.666822] systemd: 30 output lines suppressed due to ratelimiting
root@PiHole-2:~#`

Joulinar, 2021-08-15:
Looks like the NAND disk is detected as `sda` 

```
[ 9.832209] scsi 0:0:0:0: Direct-Access NAND USB2DISK 0.00 PQ: 0 ANSI: 4
[ 9.833976] sd 0:0:0:0: [sda] Attached SCSI removable disk
[ 10.196077] sd 0:0:0:0: Attached scsi generic sg0 type 0
```

Fireman1224, 2021-08-15:
I did see that, but on a rasbian no drive comes up...

It also does nothing when the microSD card is removed and the device is started up (only the power LED comes on, no blinking LED).

Any idea how to fix?

Thanks

MichaIng, 2021-08-15:
Strange that `lsblk` doesn't show it. Does fdisk do?
```sh
fdisk -l
```
Also:
> [  540.666822] systemd: 30 output lines suppressed due to ratelimiting

Would be interesting what these 30 lines are. Kernel message rate limiting can be disabled until reboot with:
```sh
echo 0 > /proc/sys/kernel/printk_ratelimit
echo 0 > /proc/sys/kernel/printk_ratelimit_burst
```

And:
```sh
vcgencmd get_throttled
```

Fireman1224, 2021-08-15:
root@PiHole-2:~# fdisk -l
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram8: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram9: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram10: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram11: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram12: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram13: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes
[... 92 lines cut ...]
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram14: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram15: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/mmcblk0: 14.9 GiB, 15931539456 bytes, 31116288 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x27504eef

Device         Boot Start      End  Sectors  Size Id Type
/dev/mmcblk0p1       8192    93802    85611 41.8M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      98304 31116287 31017984 14.8G 83 Linux
root@PiHole-2:~# [[ -b /dev/sda ]] && echo 'Block device is there'
Block device is there
root@PiHole-2:~#

Fireman1224, 2021-08-15:
```
root@PiHole-2:~# echo 0 > /proc/sys/kernel/printk_ratelimit
root@PiHole-2:~# echo 0 > /proc/sys/kernel/printk_ratelimit_burst
root@PiHole-2:~# vcgencmd get_throttled
throttled=0x0
```

MichaIng, 2021-08-15:
Can you run these:
```sh
partprobe /dev/sda
partx -u /dev/sda
parted /dev/sda print
```
The device file is there, but if not tool is able to detect it as block device, then I cannot imagine the drive content being the issue, but the hardware/device/firmware itself. I mean the tools are not even able to format the drive. If you flashed the image from Windows, obviously Windows is able to detect the drive. Does it detect the /boot FAT partition as well, else what does the Windows disk management show?

Fireman1224, 2021-08-15:
```
root@PiHole-2:~# partprobe /dev/sda
Error: Error opening /dev/sda: No medium found
root@PiHole-2:~# partx -u /dev/sda
partx: cannot open /dev/sda: No medium found
root@PiHole-2:~# parted /dev/sda print
Error: Error opening /dev/sda: No medium found
Retry/Cancel? Cancel
root@PiHole-2:~#
```
From Windows Device Manager:
[image]

From Windows Drive Manager:
[image]

MichaIng, 2021-08-15:
I guess you attached it to a USB3 port, right? Could you try (with a fresh image) to use a USB2 port? Another idea I have is that UAS causes issues with that stick. It is a small USB stick, no 2.5" SSD, right?

Fireman1224, 2021-08-15:
The pi is a 3 b+, so there are no USB 3.  My Laptop has no USB 2 ports and
I don't have a USB 2 hub.
Dee McKeel
***@***.***

MichaIng, 2021-08-15:
Ah right, then also UAS is not a topic.

> Not to mention I'm out of USB drives

You mean also on Windows you are unable to format that drives?

Fireman1224, 2021-08-15:
That's correct. It's like the partition info is lost and not recoverable.

MichaIng, 2021-08-15:
Well the partition info is not required for being able to format the drive. Re-creating the partition table is a common task. But the problem on Linux is that even the tools which are there for creating partition tables do not detect the drive as accessible block device.

Here are similar cases:
- https://unix.stackexchange.com/questions/359571/external-drive-recognized-by-ls-dev-but-no-medium-found
- https://www.lowendtalk.com/discussion/67289/how-to-revive-a-no-medium-found-usb-drive
- https://bbs.archlinux.org/viewtopic.php?id=148756

But I cannot imagine that the RPi really broke the USB stick, as this must mean serious overvoltage on the USB port, or so.
The following does not work either, does it?
```sh
dd if=/dev/zero of=/dev/sda bs=8M
```

Fireman1224, 2021-08-15:
```
root@PiHole-2:~# dd if=/dev/zero of=/dev/sda bs=8M
dd: failed to open '/dev/sda': No medium found
root@PiHole-2:~#
```

MichaIng, 2021-08-15:
So indeed it seems to be broken then. Were these known brands, probably some warranty on them, or very old ones? As said, I cannot really imagine that the RPi killed them, without them being close to dead already, but risky/expensive to really debug the case. Also strange that it worked exactly until reboot, without starting to get I/O errors earlier.

Always good to keep an eye on RPi repositories and forum, just in case someone had a similar issue, like if is something with the newest kernel/firmware release:
- https://github.com/raspberrypi/firmware/issues
- https://github.com/raspberrypi/linux/issues
- https://www.raspberrypi.org/forums/

Fireman1224, 2021-08-15:
These were cheap amazon bulk purchases, but I've not had any issues in other applications with these drives.  I'm not sure what to think.  I was going to try a name brand 16gb, but I'm a little worried it may brick it too...

MichaIng, 2021-08-16:
> I'm not sure what to think. I was going to try a name brand 16gb, but I'm a little worried it may brick it too...

Understandable. Probably try it first with Raspberry Pi OS Lite, update all packages and kernel, and see ich that works stable. If it bricks again, it is an issue with the RPi and at least it's worth trying to contact the RPi support about it. Not sure if monitoring voltage on the port of some time would give any hint...

Fireman1224, 2021-08-16:
I don't think it's likely that it was an over voltage issue because I tried it with 2 separate RPi 3's.  The likelyhood that 2 Pi's had over voltage on a USB is extremely unlikely...

Fireman1224, 2021-08-16:
@MichaIng 

I decided to try the Sandisk 16gb thumb drive.  I installed the Pi OS from their website and was able to install everything I needed and it worked properly.  I then decided to wipe the drive and re-downloaded the DietPi image from your website and try it again, this time doing installations one at a time and rebooting between each.  Basically I installed DietPi with no added software, then rebooted, installed Fail2Ban, rebooted, PiHole, rebooted, PiVPN, rebooted.

Everything seems to be working fine.  I guess the cheap memory had issues.... I really don't have a clue what could have been different about them.  I really appreciate your help with troubleshooting this!!

Thanks

MichaIng, 2021-08-16:
Okay, I'm glad the new thumb drive was not toasted as well. Then indeed it seems the other two were too cheap, although still strange that both died to close after another, like with a self-destruction timer included 😄.

closed 2021-08-16 by Fireman1224

Joulinar, 2021-08-16:
if they are still in warranty, try to return them 😃 

Personally I'm using a SanDisk Ultra Fit 16Gb. It's small and thin USB stick but working quite well.

SinghForceOne, 2025-01-31:
I've had the exact same issue whilst installing Linux mint from live Install to two separate 128gb USB sticks. The install on two separate occasions bricked both USBs. I've earlier had success with a SANDISK 64GB USB stick, but recently wanted to upgrade the install to a 128GB stick, wich has since been bricked ... twice. I get a I/O error saying it had problems writing install and that the install crashed. Then the 128gb USB is not mountable thereafter, though in DISK it comes up as /dev/sda; and when plugging in/out it rings the linux USB connection notification bell. DISKS sees it as  NAND USB2DISK, with no partitions and no actions can be performed on it. Bricked.