MichaIng/DietPi #4643: Dietpi-Software | pijuice fails to install

w1v1, 2021-08-16:
### Creating a bug report/issue

#### Required Information
- DietPi version | `7 / 5 / 0 / beta / MichaIng`
`
- Distro version | `bullseye (ID=6,RASPBIAN=0)`
- Kernel version | `Linux DietPi 5.10.52-v8+ (aarch64 GNU/Linux)`
- SBC model | `RPi 4 Model B (aarch64)`
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | pijuice-base
- Was the software title installed freshly or updated/migrated? Freshly
- Can this issue be replicated on a fresh installation of DietPi? Yes - was done on a fresh install

- Bug report ID | `7c73ceca-a5cf-447d-85f8-0ab3f99b2e0a`

#### Steps to reproduce

1. Select and install the software usign `dietpi-software`
2. See install error

#### Expected behaviour

- Successful install

#### Actual behaviour

- Failed install

#### Extra details
- Command        | `apt-get -qq --allow-change-held-packages install pijuice-base`
- Exit code      | 100
- Software title | DietPi-Software

#### Additional logs:
```
E: Unable to locate package pijuice-base
```

Joulinar, 2021-08-16:
Hi,

Thy for your message. Looks like the `pijuice-base` package is missing on Raspberry OS 64bit.

It seems you are running a fresh empty system on Bullseye. Probably you would need to use the `ARMv7` or `ARMv6` image instead.

MichaIng, 2021-08-16:
The package is architecture independent, but its a Buster vs Bullseye issue: It has not yet been build to the RPi Bullseye repository.

Temporary workaround is installing the Buster suite additionally:
```
echo 'deb https://archive.raspberrypi.org/debian/ buster main' > /etc/apt/sources.list.d/raspi-buster.list
apt update
apt install pijuice-base
```

Joulinar, 2021-08-16:
I guess we would need to disable on Bullseye for time being.

MichaIng, 2021-08-16:
Or we manually download the package, if not found in repos.

Joulinar, 2021-08-16:
or to add the buster repository for `pijuice` 😃

MichaIng, 2021-08-16:
Fixed with: https://github.com/MichaIng/DietPi/commit/0968bd628f439154c7ced6f5c2738066caa4edc3
Changelog: https://github.com/MichaIng/DietPi/commit/0d198ffc9b397cfdb043d0984508363853cba8ee

closed 2021-08-20 by MichaIng